ghdl/ghdl #851: Exception USE_ERROR

pmerinog70, 2019-06-18:
**Description**
After compiling ghdl with gcc backend, I get an exception error when I try to compile a VHDL file.

**Expected behaviour**
Error compiling a VHDL package.

**Context**
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (tarball) [Dunoon edition]
Compiled with GNAT Version: Community 2019 (20190517-83)
Target: x86_64-pc-linux-gnu

Execution of ghdl terminated by unhandled exception
raised ADA.IO_EXCEPTIONS.USE_ERROR : GNAT.Directory_Operations.Get_Current_Dir: current directory does not exist
Call stack traceback locations:
0x514912 0x51495e 0x41931c 0x49fe30 0x5049c6 0x40644f 0x7f92d2dcd0c3 0x40574b

- OS: Linux
- Origin:
  - [ ] Package manager. Repo:
  - [ ] Released binaries. Tarball:
  - [x] Built from sources. Commit SHA: Downloaded as a tar.gz directly from ghdl website

**Additional context**

**How to reproduce?**
I think that the issue is related with ghdl build with gcc with after an apparently successful build.

**Files**
N/A

**Checklist**
Before submitting your issue, please review the following checklist:

- [x] Add `GHDL Bug occurred` log block
- [ ] Add a MWE
- [x] Try the latest version

tgingold, 2019-06-18:
A way to reproduce the error would be nice, but I fear you deleted a directory or a file.  I have never seen this error.

Can you reproduce the error on your computer ?

pmerinog70, 2019-06-18:
I've got it. I was doing some tests with the ghdl (mcode backend) and moved the directory to another one using a shell in a different window. When I tried to run the new ghdl (gcc backend) I didn't notice that the directory had dissappeared.

Sorry for the inconvenience.

closed 2019-06-18 by pmerinog70