ghdl/ghdl #852: VHDL 2008

Gabriel-Degret, 2019-06-21:
**Description**
The folowing code crash:

```
library ieee;
    use ieee.std_logic_1164.all;
    use ieee.numeric_std.all;
    
entity recordOfRecord_tb is
  generic
  (       NB_CHAN_G : positive := 2
   ;      W_ADR_G   : positive := 11
   ;      W_DAT_G   : positive := 16
  );
end recordOfRecord_tb;
architecture tb of recordOfRecord_tb  is 
  type dmn_t is record 
    clk: std_ulogic;
    rst: std_ulogic;
  end record;

  signal dmn_i_s : dmn_t;
  signal stop_s : std_logic := '0';
  
  type fifo_i_t is record
    rx_cyc   : std_ulogic;                     
    rx_stb   : std_ulogic;                     
    rx_tga   : std_ulogic_vector;              
    rx_cti   : std_ulogic_vector( 1 downto 0 );
    rx_dat   : std_ulogic_vector;              
    tx_ack   : std_ulogic;                     
    tx_stall : std_ulogic;   
  end record;
  
  type fifo_o_t is record
    rx_ack   : std_ulogic;                      
    rx_stall : std_ulogic; 
    tx_cyc   : std_ulogic;                      
    tx_stb   : std_ulogic;                      
    tx_tga   : std_ulogic_vector;               
    tx_cti   : std_ulogic_vector( 1 downto 0 ); 
    tx_dat   : std_ulogic_vector;               
  end record;
  
  type fifo_array_i_t is array ( natural range <> ) of fifo_i_t;
  type fifo_array_o_t is array ( natural range <> ) of fifo_o_t; 
  
  signal arr_fifo_o 
  : fifo_array_o_t( 0 to NB_CHAN_G - 1 )
   (   tx_tga( W_ADR_G - 1 downto 0 ) 
   ,   tx_dat( W_DAT_G - 1 downto 0 )
  );

  type reg_t is record

    arr_fifo_i 
    : fifo_array_i_t( 0 to NB_CHAN_G - 1 )
     (   rx_tga( W_ADR_G - 1 downto 0 ) 
     ,   rx_dat( W_DAT_G - 1 downto 0 )
    );  

  end record;
    
  signal a
  ,      r
  : reg_t;
begin 
  
  process
  begin
    dmn_i_s.clk <= '0';
    wait for 5 ns;
    dmn_i_s.clk <= '1';
    wait for 5 ns;
    if stop_s = '1' then
      wait;
    end if;
  end process;
  
  process
  begin
    dmn_i_s.rst <= '1';
    wait for 50 ns;
    dmn_i_s.rst <= '0';
    wait;
  end process;
  
  process
  begin
    arr_fifo_o( 0 ).tx_cyc <= '0';
    arr_fifo_o( 0 ).tx_cti <= "00";
    wait until dmn_i_s.rst = '0';
    wait until rising_edge( dmn_i_s.clk );
    arr_fifo_o( 0 ).tx_cyc <= '1';
    wait until rising_edge( dmn_i_s.clk );
    wait until rising_edge( dmn_i_s.clk );
    arr_fifo_o( 0 ).tx_cyc <= '0';
    wait for 50 ns;
    stop_s <= '1';
    wait;
  end process;

  process( dmn_i_s.rst , r , arr_fifo_o )
    variable a_v : reg_t;
  begin

    a_v := r;

    if arr_fifo_o( 0 ).tx_cti = "00" then
      if arr_fifo_o( 0 ).tx_cyc = '1' and r.arr_fifo_i( 0 ).tx_ack = '1' then
        a_v.arr_fifo_i( 0 ).tx_ack := '0';
      else
        a_v.arr_fifo_i( 0 ).tx_ack := '1' and arr_fifo_o( 0 ).tx_cyc;  
      end if;
    else
      a_v.arr_fifo_i( 0 ).tx_ack := arr_fifo_o( 0 ).tx_cyc;
    end if;

    if dmn_i_s.rst = '1' then
      a_v.arr_fifo_i( 0 ).tx_ack := '0'; 
    end if;
    
    a <= a_v;
  end process;
  

  process( dmn_i_s.clk )
  begin 
    if rising_edge( dmn_i_s.clk ) then 
      r <= a;
    end if;
  end process;

-- notpad++ script
-- cd D:\vhdl
-- set cc=.\sim\ghdl\0.36-rc1-mingw32-mcode\bin\ghdl.exe

-- set sim_param= -PD:\vhdl\sim\ghdl\simSpace --workdir=D:\vhdl\sim\ghdl\simSpace --std=08

-- $(cc) -i $(sim_param) --work=ghdl_test .\ghdl_test\*.vhd

-- $(cc) -a $(sim_param) --work=ghdl_test .\ghdl_test\recordOfRecord_tb.vhd

-- $(cc) -m $(sim_param) --work=ghdl_test recordOfRecord_tb tb

-- $(cc) -r $(sim_param) --work=ghdl_test recordOfRecord_tb tb --wave=./sim/ghdl/simSpace/out/recordOfRecord_tb.ghw
end tb;
--#######################################################################################
--#######################################################################################
--#######################################################################################
--#######################################################################################
--#######################################################################################
```
Is this a bug or am I doing something wrong?

tgingold, 2019-06-21:
There is indeed a bug while dumping the waveforms.  You can still simulate without using the `--wave` option.

eine, 2020-08-05:
FTR, with `GHDL 1.0-dev (v0.37.0-924-g36f25166)` the error when opening the waveform is:

```sh
# gtkwave mywave.ghw

GTKWave Analyzer v3.3.104 (w)1999-2020 BSI

ghw_read_type: unknown type 153
ghw_read_base: error in section TYP
Error in ghw file 'mywave.ghw'.
```

umarcor, 2021-08-15:
This issue seems to be fixed with a recent version of GHDL and GTKWave. 

@Gabriel-Degret, can you please confirm?

boehmerst, 2022-07-08:
Seems like I ran into the same issue. I use unconstrained arrays of unconstrained record. 

GHDL version

```
GHDL 3.0.0-dev (tarball) [Dunoon edition]
 Compiled with GNAT Version: 9.4.0
 mcode code generator
Written by Tristan Gingold.
```

gtkwave version:

```
GTKWave Analyzer v3.3.103 (w)1999-2019 BSI

This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
gtkwave error mesage:

```
GTKWave Analyzer v3.3.103 (w)1999-2019 BSI

ghw_read_type: unknown type 153
ghw_read_base: error in section TYP
Error in ghw file 'xbar_2008.ghw'.
```

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

package xbar_2008_pkg is

  type xbar_ctrl_t is record
    addr : std_ulogic_vector;
    sel  : std_ulogic_vector;
    wr   : std_ulogic;
    en   : std_ulogic;
  end record xbar_ctrl_t;

  function dflt_xbar_crtl(t : in xbar_ctrl_t) return xbar_ctrl_t;

  type xbar_req_t is record
    ctrl  : xbar_ctrl_t;
    wdata : std_ulogic_vector;
  end record xbar_req_t;

  function dflt_xbar_req(t : in xbar_req_t) return xbar_req_t;

  type xbar_rsp_t is record
    rdata : std_ulogic_vector;
  end record xbar_rsp_t;

  function dflt_xbar_rsp(t : in xbar_rsp_t) return xbar_rsp_t;

  type xbar_req_arr_t is array(natural range<>) of xbar_req_t;
  type xbar_rsp_arr_t is array(natural range<>) of xbar_rsp_t;

end package xbar_2008_pkg;

package body xbar_2008_pkg is

  function dflt_xbar_crtl(t : in xbar_ctrl_t) return xbar_ctrl_t is
    variable v : xbar_ctrl_t(addr(t.addr'range), sel(t.sel'range));
  begin
    v.addr := (v.addr'range => '0');
    v.sel  := (v.sel'range  => '0');
[... 360 lines cut ...]
  ------------------------------------------------------------------------------
  -- clock and reset
  ------------------------------------------------------------------------------
  clk           <= not clk after clk_cycle_c/2 when sim_done = false;
  reset_n       <= '1' after 500 ns;

   ------------------------------------------------------------------------------
  -- stimuli
  ------------------------------------------------------------------------------
  stim0: process is
  begin
    wait until reset_n = '1';
    wait until rising_edge(clk);
    
    wait for 2 us;

    -- quit simulation
    wait for 10 us;
    sim_done <= true;
    assert false report "Simulation finished successfully!" severity failure;
    wait;
  end process stim0;

end architecture beh;
```

Gabriel-Degret, 2023-06-05:
Ok for me with

GHDL 3.0.0-dev (2.0.0.r1248.g1b0c3819a) [Dunoon edition]
Compiled with GNAT Version: 12.2.0
mcode code generator

and 
 gtkwave [048a5c1]

umarcor, 2023-06-05:
@boehmerst GTKWave 3.3.103 was released in November 2019: https://github.com/gtkwave/gtkwave/blob/master/gtkwave3-gtk3/ChangeLog#L1779. You need a version of GTKWave release after August 2021 (https://github.com/ghdl/ghdl/issues/852#issuecomment-899130812).

closed 2023-06-05 by umarcor