ghdl/ghdl #855: GHW wave generation fails for array in record in generic package

1138-4EB, 2019-06-24:
I'm trying to use a record defined in a generic package. The record contains a single `std_logic_vector` with the length defined by a generic parameter of the package. This produces an error when reading the generated GHW waveforms with GtkWave: `ghw_read_base: error in section HIE`.

The error is not produced if a normal package (non-generic) is used instead.

It is also possible to use VCD, which does not fail. However, this prevents any of the signals from being dumped, so it is useless.

I tried this in Debian based docker images (e.g. ghdl/ext:ls-vunit-gtkwave). I.e, the latest GHDL with LLVM backend and GtkWave v3.3.101.

``` vhdl
#>> ent.vhd
library ieee;
context ieee.ieee_std_context;

package pkg_types is
  constant g_d: positive := 32;
  type MO_RW is record
    DAT: std_logic_vector(g_d-1 downto 0);
  end record;
end package;

---

library ieee;
context ieee.ieee_std_context;

package gen_types is
  generic ( g_d: positive );
  type MO_RW is record
    DAT: std_logic_vector(g_d-1 downto 0);
  end record;
end package;

---

package wb is new work.gen_types generic map ( 16 );

---

use work.wb;
use work.pkg_types;

entity ent is
end entity;

architecture a of ent is
  signal asignal: pkg_types.MO_RW;  -- this works ok
  signal bsignal: wb.MO_RW;         -- this produces the error
begin
end;

#>> sim.sh
ghdl -a ent.vhd
ghdl --elab-run ent

#>> run.sh
docker run --rm -tv /$(pwd):/src:z -w //src ghdl/ghdl:buster-mcode sh -c ./sim.sh

#>> end
```

eine, 2020-08-07:
With `GHDL 1.0-dev (v0.37.0-949-gc689f6ad)`, a segmentation fault is produced when running `./ent --wave=wave.ghw`.

tgingold, 2020-08-07:
A PR in gtkwave has been merged yesterday.  You should try to rebuild gtkwave.

eine, 2020-08-07:
I saw that, and that's why I tried to test this issue again. However, when I opened this issue, the problem was when opening the waves with gtkwave. Today, the problem is that the simulation crashes at runtime, hence, no valid waveforms are generated. That is, this is failing before I can try GtkWave.

```sh
# ghdl -a --std=08 ent.vhd

# ghdl -e --std=08 ent

# ./ent.exe --wave=wave.ghw
Segmentation fault

# ./ent.exe --vcd=wave.vcd

# ls -lah
total 1.9M
drwxr-xr-x 1 eine None    0 Aug  7 12:18 .
drwxr-xr-x 1 eine None    0 Aug  7 12:17 ..
-rw-r--r-- 1 eine None 4.4K Aug  7 12:17 e~ent.o
-rwxr-xr-x 1 eine None 1.7M Aug  7 12:17 ent.exe
-rw-r--r-- 1 eine None 8.7K Aug  7 12:17 ent.o
-rw-r--r-- 1 eine None  702 Aug  7 12:17 ent.vhd
-rw-r--r-- 1 eine None    0 Aug  7 12:18 wave.ghw
-rw-r--r-- 1 eine None  463 Aug  7 12:18 wave.vcd
-rw-r--r-- 1 eine None  295 Aug  7 12:17 work-obj08.cf

# gtkwave.exe wave.ghw

GTKWave Analyzer v3.3.105 (w)1999-2020 BSI

Error opening ghw file 'wave.ghw', rc=-1.

# gtkwave.exe wave.vcd

GTKWave Analyzer v3.3.105 (w)1999-2020 BSI

No symbols in VCD file..nothing to do!
```

@tgingold, as you can see in the log, `wave.ghw` is empty.