ghdl/ghdl #858: GHDL LLVM installation fails at configure step

Defossez, 2019-07-02:
Hi,

I'm trying to install GHDL with LLVM on Ubuntu 18.04/Mint-19 and it fails at the configure step.
I've installed LLVM/CLANG following the steps on this [page](https://apt.llvm.org/).
- Setup the repositories for LLVM-8
- Follow the install steps given on the page
A test of the installation showed success.
Now I'm trying to build GHDL with this command
- `../configure --with-llvm-config=/usr/lib/llvm-8/bin/llvm-config --prefix=/opt/Ghdl`
results in this output in the terminal:
```
Use full IEEE library
Build machine is: x86_64-linux-gnu
Unhandled version llvm 8.0.1
```
running `/usr/lib/llvm-8/bin/llvm-config --version` shows 8.0.1

Why does the configure step stop with : Unhandled version llvm 8.0.1?

I know that the documentation says that up to LLVM 5.0 is supported, but till now I've been successfully running, on another Linux machine, GHDL with LLVM 7 and wanted to try for a new machine (Mint-19) the latest LLVM.

Many thanks in advance,

Marc

tgingold, 2019-07-02:
A few days ago I added the support for llvm-8, so it is expected to work.  Did you check out ghdl recently ?

1138-4EB, 2019-07-02:
I just tried `ubuntu:19.04` (LLVM 8.0.0) and `ubuntu:19.10` (LLVM 8.0.1) docker containers, and both of them work with latest `master`:

``` bash
# apt install git make gcc llvm-dev gnat
# llvm-config --version
# git clone https://github.com/ghdl/ghdl
# cd ghdl
# ./configure --prefix=/opt/Ghdl
```

Hence, @Defossez, you should check out the repo, as @tgingold suggests.

Defossez, 2019-07-03:
Hi,

Thanks for the fast reply!

It's a new PC everything is newly downloaded and installed on.
     Latest version of GHDL pulled for GitHub
     Latest download and install of GNAT from gnat.
     Latest version of LLVM/langc installed from apt.llvm.org

It doesn't seem a GHDL issue but rather a LLVM thing.
Can it be a path or ld_library_path issue?
The llvm apt page doesn't mention anything about paths, but then it 
doesn't mention anything about a 8.0.1 version too.

Thanks,

Marc

Defossez, 2019-07-03:
Hi,

I will uninstall/purge llvm installed from the llvm apt page.
Install it again from the default apt repositories, if that is already 
version 8, as you did.
Then try again.
And report back.

Regards,

Marc

Defossez, 2019-07-04:
Hi,

LLVM, I removed everything installed as instructed on the apt.llvm.org 
page per apt command line.
Checked if everything was removed using synaptic.
Rebooted the PC.

Then installed everything again using synaptic.
Went through the listed apt install commands fro the apt.llvm.org page.
Searched the files in synaptic and marked them one after the other for 
installation.
During this process synaptic automatically tick-marked necessary 
dependencies for some packages.
I then applied to install all indicated files and packages.

After this the configure command for GHDL works.

Why does it now work?
Don't really know but what I have seen is that installing this way 
installed extras that where indicated and tick-marked automatically by 
synaptic as necessary dependencies.

Maybe something is missing in the given apt command lines on apt.llvm.org

Anyway, it's working and this can be classified.

Thanks,

Marc

closed 2019-07-06 by tgingold