ghdl/ghdl #859: Journey installing GHDL on brand new Linux Mint 19 (Tessa) machine

Defossez, 2019-07-07:
Hi,

I opened issue #858 but got it solved so the issue is closed.
Here the description of installing GHDL on a brand new Mint 19 machine.
It might help others.

------
Brand new PC.
Installed latest version of Linux-Mint, Mint-19 (Tessa) with Cinnamon desktop.
After setting up the usual things as email and configuration of VPN, Firefox and other stuff, Install GHDL
- Go to https://github.com/ghdl/ghdl
- Clone or download latest version, save it to */Downloads*. 
  Unzip *ghdl-master.zip* into ghdl-master directory created under *Downloads*.
- Ghdl is written in ada, and following the Github page a ada compiler is required.
- Follow the given link to download ada compiler ([libre.adacore.com](http://libre.adacore.com/tools/gnat-gpl-edition/).) 
- Download the community version, press the big button on the we page,  and install it.
  - Make downloaded file executable
  - When install is run `./gnat-community-2019-20190517-x86_64-linux-bin` as normal user it wants to install Gnat in the home directory of the user.
  - Common unix/linux rules stipulate that tools need to be installed in */opt* but that is a system directory and thus under root control. 
  - Cancel the installer and rerun it now as root (sudo) 
    `sudo gnat-community-2019-20190517-x86_64-linux-bin`
  - Installer noticed the change and automatically installs Gnat in */opt* (/opt/Gnat)
  - Edit */home/<user>/.bash-aliases* or */home/<user>/.bash* and add */opt/Gnat/2019/bin* to the PATH.
  - Save, close editor and close used terminal for ada installation.
- Back to the web, browse further down the Ghdl github page and decide to install Ghdl using LLVM.
- Follow the rules and rtfm, thus click '[Building](http://ghdl.readthedocs.io/en/latest/building/Building.html)' at the bottom of the page.
- On readthedocs page find out that LLVM must be installed. Click the given link (top of page)
- llvm.org web page opens.
  - Click under [Download!] (bottom-left) [APT Packages]
  - Ending on a page with install instructions for Debian and Ubuntu. 
    Mint 19 is based on Ubuntu_18.04 thus that is OK.
  - Scroll down till Ubuntu square and copy the two, for Ubuntu-18.04 and LLVM-8, repository lines into a text editor. Save as */etc/apt/sources.list.d/llvm-clang-8.list* file.
  - Open a terminal and run: `sudo apt update`. 
    This should list all the available repositories including the added lines and report no errors.
  - Scroll down the opt.llvm.org page till the [Install (qualification Branch)] square.
  - Run one after the other the *wget* and *apt* commands provided here.
- Ada and LLVM are installed without issues, let’s install GHDL!
- Go back to the GHDL readthedocs page and click in the pro/con table the LLVM link.
  - LLVM is supported till version 5 and we installed version 8!
    - No panic, from installs on other computers I know it’s working with LLVM-7, thus it’ll work with LLVM-8 too.
  - Click the link [LLVM Backend on GNU/Linux with GCC/GNAT](https://ghdl.readthedocs.io/en/latest/building/llvm/GNULinux-GNAT.html)
  - Following the documentation the path to llvm-config is required.
  - Open *nemo* (Mint file browser) and search for llvm-config
    - It is found in `/usr/lib/llvm-8/bin` and as a link (llvm-config-8) in `/usr/bin`
- As with Gnat, Ghdl is going to be installed in */opt*
- Here we go for real:
  - Open a terminal
  - Change directory to unzipped GHDL: `cd ~/Downloads/ghdl-master`
  - Make a build directory: `mkdir build`
  -  Change into that build directory: `cd build`
  - Now these commands need to be run:
    - ```
      ../configure --with-llvm-config --prefix=PREFIX
      make
      make install
      ```
  - Run the first command: 
    `../configure –with-llvl-config=/usr/lib/llvm-8/bin –prefix=/opt/Ghdl`
    - Nearly immediately get in the terminal a message that is really an error, not from GHDL but from LLVM.
      ```
      Use full IEEE library
      Build machine is: x86_64-linux-gnu
      Unhandled version llvm 8.0.1
      ```
      But all test using clang-8 and llvm-8 work perfectly.
      Decided to remove and purge the LLVM installation and install it again.
      Install LLVM using Synaptic. Take back the *opt.llvm.org* page and install one after the other the packages mentioned there but now using Synaptic.
      Find out that some extra dependencies are installed this way!
  - Try again `../configure –with-llvl-config=/usr/lib/llvm-8/bin –prefix=/opt/Ghdl`
    Brilliant, it passes without problems.
    Up to the next command.
  - Run second command `make`
    It displays some comments and dies with a message that it cannot find *clang++*.
    ```
    make[1]: Entering directory '/home/beeb/Downloads/ghdl-master/build'
    clang++ -c `llvm-config-8 --cxxflags` -o llvm-cbindings.o ../src/ortho/llvm4-nodebug/llvm-cbindings.cpp
    /bin/sh: 1: clang++: not found
    ../src/ortho/llvm4-nodebug/Makefile:18: recipe for target 'llvm-cbindings.o' failed
    make[1]: *** [llvm-cbindings.o] Error 127
    ```
    - Open *nemo* (Mint file browser) and search for `clang++`
    - Indeed there is no clang++, but there is in */usr/bin* a link with name *clang++-8* pointing to */usr/lib/llvm-8/bin/clang++*.
    - Create and extra symbolic link: `ln -s /usr/bin/clang++-8 clang++`
      Do the same for *clang* since there is no link for it too.
  - Retry to run the command `make`
    Now it runs for a while displaying a lot of text in the terminal.
    Finally it end without problems.
  - Last step to take is *make install*
    GHDL is going to be installed in */opt/Ghdl* and */opt* is a system directory under root (sudo) control thus *make install* must be run as sudo else the installation will fail due to write permissions.
  - Run the command `sudo make install`
  - It starts of creating some directories for Ghdl.
    And the fails with the message is can not find *gnatmake*
    ```
    gnatmake -o ghdl1-llvm -aI../src/ortho/llvm4-nodebug -aI../src/ortho \
    -aI../src -aI../src/vhdl -aI../src/grt -aI../src/psl -aI../src/vhdl/translate -aI../src/ghdldrv -aI../src/ortho -aI../src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe ortho_code_main -bargs -E \
    -largs llvm-cbindings.o --LINK=clang++ \
     `llvm-config-8 --ldflags --libs --system-libs`
    /bin/sh: 1: gnatmake: not found
    ../src/ortho/llvm4-nodebug/Makefile:12: recipe for target 'ghdl1-llvm' failed
    make[1]: *** [ghdl1-llvm] Error 127
    make[1]: Leaving directory '/home/beeb/Downloads/ghdl-master/build'
    Makefile:334: recipe for target 'ghdl1-llvm' failed
    make: *** [ghdl1-llvm] Error 2
    ```
    Open a new terminal and type: *gnatmake*
    That results in a list of *gnatmake* options, being the proof Gnat is installed correctly and known (It’s add to the PATH, thus …).
  - Pondering for a while and examining what can be wrong makes me try this:
  - Change directory to */opt and `ls -la` shows that the Ghdl directory is owned by root and belong to the root group. That is how it should but then the *make install* command fails.
  - Make the user owner of the Ghdl directory and in one take drop the directory in the users group
     `sudo chown -R <user> /opt/Ghdl`
    `sudo chgrp -R <user> /opt/Ghdl`
  - Run *make install* again
    Process ends without errors or warnings.
  - GHDL is installed.
  - Add GHDL to the path.
    Edit */home/<user>/.bash-aliases* or */home/<user>/.bash* and add Ghdl to the PATH.
  - Test run is positive, now install GTKwave to be able to view something.
  - DONE!
------

tgingold, 2019-07-07:
I gnat available as a standard package in Mint ?  I suppose it would work with it.

1138-4EB, 2019-07-07:
@Defossez, thanks a lot for this very detailed report. Just some thoughts:

- As @tgingold suggested, all the required packages should be available through apt. According to https://github.com/ghdl/docker/blob/master/dockerfiles/build/ubuntu18 the following might suffice: `apt-get -y install --no-install-recommends gnat gcc make zlib1g-dev llvm-8-dev clang-8`. However, on some contexts, a user might want to get GNAT from AdaCore and LLVM from llvm.org, as you did.
- If all the dependencies are available in the path, [dist/travis/build.sh](https://github.com/ghdl/ghdl/blob/master/dist/travis/build.sh) can be used, instead of running `configure [...] && make && make install`. In this case, you would run `./dist/travis/build.sh -b llvm-8 --pkg mytarball.tgz`. Then, just `tar xzf mytarball.tgz -C /opt/ghdl`, or `mv install-llvm-8 /opt/ghdl`.

@tgingold, we need to update the list of supported LLVM versions in https://ghdl.readthedocs.io/en/latest/building/llvm/index.html#build-llvm and https://ghdl.readthedocs.io/en/latest/building/llvm/GNULinux-GNAT.html:

- Do we want to specify a list of supported versions at all?
- Is debugging still only supported with LLVM 3.5?

tgingold, 2019-07-07:
Yes, debugging is only supported with LLVM 3.5; but we should update the list of supported version in building/llvm/GNULinux-GNAT.html

tgingold, 2019-07-08:
I am closing this issue as there is no more action to do.  Feel free to complete it or to reference it.

closed 2019-07-08 by tgingold

Defossez, 2019-07-09:
Hoi,

I just wanted to see what happened when installing the tool on a brand 
new installed machine following whats written in the documentation. I 
think that somebody should bring the documentation in line with today's 
versions.

Maybe these days the main install procedure described in the 
documentation should not reference all the GNAT , LLVM an other stuff 
but just specify for Ubuntu based distributions the apt way and for 
other distributions another easy way. While the way as described now in 
the documentation could be mentioned for advance users.

My 0.02 euro

Marc

1138-4EB, 2019-07-09:
> Hoi, I just wanted to see what happened when installing the tool on a brand new installed machine following whats written in the documentation. I think that somebody should bring the documentation in line with today's versions.

We updated the list of supported LLVM versions:

- https://ghdl.readthedocs.io/en/latest/building/llvm/index.html#build-llvm
- https://ghdl.readthedocs.io/en/latest/building/llvm/GNULinux-GNAT.html

> Maybe these days the main install procedure described in the documentation should not reference all the GNAT, LLVM an other stuff but just specify for Ubuntu based distributions the apt way and for [...]

Please, do not hesitate to suggest any PR regarding this or other related enhancements.

> My 0.02 euro Marc

Thanks!