ghdl/ghdl #871: Support alias declarations in synthesis

pepijndevos, 2019-07-17:
Currently GHDL crashes on alias declarations. As discussed on Gitter, here is a simple test case.

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity pwmled is
  port (
    clk : in std_logic;
    rst : in std_logic;
    pwm_out : out std_logic);
end pwmled;

architecture impl of pwmled is
  signal counter : unsigned(15 downto 0);
  alias polarity : std_logic is counter(15);
  alias idx : unsigned(6 downto 0) is counter(6 downto 0);
  alias level : unsigned(6 downto 0) is counter(14 downto 8);
  signal pwm : std_logic;
begin
  pwm <= '1' when (idx > level) else '0';
  pwm_out <= polarity xor pwm;

  process(clk, rst)
  begin
    if(rst = '0') then
      counter <= (others => '0');
    elsif(rising_edge(clk)) then
      counter <= counter + 1;
    end if;
  end process;
end impl;
```

Xiretza, 2019-09-27:
I think this may have been fixed? `ghdl --synth testcase.vhd -e pwmled` produces synth output without any crashes on decc588e.

1138-4EB, 2019-09-30:
As @Xiretza commented, I can confirm that this seems to be fixed in master. @pepijndevos, please reopen if it does not work for you.

closed 2019-09-30 by 1138-4EB