ghdl/ghdl #874: GHDL crashes on vhdl-2008: to_string(real_number, "%0.2f");

wm2015email, 2019-07-19:
Here's a crashing testcase for you:

```
--FILE:./ghdl_crash.vhd

library ieee;
use     ieee.std_logic_1164.all;
use     ieee.math_real.all;

entity testbench is
    generic(
       c_clock_mhz :real    := 66.0
    );
end entity;

architecture rtl of testbench is
    constant c_ns_clock     :real    := (1.0/c_clock_mhz)*1000.0;
    constant c_ns_write     :real    := 10.0;
    constant c_ns_read      :real    := 25.0;

begin

-- synthesis translate_off
process 
begin
    report "sram_ctrl2   c_ns_clock    :" & to_string(c_ns_clock, "%0.2f");
    report "sram_ctrl2   c_ns_write    :" & to_string(c_ns_write, "%0.2f");
    report "sram_ctrl2   c_ns_read     :" & to_string(c_ns_read, "%0.2f");
	wait;
end process;
-- synthesis translate_on

end architecture;  
```

here's my ghdl command line:

```
ghdl.exe -a -frelaxed-rules --std=08 --ieee=synopsys --work=work ./ghdl_crash.vhd
```

here's my ghdl version:

```
$ ghdl -v
GHDL 0.35 (v0.35) [Dunoon edition]
 Compiled with GNAT Version: 5.3.0
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

I'm running GDHL on Windows 10.

wm2015email, 2019-07-19:
Here is the same code working in modelsim:

```
c:/modeltech64_10.4b/win64/vcom -2008 -work work ./sim_testcase_pkg.vhd
Model Technology ModelSim SE-64 vcom 10.4b Compiler 2015.05 May 27 2015
Start time: 14:48:51 on Jul 19,2019
vcom -2008 -work work ./sim_testcase_pkg.vhd
-- Loading package STANDARD
-- Compiling package sim_testcase_pkg
End time: 14:48:51 on Jul 19,2019, Elapsed time: 0:00:00
Errors: 0, Warnings: 0

c:/modeltech64_10.4b/win64/vcom -2008 -work work ./ghdl_crash.vhd
Model Technology ModelSim SE-64 vcom 10.4b Compiler 2015.05 May 27 2015
Start time: 14:48:51 on Jul 19,2019
vcom -2008 -work work ./ghdl_crash.vhd
-- Loading package STANDARD
-- Loading package TEXTIO
-- Loading package std_logic_1164
-- Loading package MATH_REAL
-- Compiling entity testbench
-- Compiling architecture rtl of testbench
End time: 14:48:52 on Jul 19,2019, Elapsed time: 0:00:01
Errors: 0, Warnings: 0
param: gse                            : false
param: testcase                       : testcase0
param: timeout                        : 0

c:/modeltech64_10.4b/win64/vsim -c testbench -do "log -r *; run -all; exit"
Reading C:/modeltech64_10.4b/tcl/vsim/pref.tcl

# 10.4b

# vsim -c testbench -do "log -r *; run -all; exit"
# Start time: 14:48:52 on Jul 19,2019
# ** Note: (vsim-3813) Design is being optimized due to module recompilation...
#
# //  ModelSim SE-64 10.4b May 27 2015
# //
# //  Copyright 1991-2015 Mentor Graphics Corporation
# //  All Rights Reserved.
# //
# //  THIS WORK CONTAINS TRADE SECRET AND PROPRIETARY INFORMATION
# //  WHICH IS THE PROPERTY OF MENTOR GRAPHICS CORPORATION OR ITS
# //  LICENSORS AND IS SUBJECT TO LICENSE TERMS.
# //  THIS DOCUMENT CONTAINS TRADE SECRETS AND COMMERCIAL OR FINANCIAL
# //  INFORMATION THAT ARE PRIVILEGED, CONFIDENTIAL, AND EXEMPT FROM
# //  DISCLOSURE UNDER THE FREEDOM OF INFORMATION ACT, 5 U.S.C. SECTION 552.
# //  FURTHERMORE, THIS INFORMATION IS PROHIBITED FROM DISCLOSURE UNDER
# //  THE TRADE SECRETS ACT, 18 U.S.C. SECTION 1905.
# //
# Loading std.standard
# Loading std.textio(body)
# Loading ieee.std_logic_1164(body)
# Loading ieee.math_real(body)
# Loading work.testbench(rtl)#1
# log -r *
# (vish-4014) No objects found matching '*'.
#  run -all
# ** Note: sram_ctrl2   c_ns_clock    :15.15
#    Time: 0 ns  Iteration: 0  Instance: /testbench
# ** Note: sram_ctrl2   c_ns_write    :10.00
#    Time: 0 ns  Iteration: 0  Instance: /testbench
# ** Note: sram_ctrl2   c_ns_read     :25.00
#    Time: 0 ns  Iteration: 0  Instance: /testbench
#  exit
# End time: 14:48:53 on Jul 19,2019, Elapsed time: 0:00:01
# Errors: 1, Warnings: 0

tgingold, 2019-07-24:
This is not yet handled when the arguments are static.

commit 980dc567 (2021-09-17): vhdl-evaluation: implement to_string for real with format.  Fix #874
closed 2021-09-17 by tgingold