ghdl/ghdl #875: Assertion Assert Failure : sem.adb:3301

ericschdt, 2019-07-20:
**Description**
ghdl crashes when I try to check the syntax of the following incomplete test file `test.hdl`.

**Context**
```
******************** GHDL Bug occurred ***************************                               
Please report this bug on https://github.com/ghdl/ghdl/issues                                    
GHDL release: 0.36 (v0.36) [Dunoon edition]                                                      
Compiled with GNAT Version: 9.1.0                                                                
Target: x86_64-pc-linux-gnu                                                                      
In directory: ...       
Command line:               
ghdl -s test.hdl                                                                                   
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised                                                
Exception information:                                                                           
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : sem.adb:3301                                           
Call stack traceback locations:                                                                  
0x55dbd7fac2a2 0x55dbd7f52437 0x55dbd7f230a5 0x55dbd7f602b4 0x55dbd7f60408 0x55dbd7f604b3 0x55dbd
7f0c90c 0x55dbd7f96996 0x55dbd7e879b8 0x7f891c686ee1 0x55dbd7e86c9c 0xfffffffffffffffe           
******************************************************************
```

#### System
```
Operating System: Manjaro Linux 
KDE Plasma Version: 5.16.3
KDE Frameworks Version: 5.60.0
Qt Version: 5.13.0
Kernel Version: 5.2.1-1-MANJARO
OS Type: 64-bit
```

#### Origin
  - AUR: `https://aur.archlinux.org/packages/ghdl/`

**How to reproduce?**

`ghdl -s test.hdl`
```(VHDL)
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use STD.textio.all;
use ieee.std_logic_textio.all;

entity example_file_io_tb is
end example_file_io_tb;

architecture behave of example_file_io_tb is

	------------------------------------------------------------------------------
	-- Declare the Component under Test
	------------------------------------------------------------------------------
	component module_ripple_carrier_adder is
		generic (
			g_WIDTH: natural);
		port (
			i_add_term1 : in  std_logic_vector(g_WIDTH-1 downto 0);
			i_add_term2 : in  std_logic_vector(g_WIDTH-1 downto 0);
			o_result	: out std_logic_vector(g_WIDTH downto 0);
			);
	end component module_ripple_carrier_adder;

	------------------------------------------------------------------------------
	-- Testbench Internal Signals
	------------------------------------------------------------------------------
	file file_VECTOR : text;
	file file_RESULT: text;

	constant c_WIDTH : natural := 4;

	signal r_ADD_TERM1 : std_logic_vector(c_WIDTH-1 downto 0) := (others => '0');
	signal r_ADD_TERM2 : std_logic_vector(c_WIDTH-1 downto 0) := (others => '0');
	signal w_SUM 	   : std_logic_vector(c_WIDTH downto 0);

begin
	------------------------------------------------------------------------------
	-- Instantiate and Map UUT
	------------------------------------------------------------------------------
	MODULE_RIPPLE_CARRY_ADDER_INST : module_ripple_carry_adder
		generic map (
			g_WIDTH => c_WIDTH)
		port map (
			i_addr_term1 => r_ADD_TERM1,
			i_addr_term2 => r_ADD_TERM2,
			o_result => w_SUM
			);

	------------------------------------------------------------------------------
	-- This procedure reads the file input_vectors.txt which is located in the 
	-- simulation project area. 
	-- It will read the data in and send it to the ripple carry adder component
	-- to perform the operations. The result is written to the
	-- output_results.txt file, located in the same directory.
	------------------------------------------------------------------------------

	begin
		file_open(file_VECTORS, "input_vectors.txt", read_mode);
		file_open(file_RESULTS, "output_results.txt", write_mode);
		
		while not endfile(file_VECTORS) loop
			readline(file_VECTORS, v_ILINE);
			read(v_ILINE, v_ADD_TERM1);
			read(v_ILINE, v_SPACE);
			read(v_ILINE, V_ADD_TERM2);
			
			-- Pass the variable to a signal to allow the ripple-carry to use it
			r_ADD_TERM1 <= v_ADD_TERM1;
			r_ADD_TERM2 <= v_ADD_TERM2;

			wait for 60 ns;
			
			write(v_OLINE, w_SUM, right, c_WIDTH);
			writeline(file_RESULTS, v_OLINE);
		end loop;

		file_close(file_VECTORS);
		file_close(file_RESULTS);

		wait;
	end process;

end behave;
```

tgingold, 2019-07-21:
The current version doesn't crash anymore.  You now get:

```
test.vhdl:21:80:error: extra ';' at end of interface list
test.vhdl:58:8:error: unexpected token 'begin' in a concurrent statement list
test.vhdl:62:16:error: sequential statement only allowed in processes
test.vhdl:82:11:error: missing ";" at end of architecture
test.vhdl:84:1:error: missing entity, architecture, package or configuration
```

ericschdt, 2019-07-21:
> The current version doesn't crash anymore.  You now get:

Which commit/s has/have resolved the issue?

I just tried the AUR-Git packages and received:

```
test.hdl:1:1:error: missing entity, architecture, package or configuration
test.hdl:1:31:error: character '$' can only be used in strings or comments
test.hdl:4:45:error: an identifier is expected instead of 'configuration'
test.hdl:4:45:error: 'is' is expected instead of 'configuration'
test.hdl:4:45:error: 'end' is expected instead of 'configuration'
test.hdl:4:44:error: missing ";" at end of package declaration
test.hdl:4:59:error: an identifier is expected instead of 'file'
(...)
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-486-g44a19c53) [Dunoon edition]
Compiled with GNAT Version: 9.1.0
Target: x86_64-pc-linux-gnu
/path/
Command line:
ghdl -s test.hdl
Exception STORAGE_ERROR raised
Exception information:
raised STORAGE_ERROR : stack overflow or erroneous memory access
Call stack traceback locations:
0x7f10d1ba87de 0x55c54a5e03b1 0x55c54a660d9d 0x55c54a6b077a 0x55c54a6b289e 0x55c54a683565 0x55c54a6c0d5b 0x55c54a6c0ead 0x55c54a6c0f8c 0x55c54a67e53b 0x55c54a6f6bc8 0x55c54a5d6a2e 0x7f10d1b94ee1 0x55c54a5d5c6c 0xfffffffffffffffe
******************************************************************
```

closed 2019-07-22 by tgingold

1138-4EB, 2019-07-22:
> Which commit/s has/have resolved the issue?
> 
> I just tried the AUR-Git package and received:

It must be one of the latest 7 commits, because:

``` bash
root@55a1cc0ef37e:~# ls
test.hdl
root@55a1cc0ef37e:~# ghdl --version
GHDL 0.37-dev (v0.36-493-g654af3d6) [Dunoon edition]
 Compiled with GNAT Version: 8.3.0
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2019 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
root@55a1cc0ef37e:~# ghdl -a test.hdl 
test.hdl:21:80:error: extra ';' at end of interface list
test.hdl:58:8:error: unexpected token 'begin' in a concurrent statement list
test.hdl:62:16:error: sequential statement only allowed in processes
test.hdl:82:11:error: missing ";" at end of architecture
test.hdl:84:1:error: missing entity, architecture, package or configuration
root@55a1cc0ef37e:~# 
```

commit 66652290 (2019-07-22): vhdl: add testcase.

Close #875