ghdl/ghdl #877: Compilation error on discrete subtype indication as discrete range

go2sh, 2019-07-29:
**Description**
The LRM specifies that, slice has a discrete range as argument.
```
slice_name ::= prefix ( discrete_range )
```
A discrete range can be a range or a discrete subtype indication. 
```
discrete_range ::= discrete_subtype_indication | range
```

So the following code should (tm) be legal:
```
package aa is
constant a : integer_vector(0 to 1) := (0, 1);
constant b : integer_vector(0 to 0) := a(integer range 0 to 0)
end package aa;
```

**Expected behaviour**
Code gets analyzed.

**Context**
```
test1.vhd:3:49:error: ')' is expected instead of 'range'
test1.vhd:3:49:error: missing ";" at end of object declaration
ghdl:error: compilation error
```

- OS: Linux (Arch LLVM Backend)
- Origin:
  - [ ] Package manager. Repo:
  - [ ] Released binaries. Tarball:
  - [X] Built from sources. Commit SHA: 2068fb5a

**Checklist**
Before submitting your issue, please review the following checklist:

- [ ] Add `GHDL Bug occurred` log block
- [ ] Add a MWE
- [x] Try the latest version

go2sh, 2019-07-29:
More Context which makes me think this is allowed: (LRM 5.3.2.1)
```
The direction of a discrete range is the same as the direction of the range or the discrete subtype indication
that defines the discrete range. If a subtype indication appears as a discrete range, the subtype indication
shall not contain a resolution indication.
```

tgingold, 2019-07-30:
Wow, are you looking for each possible mistake ?
It looks like you are right, but I have never seen anybody use that notation for slice names!

go2sh, 2019-07-30:
I don't want to bother you with those kinds of things, I just post them here for completeness and to document some of the "awesome" language features. I don't have a use-case for this nor do I expect you to fix this, feel free to mark it as "won't fix". I can totally understand that.

I'am writing an advanced formating tool for VHDL, which does some kind of parsing of the design file. This is just one thing, I stumbled on while writing an combined expression parser. Since this special case felt so awkward, I tested it with ghdl and modelsim. ;-) And the result is documented here.

`discrete_range` is also used in for-loops, aggregates and in subtype indications... So you can nest them on level like:
```
subtype array_sub is array_type(integer range 0 to 5);
```

tgingold, 2019-08-11:
Now fixed.

closed 2019-08-11 by tgingold