ghdl/ghdl #881: GHDL wrongly reports negative timeout clause

obruendl, 2019-08-02:
**Description**
GHDL crashes when simulating my code. It tells me that there is a negative timeout clause but the clause is not negative. That the clause is not negative can be deduced from the code itself as well as from the fact that the simulation runs fine on Modelsim.

**Expected behaviour**
Simulation should run without errors.

**Context**
C:\Program Files (x86)\GHDL\bin\ghdl.exe:error: negative timeout clause at D:/github/psi_fpga_all/VivadoIp/vivadoIP_clock_measure/tb/single_clock_measurement_tb.vhd line 97
C:\Program Files (x86)\GHDL\bin\ghdl.exe:error: simulation failed
child process exited abnormally

- OS: Win10 / 64 bit

**Additional context**
The line that fails looks as follows:
"wait for Period_v/2"
Period_v is a variable defined in a process. It is calculated:
"Period_v := (1 sec) / HzTest;"
where HzTest is a signal that changes its value during the simulation but is always positive.

**How to reproduce?**
1. Checkout this repo: https://github.com/paulscherrerinstitute/psi_fpga_all
  use "git checkout --recurse-submodules https://github.com/paulscherrerinstitute/psi_fpga_all"
2. Modify the file psi_fpga_all/VivadoIp/vivadoIP_clock_measure/sim/run.tcl to use GHDL:
  change line 11 to "psi::sim::init -ghdl"
3. open a TCL shell in psi_fpga_all/VivadoIp/vivadoIP_clock_measure/sim
4. execute "source ./run.tcl"

**Files**
https://github.com/paulscherrerinstitute/psi_fpga_all

**Checklist**
Before submitting your issue, please review the following checklist:

tgingold, 2019-08-02:
Interesting.  You are using HzTest <= 0.0001,
so Period would be 10_000 sec, which overflows the 64 bit time type.

It works if the 'Zero' period is 0.001
Unfortunately, you cannot reduce the time resolution because of textio.  I will fix that.

obruendl, 2019-08-05:
Okay, I will change the period to 0.001 as workaround, so we can continue using GHDL until it is fixed.

Thank you for the quick and precise response. This is well appreciated :-)

tgingold, 2019-08-08:
I think this is now fixed.  When using the mcode backend you can now set the time resolution either to auto `--time-resolution=auto` or to a larger time unit (`--time-resolution=ps`).

closed 2019-08-08 by tgingold

1138-4EB, 2019-09-16:
@tgingold, is it possible to have this feature with other backends? I'm asking for documentation purposes.

tgingold, 2019-09-16:
No, because you need a global analysis to automatically set the resolution.