ghdl/ghdl #883: Build failure with sid-gcc-9.1.0

sjaeckel, 2019-08-06:
**Description**
VUnit `vhdl/verification_components/src/axi_stream_monitor.vhd` can't be compiled as soon as coverage is enabled.

**Expected behaviour**
That the compilation also works with coverage enabled.

**Context**

```
...
Compiling into vunit_lib: ../usr/local/lib/python3.7/dist-packages/vunit/vhdl/verification_components/src/axi_stream_monitor.vhd          failed
=== Command used: ===
/usr/local/bin/ghdl -a --workdir=/work/vunit_out/ghdl/libraries/vunit_lib --work=vunit_lib --std=08 -P/work/vunit_out/ghdl/libraries/vunit_lib -P/work/vunit_out/ghdl/libraries/osvvm -P/work/vunit_out/ghdl/libraries/t -g -fprofile-arcs -ftest-coverage /usr/local/lib/python3.7/dist-packages/vunit/vhdl/verification_components/src/axi_stream_monitor.vhd

=== Command output: ===
during IPA pass: profile
/usr/local/lib/python3.7/dist-packages/vunit/vhdl/verification_components/src/axi_stream_monitor.vhd: In function ‘vunit_lib__axi_stream_monitor__ARCH__a__axi_stream_protocol_checker_generate__B1__axi_stream_protocol_checker_inst__COMP_ELAB’:
/usr/local/lib/python3.7/dist-packages/vunit/vhdl/verification_components/src/axi_stream_monitor.vhd:35: internal compiler error: in coverage_begin_function, at coverage.c:656
   35 |     variable axi_stream_transaction : axi_stream_transaction_t(
      | 
0x609efc coverage_begin_function(unsigned int, unsigned int)
        ../../gcc-srcs/gcc/coverage.c:458
0xac5731 branch_prob(bool)
        ../../gcc-srcs/gcc/profile.c:1233
0xbff182 tree_profiling
        ../../gcc-srcs/gcc/tree-profile.c:793
0xbff182 execute
        ../../gcc-srcs/gcc/tree-profile.c:898
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <https://gcc.gnu.org/bugs/> for instructions.
/usr/local/bin/ghdl: exec error
```
- OS: 
  - not relevant (can be provided if required)
  - Docker version 19.03.1, build 74b1e89e8a

- Origin:
  - Docker image **sid-gcc-9.1.0** built on "August 5, 2019 07:46 AM"

**Additional context**
This does not happen with the **buster-gcc-7.4.0** image

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE), that is compatible with [issue-runner](https://github.com/1138-4EB/issue-runner). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```
#>> run.py 
from os.path import join, dirname
from vunit import VUnit

ui = VUnit.from_argv()

ui.add_osvvm()
ui.add_verification_components()

ui.add_source_files_from_csv(join(dirname(__file__), "f.csv"))

ui.set_sim_option("enable_coverage", True)
ui.set_compile_option("ghdl.flags",["-g", "-fprofile-arcs", "-ftest-coverage"])
ui.set_sim_option("ghdl.elab_flags",["-Wl,-lgcov", "-Wl,--coverage"])

# Run vunit function
ui.main()

#>> f.csv 
t,t.vhd

#>> t.vhd 
library IEEE;
use IEEE.STD_LOGIC_1164.all;
use IEEE.NUMERIC_STD.all;

library vunit_lib;
context vunit_lib.vunit_context;
context vunit_lib.vc_context;

library osvvm;
use osvvm.RandomPkg.all;

entity tb_t is
        generic(
                runner_cfg : string);
end entity tb_t;

architecture tb of tb_t is
begin
        main : process
        begin
                test_runner_setup(runner, runner_cfg);

                while test_suite loop
                        if run("test1") then
                        end if;
                end loop;

                test_runner_cleanup(runner);
                wait;
        end process;
        test_runner_watchdog(runner, 100 ms);

end architecture tb;

#>> Dockerfile
FROM ghdl/ghdl:sid-gcc-9.1.0

#---

RUN apt-get update -qq \
 && apt-get -y install libgnat-8 gcc libc6-dev curl python3 python3-pip lcov git vim \
 && apt-get autoclean -y && apt-get clean -y && apt-get autoremove -y \
 && pip3 install vunit_hdl

RUN mkdir /work && chmod 777 /work

#>> run.sh
docker build -t test/ghdl:sid-gcc-9.1.0 .
docker run -v /$(pwd)://work --rm -it -w //work test/ghdl:sid-gcc-9.1.0 sh -c 'VUNIT_SIMULATOR=ghdl; python3 run.py -p 8 ;'

#>> end
```

_I hope that's all correct :-)_

**Files**
none

**Checklist**
Before submitting your issue, please review the following checklist:

- [ ] Add `GHDL Bug occurred` log block
- [x] Add a MWE
- [x] Try the latest version

I'm not sure where the issue lies, probably also gcc, but I thought I'd start the report here.

1138-4EB, 2019-08-06:
@sjaeckel, thanks for reporting this issue. This is a rather complex issue, because your use case includes multiple tools (GHDL, VUnit, GCC, Docker). So, let's go step by step.

The title seems misleading to me, because I understand 'build' as compiling GHDL itself. However, this issue is about an error that is produced when analyzing a VHDL file. Would you mind renaming it so this is explicit?

> VUnit `vhdl/verification_components/src/axi_stream_monitor.vhd` can't be compiled as soon as coverage is enabled.

Same as above, let's better use 'analysis' or 'elaboration', even though we know that the target is to build/compile and executable binary.

> `/usr/local/bin/ghdl -a --workdir=/work/vunit_out/ghdl/libraries/vunit_lib --work=vunit_lib --std=08 -P/work/vunit_out/ghdl/libraries/vunit_lib -P/work/vunit_out/ghdl/libraries/osvvm [...]

Is the error reproducible if we analyse this file isolated? I.e., copy the file to a separate dir and just `/usr/local/bin/ghdl -a --work=vunit_lib --std=08 -g -fprofile-arcs -ftest-coverage axi_stream_monitor.vhd`.

> OS: not relevant (can be provided if required)

This refers to the environment where GHDL is being executed, not the host. Hence, the OS is GNU/Linux, precisely Debian, since we do not provide windows containers/images. As you say, the host is not relevant when docker is used.

> This does not happen with the buster-gcc-7.4.0 image

This might be meaningful. Currently, we build images with five different versions of GCC (https://travis-ci.com/ghdl/docker/builds/121904609). It'd be interesting to try it with 8.1.0 and/or 8.2.0.

> How to reproduce?

In this case, I think that using the CSV file is not required. It should be equivalent to replace `ui.add_source_files_from_csv(join(dirname(__file__), "f.csv"))` with `vu.add_library('t').add_source_files(join(dirname(__file__), 't.vhd'))`.

> I hope that's all correct :-)

It looks so. But we can probably simplify it a little more:

- OSVVM seems not to be used. Is the error reproducible when we remove it? Note that we should also remove `ui.add_osvvm()` from `run.py`.
- Since there is a single test, we can get rid of the `while test_suite` loop.

```
library IEEE;
use IEEE.STD_LOGIC_1164.all;
use IEEE.NUMERIC_STD.all;

library vunit_lib;
context vunit_lib.vunit_context;
context vunit_lib.vc_context;

entity tb_t is
        generic( runner_cfg : string );
end entity tb_t;

architecture tb of tb_t is
begin
        main : process
        begin
                test_runner_setup(runner, runner_cfg);
				info("HELLO");
                test_runner_cleanup(runner);
                wait;
        end process;
        test_runner_watchdog(runner, 100 ms);
end architecture tb;
```

> #>> Dockerfile

As a side note. Currently, we do not provide any image with VUnit/Gtkwave and GHDL with GCC backend. There is `ghdl/ext:vunit-gtkwave`, which includes GHDL with mcode backend; and `ghdl/ext:ls-vunit-gtkwave`, which includes GHDL with LLVM and `--default-pic`. Would you be interested if we provided a third image (e.g. `ghdl/ext:coverage`) which includes GHDL with GCC backend and VUnit, and optionally GtkWave?

I ask it because I don't know if the Dockerfile you provided is some portion of a larger Dockerfile that you use internally/privately. Should you be a potential user of an upstream image, please let us know:

- Do you want the 'latest' GHDL and VUnit from their respective `master` branches, or the latest stable release of each of them?
- Do you want GtkWave included or do you prefer o more lightweight image without GUI dependencies?
- Is Debian Buster a good base image for you or do you use Fedora/Ubuntu?

> apt-get -y install libgnat-8 gcc libc6-dev curl python3 python3-pip lcov git vim

`libgnat-8`, `gcc` and `libc6-dev` should already be installed in `ghdl/ghdl:sid-gcc-9.1.0`, because these are runtime dependencies for GHDL (see https://github.com/ghdl/docker/blob/master/dockerfiles/run_debian). If you really need to install them again, something might be wrong in our build process.

> I'm not sure where the issue lies, probably also gcc, but I thought I'd start the report here.

We need to start somewhere, so this is ok. As commented above, let's try to narrow the MWE and that will let us know where to ask next.

1138-4EB, 2019-08-06:
I can confirm that the error is only reproducible with `sid-gcc-9.1.0`. All others (`stretch-gcc-5.5.0`, `buster-gcc-7.4.0`, `fedora28-gcc-8.1.0` and `fedora29-gcc-8.2.0`) work ok. Hence, this might be an issue with GCC 9. I don't know if @tgingold has checked it yet. I know that GCC 8 is required for some synth features, but no idea about GCC 9.

@sjaeckel, I did the tests with the three changes suggested above. So, it is up to you to update the first comment.

Replying to myself:

> Is the error reproducible if we analyse this file isolated?

No. In order to analyze the file many other VUnit sources are required.

> If you really need to install them again, something might be wrong in our build process.

They are not required:

``` bash
winpty docker run --rm -itv /$(pwd)://work -w //work ghdl/ghdl:sid-gcc-9.1.0 sh -c 'apt-get update -qq; apt-get -y install python3 python3-pip lcov; pip3 install vunit_hdl; VUNIT_SIMULATOR=ghdl; python3 run.py -p 8 ;'
winpty docker run --rm -itv /$(pwd)://work -w //work ghdl/ghdl:stretch-gcc-5.5.0 sh -c 'apt-get update -qq; apt-get -y install python3 python3-pip lcov; pip3 install vunit_hdl; VUNIT_SIMULATOR=ghdl; python3 run.py -p 8 ;'
winpty docker run --rm -itv /$(pwd)://work -w //work ghdl/ghdl:buster-gcc-7.4.0 sh -c 'apt-get update -qq; apt-get -y install python3 python3-pip lcov; pip3 install vunit_hdl; VUNIT_SIMULATOR=ghdl; python3 run.py -p 8 ;'
winpty docker run --rm -itv /$(pwd)://work -w //work ghdl/ghdl:fedora28-gcc-8.1.0 sh -c 'dnf update -y; dnf install -y python3 python3-pip lcov; pip3 install vunit_hdl; VUNIT_SIMULATOR=ghdl; python3 run.py -p 8 ;'
winpty docker run --rm -itv /$(pwd)://work -w //work ghdl/ghdl:fedora29-gcc-8.2.0 sh -c 'dnf update -y; dnf install -y python3 python3-pip lcov; pip3 install vunit_hdl; VUNIT_SIMULATOR=ghdl; python3 run.py -p 8 ;'
```

sjaeckel, 2019-08-07:
@1138-4EB thanks for your time, further analysis and improvement of the initial issue.

I've updated the title, description and vunit testsuite as proposed.

Regarding your questions:

> Would you be interested if we provided a third image (e.g. `ghdl/ext:coverage`) which includes GHDL with GCC backend and VUnit, and optionally GtkWave?

I would be very interested in such an image ;-) 

> * Do you want the 'latest' GHDL and VUnit from their respective `master` branches, or the latest stable release of each of them?

I'd say the latest stable release should be what is used in production, but sometimes it would also be handy to have master available.

> * Do you want GtkWave included or do you prefer o more lightweight image without GUI dependencies?

I personally don't need GtkWave yet as I'm using GHDL and VUnit only in CI, but that might also change in the future.

> * Is Debian Buster a good base image for you or do you use Fedora/Ubuntu?

Debian Buster is perfect for me.

> I ask it because I don't know if the Dockerfile you provided is some portion of a larger Dockerfile that you use internally/privately.

This Dockerfile is the entire file, I started by using ghdl/ext:vunit in production but now I also want coverage so I had to merge the relevant parts of ghdl-with-gcc and ghdl/ext:vunit.

> `libgnat-8`, `gcc` and `libc6-dev` should already be installed in `ghdl/ghdl:sid-gcc-9.1.0`, because these are runtime dependencies for GHDL (see [...]

I didn't check if they are really required, so it's very possible that this is not the case.

If I can help with anything, please feel free to drop me another comment!

sjaeckel, 2019-08-07:
> > Would you be interested if we provided a third image (e.g. `ghdl/ext:coverage`) which includes GHDL with GCC backend and VUnit, and optionally GtkWave?
> 
> I would be very interested in such an image ;-)

and btw. I'd happily also test it!

1138-4EB, 2019-08-21:
Hi again @sjaeckel! Regarding new docker images for CI/coverage, please have a look at ghdl/docker#11. Let's keep this 'clean' for possible GCC 9 support issues.

@tgingold, whenever you look at this, please ignore all the noise about docker. See first paragraph in [this](https://github.com/ghdl/ghdl/issues/883#issuecomment-518815957) comment above.

tgingold, 2021-09-14:
I think this is now fixed.

closed 2021-09-14 by tgingold

wfjm, 2022-07-09:
I use a ghdl 2.0.0 under Ubuntu 20.04LTS build with gcc 9.3.0 (default gcc used in Ub 20.04).  
All works perfect, but when I try to build a model with coverage enabled I get
```
    during IPA pass: profile
      .../e~tb_w11a_n4: In function
        ‘work__tbcore_rlink__ARCH__sim__DEFAULT_CONFIG__simclkcnt’:
      .../e~tb_w11a_n4:28: internal compiler error:
        in coverage_begin_function, at coverage.c:656
          0x60fdbe coverage_begin_function(unsigned int, unsigned int)
                  ../../../gcc-9.3.0/gcc/coverage.c:656
          0xb0e5ec branch_prob(bool)
                  ../../../gcc-9.3.0/gcc/profile.c:1233
          0xc51912 tree_profiling
                  ../../../gcc-9.3.0/gcc/tree-profile.c:793
          0xc51912 execute
                  ../../../gcc-9.3.0/gcc/tree-profile.c:898
```

Thus exactly the type of error that triggered this issue.  
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=100788 states that this is fixed in gcc-10 and gcc-11.  
So I re-build ghdl with gcc 11.2.0 (the default gcc in Ub 22.04 LTS).  
And still get the same error
```
    during IPA pass: profile
      .../e~tb_w11a_n4:
        In function ‘work__tbcore_rlink__ARCH__sim__DEFAULT_CONFIG__simclkcnt’:
      .../e~tb_w11a_n4:28: internal compiler error:
        in coverage_begin_function, at coverage.c:662
          0x680226 coverage_begin_function(unsigned int, unsigned int)
            ../../../gcc-11.2.0/gcc/coverage.c:662
          0xbeb148 branch_prob(bool)
            ../../../gcc-11.2.0/gcc/profile.c:1328
          0xd3e5e2 tree_profiling
            ../../../gcc-11.2.0/gcc/tree-profile.c:782
          0xd3e5e2 execute
            ../../../gcc-11.2.0/gcc/tree-profile.c:888
```
The line numbers are different, and path names show that I picked the new build.

Question: How to avoid this supposedly solved problem ?

tgingold, 2022-07-10:
I think this is the issue that has been fixed in #1817.

wfjm, 2022-07-13:
I'm using 2.0.0, thus commit 8185ed8 from 2022-03-01.  
The issue #1817 was apparently closed with 51d8774 on 2021-07-23.  
And as one would expect, 51d8774 is included in 8185ed8.  
So in theory all should be fine with 2.0.0.  
It seems that the fixes done for #1817 didn't resolve the full problem.  
I even don't understand the structure of the problem.  
From https://gcc.gnu.org/bugzilla/show_bug.cgi?id=100788 I took that it's a gcc 9 issue, and solved for gcc 10.  
From #1817 I took that ghdl was modified too. Where that workarounds for a still luring problem in gcc 10+ ?

tgingold, 2022-07-13:
Yes you're right.  I have to investigate what is happening with gcc 11.2

tgingold, 2022-07-13:
Also, can you reproduce the issue with the testcase from 1817 ?

wfjm, 2022-07-14:
I've built a gcc 2.0.0 with gcc 10.2.0, see same problem for a simple unit test bench.  
And that should definitively work.  
I've to look into my model build procedures and check from where -lgcov is loaded.