ghdl/ghdl #884: Merge libghdlvpi and/or libghdlsynth into libghdl

1138-4EB, 2019-08-07:
Currently, `libghdl` is built with GHDL by default (unless `--disable-libghdl` is used). `libghdlvpi.so`, is also built by default.

Moreover, in order to use ghdlsynth-beta as a module, `libghdlsynth.so` needs to be built from this repo. Last, `ghdl.so` is built, which is the interface to yosys.

As a result, when ghdlsynth-beta is used, the user finds up to four shared libraries generated by GHDL in `GHDL_PREFIX/lib`. For example:

```
drwxr-xr-x 3 root root 4.0K Aug  7 20:59 .
drwxr-xr-x 5 root root 4.0K Aug  7 20:59 ..
drwxr-xr-x 8 root root 4.0K Aug  7 20:59 ghdl
-rwxr-xr-x 1 root root 1.3M Aug  7 20:59 ghdl.so
-rwxr-xr-x 1 root root 4.3M Aug  7 20:59 libghdl-0_37_dev.so
-rwxr-xr-x 1 root root 5.1M Aug  7 20:59 libghdlsynth.so
-rwxr-xr-x 1 root root  17K Aug  7 20:59 libghdlvpi.so
```

**Describe the solution you'd like**

- Since `libghdl` and `libghdlvpi` are built by default, merge the later into the former.
- If possible, merge `libghdlsynth` into `libghdl` when `--enable-synth` is used; leave the functionality out otherwise.
    - Currently, `--enable-synth` is used to enable command/option `ghdl --synth`, which is functionally similar to `libghdlsynth` but the implementation is different. Therefore, it might be better to add a new independent switch (say `--enable-synthlib`) to include the funciontality in `libghdl`.

**Additional context**

Although the suggested procedure in `ghdlsynth-beta` is to install `ghdl.so` in `YOSYS_PREFIX/share/yosys/plugins/ghdl.so`, I think that having it installed along with GHDL is a use case that we should support. See `install.ghdl` in tgingold/ghdlsynth-beta#31.

1138-4EB, 2019-08-27:
This changed in https://github.com/ghdl/ghdl/commit/7e91b1419cde2a00c9288c02cda256a054ed6a3f. Now, `libghdlsynth.so` is merged into `libghdl-0_37_dev.so` if ghdl is configured with `--enable-libghdl --enable-synth`.

@tgingold, could you please clarify these contexts:

- Since libghdl is enabled by default, configuring with only `--enable-synth`, is equivalent to `--enable-libghdl --enable-synth`.
- `--enable-synth --disable-libghdl` provides `ghdl --synth` only.
- It is not possible to add synthesis support to libghdl only, without enabling `ghdl --synth` at the same time.
- Make target `libghdlsynth.so` still exists, which a kind-of subset of libghdl for synthesis. This might be useful for users/developers that want a shared library for synthesis only (without building libghdl). Is this a suggested workflow? Or is it deprecated/for internal purposes only?

Moreover, what about libghdlvpi? Does it make sense it merge it into libghdl? Otherwise, I think we can close this issue...

tgingold, 2019-08-28:
> 	• Since libghdl is enabled by default, configuring with only --enable-synth, is equivalent to --enable-libghdl --enable-synth.

Correct.

> 	• --enable-synth --disable-libghdl provides ghdl --synth only.

Correct.

> 	• It is not possible to add synthesis support to libghdl only, without enabling ghdl --synth at the same time.

Correct.

> 	• Make target libghdlsynth.so still exists, which a kind-of subset of libghdl for synthesis. This might be useful for users/developers that want a shared library for synthesis only (without building libghdl). Is this a suggested workflow? Or is it deprecated/for internal purposes only?

It will be deprecated.  Maybe we should keep libghdlsynth.a or create libghdl.a

> Moreover, what about libghdlvpi? Does it make sense it merge it into libghdl? Otherwise, I think we can close this issue...

No, it is fully independent from libghdl.  So I think you can close this issue.

tgingold, 2019-08-28:
That case: to build yosys with ghdl but without external shared libraries.

tgingold, 2019-08-28:
Yes, you always need an installation of ghdl (at least for the libraries).  So I am not sure that using the .a is really an advantage.

closed 2019-08-28 by 1138-4EB

1138-4EB, 2019-08-28:
Thanks!

> It will be deprecated. Maybe we should keep libghdlsynth.a or create libghdl.a

In case of keeping one, I think it makes sense for it to be `libghdl.a`, in coherence with shared variants.

However, since https://github.com/tgingold/ghdlsynth-beta/commit/d9ce5d06ecefc5cd3c3a8cbd6cb77afcee5185e9#diff-04c6e90faac2675aa89e2176d2eec7d8, `libghdlsynth.a` seems not to be used anywhere. Unless it is implicitly built when ghdlsynth-beta is built as part of yosys... What is the expected use case?

1138-4EB, 2019-08-28:
> That case: to build yosys with ghdl but without external shared libraries.

That's what was removed in https://github.com/tgingold/ghdlsynth-beta/commit/d9ce5d06ecefc5cd3c3a8cbd6cb77afcee5185e9. Previously, instructions suggested to execute `make libghdlsynth.a` and `make install.libghdlsynth`. Now, instructions say `Get and build ghdl as in the previous section` (see https://github.com/tgingold/ghdlsynth-beta#build-as-part-of-yosys-not-recommended).

Are there two variants of 'Build as part of yosys' (one with shared ghdl libraries and another one with static libraries)? If that is the case, I think we should explain how to build the latter.

Moreover, we know that, in practice, a working GHDL installation is a runtime dependency of ghdlsynth as a module. I believe that this is still true, even if `libghdlsynth.a`/`libghdl.a` is used to build ghdlsynth as part of yosys. Hence, is there any technical advantage in using `*.a` libs instead of shared ones? E.g., should it be faster?