ghdl/ghdl #886: vhdl-sem_names.adb

geekskick, 2019-08-09:
**Description**
Static assertion failure during analysis.

**Expected behaviour**
Expecting meaningful error or passing analysis

**Context**
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-133-g3d528d7c) [Dunoon edition]
Compiled with GNAT Version: Community 2018 (20180523-73)
Target: x86_64-apple-darwin16.7.0
In directory: <...path...>/pong/src/
Command line:
ghdl -a -v --workdir=<...path...>/pong/work paddle_mover.vhd
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : vhdl-sem_names.adb:4222
Load address: 0x103570000
Call stack traceback locations:
0x10387ae71 0x10376618a 0x103779d96 0x10377b46e 0x10378309e 0x103760ff9 0x103764f52 0x10377b234 0x10377c372 0x103776b81 0x103779b7a 0x10377c2f7 0x10377d20f 0x10377e2d5 0x10378a6a8 0x103791ba4 0x10374be1b 0x1037a9710 0x103703968 0x1038628ee 0x10357313c
******************************************************************
- OS: macOS
- Origin:
  - [ ] Package manager. Repo:
  - [ ] Released binaries. Tarball:
  - [X] Built from sources. Commit SHA: g3d528d7c

**How to reproduce?**
```
#>> ent.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

package pong_pkg is
    subtype short is signed(16 downto 0);

    type direction is (UP, DOWN, NONE);
    type dimension is record
        w : short;
        h : short;
    end record;

    constant SCREEN_SIZE : dimension := (   w => to_signed(640, short'length), 
                                            h => to_signed(480, short'length) );

    type location is record
        r : short;
        c : short;
    end record;    
    
    type paddle is record
        loc : location;
        dir : direction;
    end record;    

    component paddle_mover is
        generic(
            paddle_size : dimension;
            screen_size : dimension;
            reset_loc   : location
        );
        port(
            clk : in std_logic;
            en  : in std_logic;
            rst : in std_logic;
            dir : in direction;
            q   : in paddle;
            d   : out paddle
        );
    end component;
end package;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

use work.pong_pkg.all;

entity paddle_mover is
generic(
    paddle_size : dimension := (w => to_signed(1, short'length), h => to_signed(1 short'length));
    screen_size : dimension := (w => to_signed(1, short'length), h => to_signed(1 short'length));
    reset_loc   : location  := (r => to_signed(1, short'length), c => to_signed(1 short'length))
);
port(
    clk : in std_logic;
    en  : in std_logic;
    rst : in std_logic;
    dir : in direction;
    q   : in paddle;
    d   : out paddle
);
end entity;

architecture beh of paddle_mover is
        
    constant velocity   : short     := to_signed(1, short'length);

    signal next_candidate :paddle   := (dir => NONE, loc => (others => 1));
    signal next_paddle  : paddle    := (dir => NONE, loc => (others => 1));
    signal next_moves   : std_logic := '0';
    signal off_bottom   : std_logic := '0';
    signal off_top      : std_logic := '0';
begin
   
    next_candidate.dir <= q.dir;
    next_candidate.loc.c <= q.loc.c;

    next_candidate.loc.r <= q.loc.r + velocity when dir = DOWN else 
                            q.loc.r - velocity when dir = UP else 
                            q.loc.r;
    
    off_bottom <= (next_candidate.loc.r + paddle_size.h) >= screen_size.h;
    off_top    <= next_candidate.loc.r <= to_unsigned(0, short'length);

    next_moves <= off_bottom nor off_top;

    next_paddle.dir   <= next_candidate.dir;
    next_paddle.loc.c  <= next_candidate.loc.c;
    next_paddle.loc.r  <= next_candidate.loc.r when next_moves = '1' else q.loc.r;
    
    process(clk)
    begin
        if rising_edge(clk) then
            if rst = '1' then
                d.loc <= reset_loc;
            elsif en = '1' then
                d <= next_paddle;
            end if;
        end if;
    end process;      
end architecture; 

#>> sim.sh
ghdl -a ent.vhd

#>> run.sh
docker run --rm -tv /$(pwd):/src:z -w //src ghdl/ghdl:buster-mcode sh -c ./sim.sh

#>> end
```

**Checklist**
Before submitting your issue, please review the following checklist:

- [X] Add `GHDL Bug occurred` log block
- [X] Add a MWE
- [X] Try the latest version

tgingold, 2019-08-10:
The issue is missing comma between `1` and `short'length`.  I am fixing it.

diogratia, 2019-08-10:
The errors causing the GHDL Bug occurred report are the lack of separators in the parameter association lists for the to_signed calls in generic clause:

```vhdl
generic (
    paddle_size : dimension := (w => to_signed(1, short'length), h => to_signed(1, short'length));
    screen_size : dimension := (w => to_signed(1, short'length), h => to_signed(1, short'length));
    reset_loc   : location  := (r => to_signed(1, short'length), c => to_signed(1, short'length))
);
```
These are lexical errors. (The missing commas have been added above).

There are semantic errors in the signal declarations for next_candidate and next_paddle initial values. There may be further errors.

commit ffd18264 (2019-08-11): vhdl: avoid crash on incorrect unit name.

Fix #886
closed 2019-08-11 by tgingold