ghdl/ghdl #887: GHDL crash at run start

johosc, 2019-08-09:
**Description**
Trying to simulate a block making use of ieee fixed point library (VHDL 2008 enabled for both analysis and run steps). When issuing the "ghdl -r" it crashes immediatly with a crash report, context below.

**Expected behaviour**
Simulation should start, but crashes instead.

**Context**
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36 (v0.36) [Dunoon edition]
Compiled with GNAT Version: 7.3.0
Target: i686-w64-mingw32
In directory: C:\cygwin64\home\johan.oscarsson\dev\vhdl\filt\sim\
Command line:
C:\ghdl-0.36\bin\ghdl.exe -r --std=08 -P../lib --workdir=../lib --work=tbFiltLib tb
Exception ORTHO_CODE.SYNTAX_ERROR raised
Exception information:
raised ORTHO_CODE.SYNTAX_ERROR : ortho_code-exprs.adb:451
Call stack traceback locations:
0x6888dd 0x6889af 0x6895f1 0x674b26 0x673621 0x678861 0x638eb6 0x63a478 0x63ad5d 0x634007 0x630544 0x630cda 0x630abc 0x630d6d 0x630abc 0x630dd1 0x6963ce 0x60bffe 0x609896 0x6149d2 0x699569 0x5f2674 0x589224 0x69b94f 0x402635 0x401389 0x763f6357 0x77947a92 0x77947a62
******************************************************************

- OS: Windows/cywin
- Origin:
  - [ ] Package manager. Repo:
  - [ x] Released binaries. Tarball: ghdl-0.36-mingw32-mcode.zip
  - [ ] Built from sources. Commit SHA:

**Additional context**

**How to reproduce?**
No MWE yet, will try to create one

tgingold, 2019-08-10:
We don't really need a minimal reproducer, but a standalone one is welcome.

Maybe this issue has been fixed since 0.36

johosc, 2019-08-13:
I have managed to make a standalone example of the problem. It seems to be related to having unconstrained arrays of u_sfixed as input to a component. I get the following elaboration error when trying to run the design:

C:\ghdl-0.36\bin\ghdl.exe:error: NULL access dereferenced
  from: [unknown caller]
  from: work.filter(rtl).STMT_ELAB at test.vhd:38
  from: work.tb(test).filter_1.CMP_ELAB at test.vhd:28
  from: work.tb(test).DECL_ELAB at test.vhd:67
  from: Elaboration of design
C:\ghdl-0.36\bin\ghdl.exe:error: error during elaboration

The code, test.vhd:
`
---------------------------------------------------------------------------
-- A package defining a type (unconstrained array of unconstrained vectors)
---------------------------------------------------------------------------
library ieee;
use ieee.fixed_float_types.all;
use ieee.fixed_pkg.all;

package filtPkg is
  type u_sfixed_array is array (integer range <>) of u_sfixed;
end package filtPkg;

-------------------------------------------------------------------
-- A component using the array type for input (still unconstrained)
-------------------------------------------------------------------
library ieee;
use ieee.std_logic_1164.all;
use ieee.fixed_float_types.all;
use ieee.fixed_pkg.all;

use work.filtPkg.all;

entity filter is
  generic(
    intbits_g : natural;
    decbits_g : natural);
  port(
    clk   : in  std_ulogic;
    c     : in  u_sfixed_array);         -- filter coefficient
end entity filter;

architecture rtl of filter is

  -- Test signal to check assignement from the input array
  signal tmp : u_sfixed(intbits_g-1 downto -decbits_g);
  
begin

  tmp <= c(0);  -- This does not work, c does not seem to exist (NULL access dereferenced)

end architecture rtl;

------------------------------------
-- A top instantiating the component
------------------------------------
library ieee;
use ieee.std_logic_1164.all;
use ieee.fixed_float_types.all;
use ieee.fixed_pkg.all;

use work.filtPkg.all;

entity tb is
end entity tb;

architecture test of tb is
  constant intbits_c : natural := 4; -- Number of integer bits
  constant decbits_c : natural := 6; -- Number of decimal bits

  -- For simulation purpose
  signal clk : std_ulogic := '0';
  signal done : boolean := false;

  -- Input data to the component, here is where the size is constrained
  signal coeff : u_sfixed_array(0 to 5)(intbits_c-1 downto -decbits_c) := (others => (others => '0'));

  -- A test signal to check assignment from previous array
  signal tmp : u_sfixed(intbits_c-1 downto -decbits_c);

begin
  -- Clock generation
  clk  <= transport not clk after 10 ns when not done else '0';
  done <= transport true after 500 ns;
  
  tmp <= coeff(0); -- This works, I guess range is well defined here

  -- Component instantiation
  filter_1 : entity work.filter
    generic map (
      intbits_g => intbits_c,
      decbits_g => decbits_c)
    port map (
      clk   => clk,
      c     => coeff);

end architecture test;
`

Build and run:
`
ghdl -a --std=08 test.vhd

ghdl -r --std=08 tb
`

johosc, 2019-08-13:
I have also made a version where my input array type is an array of std_logic_vector. In this case I don't get any error:

test2.vhd:
`
---------------------------------------------------------------------------
-- A package defining a type (unconstrained array of unconstrained vectors)
---------------------------------------------------------------------------
library ieee;
use ieee.std_logic_1164.all;

package filtPkg2 is
  type std_logic_vector_array is array (integer range <>) of std_logic_vector;
end package filtPkg2;

-------------------------------------------------------------------
-- A component using the array type for input (still unconstrained)
-------------------------------------------------------------------
library ieee;
use ieee.std_logic_1164.all;

use work.filtPkg2.all;

entity filter2 is
  generic(
    high_g : natural;
    low_g  : natural);
  port(
    clk : in  std_ulogic;
    c   : in  std_logic_vector_array);         -- filter coefficient
end entity filter2;

architecture rtl of filter2 is

  -- Test signal to check assignement from the input array
  signal tmp : std_logic_vector(high_g downto low_g);
  
begin

  tmp <= c(0);  -- This works.

end architecture rtl;

------------------------------------
-- A top instantiating the component
------------------------------------
library ieee;
use ieee.std_logic_1164.all;

use work.filtPkg2.all;

entity tb2 is
end entity tb2;

architecture test of tb2 is
  constant high_c : natural := 7;
  constant low_c  : natural := 0;

  -- For simulation purpose
  signal clk  : std_ulogic := '0';
  signal done : boolean := false;

  -- Input data to the component, here is where the size is constrained
  signal coeff : std_logic_vector_array(0 to 5)(high_c downto low_c) := (others => (others => '0'));

  -- A test signal to check assignment from previous array
  signal tmp : std_logic_vector(high_c downto low_c);

begin
  -- Clock generation
  clk  <= transport not clk after 10 ns when not done else '0';
  done <= transport true after 500 ns;
  
  tmp <= coeff(0); -- This works

  -- Component instantiation
  filter2_1 : entity work.filter2
    generic map (
      high_g => high_c,
      low_g  => low_c)
    port map (
      clk   => clk,
      c     => coeff);

end architecture test;
`

Build and run:
`
ghdl -a --std=08 test2.vhd

ghdl -r --std=08 tb2
`

tgingold, 2019-08-14:
I don't get a crash for tb2, so I suppose this issue was partially fixed after release 0.36 (but not completely!)

johosc, 2019-08-14:
That is the same as me, I get a crash for tb but not for tb2. So it seems like the type used for the array matters (u_sfixed vs std_ulogic_vector).

tgingold, 2020-08-07:
Look fixed.

closed 2020-08-07 by tgingold