ghdl/ghdl #888: Fedora packaging: upstream patches, copr and coordination

1138-4EB, 2019-08-09:
While having a look at Fedora packaging for ghdl (https://src.fedoraproject.org/rpms/ghdl/tree/master), I found some patches that are not required anymore, and some other which are suitable to be upstreamed. Precisely:

- There is a patch for LLVM8 (ghdl-llvm8.patch), which is already supported by latest ghdl.
- There is a patch file (`ieee-mathreal.patch`) that is not used anymore.
- ghdlver is incorrectly set to 0.35dev; it should be updated to 0.37dev.
- A dummy UPF package is added (`upf.patch`), which was proposed in #436, but it is not merged yet.

On the one hand, I am preparing a PR to fix all these issues at once: https://github.com/1138-4EB/rpm-ghdl/commits/fix

On the other hand, there are 12 patches related to GCC 9 that are likely to be interesting/useful in order to add support for it here. Precisely, I think we are hitting some problem with GCC 9 in #883.

---

Moreover, since Fedora's forge supports opening 'Remote PRs' (e.g. from GitHub), I think it would be interesting to move or fork 1138-4EB/rpm-ghdl to ghdl/rpm. This would allow us to keep a better coordination.

---

We can also use copr (https://copr.fedorainfracloud.org/) to build RPM packages for Fedora rawhide, Fedora latest (30) and Fedora previous (29); for aarch64, i386 and x86_64. Potentially, other archs (ppc64le) and platforms can be added to (epel-6, epel-7, mageia-6, mageia-7, mageia-8, opensuse leap, opensuse tumbleweed...).

Note that the objective is not to replace our current CI jobs at ghdl/docker. This is specially so because our container-based CI jobs are very optimized and they take around 35'-40' instead of 45'-120'. However, in the long term, we might want to distribute 'nightly' RPMs through https://copr.fedorainfracloud.org/coprs/ghdl/ghdl, as a complement to 'stable' official releases; instead of the release tarballs for Fedora that we publish now.

Overall, I think that copr is a good playground to try the integration of ghdl, yosys and ghdlsynth, as commented in tgingold/ghdlsynth-beta#31. I.e., we can create non-official packages for ghdl and yosys, and play with different approaches to provide ghdlsynth.

---

/cc @tsailer

tgingold, 2019-08-11:
I doubt that the gcc patches fix the crash.  They appear to be small changes for configuration.

But yes, It would be nice to improve the coordination with rpm and other integrations.

sharkcz, 2019-08-23:
Hi, yes, having nightly builds in copr would be great. FYI - the current Fedora rpm builds and runs just fine on ppc64le. Don't hesitate to ask, I'm sure there are people (like me) in the Fedora community willing to help.

1138-4EB, 2019-08-23:
@sharkcz, thanks for stepping by!

> FYI - the current Fedora rpm builds and runs just fine on ppc64le

Do you know if [this](https://src.fedoraproject.org/rpms/ghdl/blob/master/f/ghdl-ppc64abort.patch) patch is still needed? According to [this](https://src.fedoraproject.org/rpms/ghdl/blob/master/f/ghdl.spec#L163) comment, it was submitted to https://gna.org/bugs/index.php?13390 10 years ago, but the site is now gone...

> Don't hesitate to ask, I'm sure there are people (like me) in the Fedora community willing to help.

Actually, I'd be glad if you could help me. Currently, this branch is successfully built: https://github.com/1138-4EB/rpm-ghdl/commits/fix However, `ghdlgitver` needs to be updated to any commit after https://github.com/ghdl/ghdl/commit/b814dbe04aa1658dc294b7db1f9005c497205f25 I've had a hard time trying to do so. I created a new tarball and I updated the `sources` file and `ghdlgitrev` in the spec file:

```
mkdir ghdl
curl -fsSL https://github.com/ghdl/ghdl/archive/f977ba0dd5b152e97619ecfe1d848d83f2e062ff.tar.gz | tar xzf - -C ghdl --strip-components=1
tar cvJf ghdl20190820gitf977ba0d.tar.bz2 --exclude-vcs ghdl
sha512sum ghdl20190820gitf977ba0d.tar.bz2
```

However, I am unable to have the new `.tar.bz2` pushed to copr. `fedpkg copr-build ghdl` from `master` of [1138-4EB/rpm-ghdl](https://github.com/1138-4EB/rpm-ghdl) is successful, so the credentials are properly set up. Moreover, I could successfully upload the tarball two weeks ago. I don't know what I'm doing wrong now, but I remember it took me a time to guess the command back then.

BTW, do you know how to update `%global DATE` and/or `Release: 5%{ghdlgitrev}.0%{?dist}`? I.e., is it necessary to update all the tarballs at once, so there is a single date? Or is it allowed to update a tarball only?

tsailer, 2019-08-23:
> Do you know if [this](https://src.fedoraproject.org/rpms/ghdl/blob/master/f/ghdl-ppc64abort.patch) patch is still needed? According to [...]

I added this patch to address a build failure on ppc64 (and I copied it from somewhere I don't remember). If it builds without - fine, but I haven't tried lately...

sharkcz, 2019-08-23:
/me tries a build without the patch ...

1138-4EB, 2019-08-23:
> Moreover, I could successfully upload the tarball two weeks ago. I don't know what I'm doing wrong now, but I remember it took me a time to guess the command back then.

Never mind. The tarball had to be `ghdl.20190820gitf977ba0d.tar.bz2` instead of `ghdl20190820gitf977ba0d.tar.bz2`. I'm testing it now...

sharkcz, 2019-08-23:
`spectool` from `rpmdevtools` package does some spec file parsing, so it can download the SourceX archives for example. But don't think it's possible with variable that are commented out.

sharkcz, 2019-08-23:
> /me tries a build without the patch ...

uff, it still fails build when compiling `src/std/textio.v87`
````
/home/dan/projects/fedora/ghdl/gcc-9.1.1-20190503/obj-ppc64le-fedora_ghdl-linux/gcc//ghdl -a --GHDL1=/home/dan/projects/fedora/ghdl/gcc-9.1.1-20190503/obj-ppc64le-fedora_ghdl-linux/gcc//ghdl1 --std=87 --bootstrap --work=std ../../src/std/textio.v87
during RTL pass: final
/home/dan/projects/fedora/ghdl/gcc-9.1.1-20190503/ghdl/lib/ghdl/std/v87/../../src/std/textio.v87: In function 'std__textio__ELAB_SPEC':
/home/dan/projects/fedora/ghdl/gcc-9.1.1-20190503/ghdl/lib/ghdl/std/v87/../../src/std/textio.v87:1: internal compiler error: in rs6000_output_function_epilogue, at config/rs6000/rs6000.c:29165
    1 | --  Std.Textio package declaration.  This file is part of GHDL.
      | 
0x10c589ef rs6000_output_function_epilogue
	../../gcc/config/rs6000/rs6000.c:29165
0x10527a37 final_end_function()
	../../gcc/final.c:1887
0x1052d413 rest_of_handle_final
	../../gcc/final.c:4667
0x1052d413 execute
	../../gcc/final.c:4737
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <http://bugzilla.redhat.com/bugzilla> for instructions.
/home/dan/projects/fedora/ghdl/gcc-9.1.1-20190503/obj-ppc64le-fedora_ghdl-linux/gcc//ghdl:error: exec error
make[1]: *** [libraries/Makefile.inc:157: lib/ghdl/std/v87/std-obj87.cf] Error 1
make[1]: Leaving directory '/home/dan/projects/fedora/ghdl/gcc-9.1.1-20190503/ghdl'
make: *** [Makefile:255: libs.vhdl.gcc] Error 2
````

sharkcz, 2019-08-23:
For the record in gcc 10 it's in `gcc/config/rs6000/rs6000-logue.c`, the `rs6000.c` got split into multiple files.

1138-4EB, 2019-08-23:
> > /me tries a build without the patch ...
> 
> uff, it still fails build when compiling `src/std/textio.v87`

Thanks for trying! Then, I'll keep the patch, since I'm preparing a PR of minor fixes.

Precisely, I think that https://github.com/1138-4EB/rpm-ghdl/commits/update is ready, since builds for 29, 30 and rawhide pass on i386, x86_64, aarch64 and ppc64le. Shall I open an external PR in https://src.fedoraproject.org/rpms/ghdl/tree/master or will any of you pick them?

sharkcz, 2019-08-23:
I have made a slightly better patch, more in line with the current logic in `rs6000_output_function_epilogue()`, testing now.
````
diff -up gcc/config/rs6000/rs6000.c.orig gcc/config/rs6000/rs6000.c
--- gcc/config/rs6000/rs6000.c.orig	2019-08-23 13:21:47.463294109 +0200
+++ gcc/config/rs6000/rs6000.c	2019-08-23 13:22:10.990328958 +0200
@@ -29147,7 +29147,8 @@ rs6000_output_function_epilogue (FILE *f
 	  || ! strcmp (language_string, "GNU GIMPLE")
 	  || ! strcmp (language_string, "GNU Go")
 	  || ! strcmp (language_string, "GNU D")
-	  || ! strcmp (language_string, "libgccjit"))
+	  || ! strcmp (language_string, "libgccjit")
+	  || ! strcmp (language_string, "vhdl"))
 	i = 0;
       else if (! strcmp (language_string, "GNU F77")
 	       || lang_GNU_Fortran ())
````

sharkcz, 2019-08-23:
And when talking about packaging and Fedora, I've met a strange build error
````
BUILDSTDERR: gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -g -gnata -I- /builddir/build/BUILD/gcc-9.1.1-20190503/ghdl-llvm/src/vhdl/translate/trans_decls.ads
BUILDSTDERR: gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -g -gnata -I- /builddir/build/BUILD/gcc-9.1.1-20190503/ghdl-llvm/src/vhdl/vhdl-sem_assocs.adb
BUILDSTDERR: gnatmake: "vhdl-sem_assocs.ali" incompatible ALI file, please recompile
BUILDSTDERR: gnatmake: "/builddir/build/BUILD/gcc-9.1.1-20190503/ghdl-llvm/src/vhdl/vhdl-sem_assocs.adb" compilation error
BUILDSTDERR: make: *** [Makefile:318: ghdl_llvm] Error 4
````
first in F-31 on x86_64 (https://koji.fedoraproject.org/koji/taskinfo?taskID=37232860) and a moment ago in local rebuild in F-30 on ppc64le. Sounds like a parallel make issue.

1138-4EB, 2019-08-23:
Once I guessed how to automate updating the ghdl tarball, I created a workflow in GitHub Actions that runs `fedpkg local` in a `fedora` docker container: https://github.com/1138-4EB/rpm-ghdl/actions It clones the latest `master` from ghdl/ghdl, and it generates the ghdlgitver variable through `git log` and `git rev-parse`. Each run takes about 1h, so it is not samething to be executed multiple times each day. However, it is significantly faster that copr, because a single architecture and version are tested. Hence, we might set this to be executed one or twice a week. It will hopefully allow us to:

- Detect architecture-independent issues before going to copr/fedora. This allows packagers/maintainers of `fedora` to focus on specific issues regarding their multi-platform and multi-version support approach.
- Provide `srpm` and/or `rpm` packages for amd64, which include all the three backends; instead of providing two tarballs (one for mcode and another one for llvm) only: https://github.com/ghdl/ghdl/releases.
- Provide 'nightly' `rpm` packages. GitHub Actions allow to collect the artifacts of each job, without explicitly pushing them anywhere. I have to guess how to define the list of objects to be preserved yet, tho.

Precisely, by checking [this log](https://github.com/1138-4EB/rpm-ghdl/commit/ad175d89acab8550935f964555920a1e7297d71e/checks), I found that rpms generated by https://github.com/1138-4EB/rpm-ghdl/commits/update are not correct. Some files are not included in the packages:

```
2019-08-23T17:26:22.0799392Z warning: Installed (but unpackaged) file(s) found:
2019-08-23T17:26:22.0800661Z    /usr/bin/gcov-dump
2019-08-23T17:26:22.0801038Z    /usr/bin/x86_64-fedora_ghdl-linux-gcc-9
2019-08-23T17:26:22.0801198Z    /usr/include/ghdlsynth.h
2019-08-23T17:26:22.0801487Z    /usr/include/ghdlsynth_gates.h
2019-08-23T17:26:22.0801849Z    /usr/lib/debug/usr/bin/gcov-dump-0.37dev-5.20190820gitf977ba0d.0.fc32.x86_64.debug
2019-08-23T17:26:22.0802314Z    /usr/lib/debug/usr/bin/x86_64-fedora_ghdl-linux-gcc-9-0.37dev-5.20190820gitf977ba0d.0.fc32.x86_64.debug
2019-08-23T17:26:22.0802635Z    /usr/lib/debug/usr/lib/libghdl-0_37_dev.so-0.37dev-5.20190820gitf977ba0d.0.fc32.x86_64.debug
2019-08-23T17:26:22.0802897Z    /usr/lib/libghdl-0_37_dev.so
```

This is surprising to me, because all the copr builds passed, for all the platforms! I'd argue that the testing procedure is partially broken if packages are not tested in an environment with runtime dependencies only. Nonetheless, at least three of the files do need to be added to the package (`ghdlsynth.h`, `ghdlsynth_gates.h` and `libghdl-0_37_dev.so`). I am not sure about `gcov-dump` and `x86_64-fedora_ghdl-linux-gcc-9`.

> I have made a slightly better patch

Is it available in any public git repo? I'd rather cherry-pick a commit, if possible.

> And when talking about packaging and Fedora, I've met a strange (and transient) build error

I've seen that quite frequently in the jobs we have in Travis. I found it to be an issue also when building ghdl inside a container running with kata-containers runtime (which are docker containers on a QEMU KVM). So, on the one hand make is not the best tool for parallel builds. On the other hand, I don't know why this is more common on fedora than in debian-based containers. Nevertheless, I also found that debian-based containers (specially ubuntu) have more frequent connection issues than fedora. I guess there is little to be done here.

sharkcz, 2019-08-23:
The "unpackaged files" feature is taken from the gcc spec file, there is a variable in rpm to allow that (https://src.fedoraproject.org/rpms/gcc/blob/master/f/gcc.spec#_10). I'll look into it, but we will likely need a -devel subpackage for the headers.

One little thing I found is that the `tar` command from the spec generates an xz archive of ghdl with the `-J` option, rather than bz2.

tgingold, 2019-08-23:
Yes, do not use parallel make.  I will investigate if we can do something better than no parallel.

1138-4EB, 2019-08-23:
> The "unpackaged files" feature is taken from the gcc spec file, there is a variable in rpm to allow that (https://src.fedoraproject.org/rpms/gcc/blob/master/f/gcc.spec#_10).

Thanks. I saw that it is unset (`%global _unpackaged_files_terminate_build 0`) in `ghdl.spec`.

> I'll look into it, but we will likely need a -devel subpackage for the headers.

Let's ask @tgingold, but I think that you are correct, since those headers are currently only used when [tgingold/ghdlsynth-beta](https://github.com/tgingold/ghdlsynth-beta) is built.

> One little thing I found is that the `tar` command from the spec generates an xz archive of ghdl with the `-J` option, rather than bz2. And maybe using `git archive` would be more convenient, for example in form `date=20190820; hash=f977ba0d; git archive --format=tar --prefix=ghdl/ $hash | xz > ../ghdl.${date}git${hash}.tar.xz`

`git archive` is used in the comments for `nvptx-tools` and `nvptx-newlib`, indeed:

```spec
# The source for this package was pulled from upstream's vcs.  Use the
# following commands to generate the tarball:
# GCC_TARNAME="gcc-%%{version}-%%{DATE}"
# svn export svn://gcc.gnu.org/svn/gcc/branches/redhat/gcc-9-branch@%%{SVNREV} $GCC_TARNAME
# tar cf - $GCC_TARNAME | xz -9e > $GCC_TARNAME.tar.xz
Source0: gcc-%{gcc_version}-%{DATE}.tar.xz
# The source for nvptx-tools package was pulled from upstream's vcs.  Use the
# following commands to generate the tarball:
# git clone https://github.com/MentorEmbedded/nvptx-tools.git
# cd nvptx-tools
# git archive origin/master --prefix=nvptx-tools-%%{nvptx_tools_gitrev}/ | xz -9e > ../nvptx-tools-%%{nvptx_tools_gitrev}.tar.xz
# cd ..; rm -rf nvptx-tools
Source1: nvptx-tools-%{nvptx_tools_gitrev}.tar.xz
# The source for nvptx-newlib package was pulled from upstream's vcs.  Use the
# following commands to generate the tarball:
# git clone https://github.com/MentorEmbedded/nvptx-newlib.git
# cd nvptx-newlib
# git archive origin/master --prefix=nvptx-newlib-%%{nvptx_newlib_gitrev}/ | xz -9 > ../nvptx-newlib-%%{nvptx_newlib_gitrev}.tar.xz
# cd ..; rm -rf nvptx-newlib
Source2: nvptx-newlib-%{nvptx_newlib_gitrev}.tar.xz
```

I'd move all these comment to a separate `update_tarballs.sh` script. wdyt?

This is how I do it for GHDL:

```
git clone --depth=1 https://github.com/ghdl/ghdl
cd ghdl

GHDL_SHA="$(git rev-parse --short=8 --verify HEAD)"
GHDL_GITVER="${GHDL_SHA}git$(git rev-parse --short=8 --verify HEAD)"

git archive --format=tar --prefix=ghdl/ "$GHDL_SHA" | xz > "../ghdl.${GHDL_GITVER}.tar.xz"

cd ..
rm -rf ghdl

sed -i.bak \
"s/.*ghdl.*/$(sha512sum ghdl.${GHDL_GITVER}.tar.bz2 | sed 's/\(.*\) \(.*\)/SHA512 (\2) = \1/g')/g" \
sources

sed -i.bak \
's/\(%global ghdlgitrev\).*/\1 .'"${GHDL_GITVER}"'/g' \
ghdl.spec
```

BTW, I tried to parse it with `spectool`, but it does not work.

1138-4EB, 2019-08-23:
> Yes, do not use parallel make. I will investigate if we can do something better than no parallel.

Should we change https://github.com/ghdl/ghdl/blob/master/dist/travis/build.sh#L126?

tgingold, 2019-08-24:
I have never seen a build failure on travis due to parallel.  Maybe nproc is 1.
I think this is an issue only for llvm builds.

sharkcz, 2019-08-24:
I have sent my spec file updates via https://github.com/1138-4EB/rpm-ghdl/pull/1
No uninstalled files are reported, confirmed it compiles the microwatt core.

tgingold, 2019-08-24:
Great!

1138-4EB, 2019-08-24:
> I have never seen a build failure on travis due to parallel. Maybe nproc is 1. I think this is an issue only for llvm builds.

I normally restart them if I see that errors are transitient. Unless you explicitly wait until the tests are executed, it might be my fault to be hiding this from you. Do you want me not to restart jobs in ghdl/ghdl?

> I think this is an issue only for llvm builds.

I think so. Errors in mcode builds are normally related to connection issues.

---

> I have sent my spec file updates via 1138-4EB/rpm-ghdl#1

Thanks! I rebased them. The run was successful, and no unpackaged files are reported: https://github.com/1138-4EB/rpm-ghdl/commit/bcba79a267f055a863fefa5a9e3e4a9879e03215/checks

I saw that you removed `gcov-*` binaries. GHDL supports three backends, but only one of them supports code coverage (GCC). In order to use it `lcov` is required. On the one hand, I don't know if these binaries have anything to do with it. On the other hand, would it make sense to define `lcov` as a dependency of GHDL (when GCC backend is used)?

> No uninstalled files are reported, confirmed it compiles the https://github.com/antonblanchard/microwatt core.

off-topic (ref antonblanchard/microwatt#1): did you try to synthesize microwatt with GHDL? I expect it to fail, but it might be a surprise.

---

Pending tasks before requesting a merge of [1138-4EB/rpm-ghdl@update](https://github.com/1138-4EB/rpm-ghdl/commits/update) into https://src.fedoraproject.org/rpms/ghdl/:

- [ ] update `%global DATE`?
- [ ] update `Release: 5%{ghdlgitrev}.0%{?dist}`?
- [ ] update changelog
- [ ] add `lcov` as a dependency of ghdl (GCC)?
- [ ] use `git archive .. | xz ...`
  - create `update_tarballs.sh`?
- [ ] set `%global _unpackaged_files_terminate_build 1`?

As a general comment, I am not sure about being correct to make package `ghdl` default to GCC backend on fedora. I think that `ghdl-gcc` should exist, and `ghdl` should be reference to one of `ghdl-gcc`, `ghdl-llvm` or `ghdl-mcode`. Unless code coverage is required, mcode should be the default, since it is much more lightweight and faster. If cross-platform support is a priority, LLVM should be the default. Nevertheless, I'd suggest not to use `ghdl`, but to require the user to choose a backend explicitly. @tsailer, @sharkcz, do you know whether this has been discussed anywhere?

sharkcz, 2019-08-27:
some updates from me
- removal of `gcov*` - the only part of the gcc build tree that's distributed is the vhdl backend, there was one `gcov-*` tool not removed, which I fixed. The gcov tools are available via `Requires: gcc`, meaning everything the `gcc` package contains is always available when `ghdl` is installed
- unless `lcov` is part of the regular user workflows, I would leave it on the user to install it and not introduced any dependency on it
- we can drop `_unpackaged_files_terminate_build` redefine to avoid any packaging bugs, now all the unneeded gcc files should be removed correctly
- regarding the variants - gcc/llvm/mcode - I think we can use the `alternatives` mechanism which presets a symlink to `ghdl` and allows the user to change it per their preference
- when building with the synth feature enabled I get a missing symbol error during linking (it's on ppc64le)
````
make[2]: Leaving directory '/home/dan/projects/fedora/ghdl/gcc-9.1.1-20190503/obj-ppc64le-fedora_ghdl-linux/libcpp'
make[2]: Entering directory '/home/dan/projects/fedora/ghdl/gcc-9.1.1-20190503/obj-ppc64le-fedora_ghdl-linux/libdecnumber'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/dan/projects/fedora/ghdl/gcc-9.1.1-20190503/obj-ppc64le-fedora_ghdl-linux/libdecnumber'
make[2]: Entering directory '/home/dan/projects/fedora/ghdl/gcc-9.1.1-20190503/obj-ppc64le-fedora_ghdl-linux/gcc'
gnatmake -c -aI../../gcc/vhdl --subdirs=vhdl ortho_gcc-main \
 -cargs -O2 -g -Wall -Wformat-security -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -mcpu=power8 -mtune=power8 -fasynchronous-unwind-tables -fstack-clash-protection -gnata -gnaty3befhkmr -gnatwa
gnatmake: objects up to date.
gnatmake -o ghdl1 -aI../../gcc/vhdl -aOvhdl ortho_gcc-main \
 -bargs -E -cargs -O2 -g -Wall -Wformat-security -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -mcpu=power8 -mtune=power8 -fasynchronous-unwind-tables -fstack-clash-protection -gnata -gnaty3befhkmr -gnatwa \
 -largs --LINK=g++ -no-pie   -O2 -g -Wformat -Wformat-security -Wp,-D_GLIBCXX_ASSERTIONS -fstack-protector-strong -grecord-gcc-switches -mcpu=power8 -mtune=power8 -fasynchronous-unwind-tables -fstack-clash-protection  -DIN_GCC     -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -static-libstdc++ -static-libgcc  attribs.o vhdl/ortho-lang.o \
 libbackend.a libcommon-target.a libcommon.a ../libcpp/libcpp.a ../libdecnumber/libdecnumber.a libcommon.a ../libcpp/libcpp.a   ../libbacktrace/.libs/libbacktrace.a ../libiberty/libiberty.a ../libdecnumber/libdecnumber.a  -ldl -lmpc -lmpfr -lgmp -rdynamic -ldl  -lz
gnatmake: "ghdl1" up to date.
gnatmake -o ghdl -aI../../gcc/vhdl -aI../../gcc/vhdl/ghdldrv \
 -aIvhdl -aOvhdl ghdl_gcc \
 -bargs -E -cargs  -gnata -gnaty3befhkmr -gnatwa -largs libcommon.a ../libcpp/libcpp.a   ../libbacktrace/.libs/libbacktrace.a ../libiberty/libiberty.a ../libdecnumber/libdecnumber.a 
gnatbind -aI../../gcc/vhdl -aI../../gcc/vhdl/ghdldrv -aIvhdl -aOvhdl -E -x ghdl_gcc.ali
gnatlink ghdl_gcc.ali -o ghdl libcommon.a ../libcpp/libcpp.a ../libbacktrace/.libs/libbacktrace.a ../libiberty/libiberty.a ../libdecnumber/libdecnumber.a
/usr/bin/ld: ./synth-expr.o: undefined reference to symbol 'ceil@@GLIBC_2.17'
/usr/bin/ld: //usr/lib64/libm.so.6: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
gnatlink: error when calling /usr/bin/gcc
gnatmake: *** link failed.
make[2]: *** [../../gcc/vhdl/Make-lang.in:115: ghdl] Error 4
make[2]: Leaving directory '/home/dan/projects/fedora/ghdl/gcc-9.1.1-20190503/obj-ppc64le-fedora_ghdl-linux/gcc'
make[1]: *** [Makefile:4300: all-gcc] Error 2
make[1]: Leaving directory '/home/dan/projects/fedora/ghdl/gcc-9.1.1-20190503/obj-ppc64le-fedora_ghdl-linux'
make: *** [Makefile:937: all] Error 2
````
so `-lm` is missing somewhere, probably in `ghdl/src/ortho/gcc/Makefile` in `HOST_LIBS`?

tgingold, 2019-08-28:
That has to be fixed (probably in Makefile.in)

1138-4EB, 2019-08-28:
@sharkcz, thanks! I updated the tasklist according to your comments.

> when building with the synth feature enabled I get a missing symbol error during linking (it's on ppc64le)
> 
> so `-lm` is missing somewhere, probably in `ghdl/src/ortho/gcc/Makefile` in `HOST_LIBS`?

@tgingold, is this worth a separate issue?

1138-4EB, 2019-09-01:
While trying to run the testsuite after installing the `rpm` packages produced with `fedpkg local`, I'm getting the following error:

```bash
2019-08-31T22:26:41.3287251Z gna issue450: [31mfailed[0m
2019-08-31T22:26:41.3298512Z analyze disptree.vhdl
2019-08-31T22:26:41.3299874Z elaborate disptree
2019-08-31T22:26:41.3300346Z /usr/bin/gcc -c vpi2.c -I/usr/include -fPIC
2019-08-31T22:26:41.3300453Z vpi2.c:9:10: fatal error: vpi_user.h: No such file or directory
2019-08-31T22:26:41.3300498Z     9 | #include <vpi_user.h>
2019-08-31T22:26:41.3300539Z       |          ^~~~~~~~~~~~
2019-08-31T22:26:41.3300599Z compilation terminated.
```

The file should be installed in `/usr/include/ghdl/vpi_user.h`:

```bash
2019-08-31T22:22:26.2118211Z + /usr/bin/mv /root/rpmbuild/BUILDROOT/ghdl-0.37dev-5.20190831gitbe3a72d2.0.fc32.x86_64/usr/include/vpi_user.h /root/rpmbuild/BUILDROOT/ghdl-0.37dev-5.20190831gitbe3a72d2.0.fc32.x86_64/usr/include/ghdl
```

See full log: https://github.com/1138-4EB/rpm-ghdl/runs/208895177

sharkcz, 2019-09-03:
The testsuite should allow having the headers in a `ghdl` subdirectory, which conflicts with the `gcc` call
````
2019-08-31T22:26:41.3300346Z /usr/bin/gcc -c vpi2.c -I/usr/include -fPIC
````

Using per project subdir under `/usr/include` is a good practice.

1138-4EB, 2019-09-03:
The point is that I don't think the testsuite is doing this. The command in the testsuite is `$GHDL --vpi-compile -v gcc -c vpi2.c`. Hence, I think that it is ghdl which is internally adding `-I/usr/include -fPIC`. Maybe this has always been 'broken' but Fedora users either i) don't install GHDL through dnf; ii) don't use VPI; or iii) compile C sources explicitly, instead of using ghdl.

FTR, our docker builds and tests work because we use the default 'make install' and, therefore, header files are located in `/usr/local/include`.

sharkcz, 2019-09-05:
@tsailer, do you have objections if I move the Fedora ghdl package forward with the changes discussed here?

tsailer, 2019-09-05:
Not at all, thanks for the work you do!

sharkcz, 2019-09-05:
So the changes have been pushed out and a new build for rawhide started. And I have already next steps in mind :-)
- update to newer ghdl snapshot
- update to gcc 9.2.1
- possibly cleanup of the gcc build and other cleanups
- implement the backend selection with alternatives

1138-4EB, 2019-09-14:
@sharkcz thanks for pushing these updates!

I saw that you updated ghdl so fixes for synthesis are included: https://src.fedoraproject.org/rpms/ghdl/c/caf3b7d591df4b7cfb907cec1aeaf380ef4ee0c2?branch=master

However, I am a little confused about the new mechanism to retrieve ghdl sources: `https://github.com/ghdl/ghdl/archive/%{ghdlcommit}/%{name}-%{ghdlshortcommit}.tar.gz` is downloaded as `%{name}-%{ghdlshortcommit}.tar.gz`, which is coherent with the file name in `sources`; then, you do `mv ghdl-%{ghdlcommit} ghdl`.

On the one hand, I assume that `wget` is used to retrive the tarball. So, `%{name}-%{ghdlshortcommit}.tar.gz` defines the name of the file. Why not `ghdl.tar.gz`?

On the other hand, I assume that the tarball is extracted automatically, and `ghdl-%{ghdlcommit}` is that name of the tar file, independently of the `tar.gz` wrapper. I wonder if it is possible to provide custom arguments to `tar` in order to avoid the `mv` command. I found little info about it (https://rikers.org/rpmbook/node68.html#SECTION03526100000000000000).

sharkcz, 2019-09-14:
> @sharkcz thanks for pushing these updates!
>
> I saw that you updated ghdl so fixes for synthesis are included: [...]

I have switched to using a snapshot directly provided by github, so no manual actions are needed. It's more compliant with the Fedora Guidelines, see https://fedoraproject.org/wiki/Packaging:SourceURL#Git_Hosting_Services for details.

1138-4EB, 2019-09-16:
@sharkcz, thanks for the reference! I see that `%{name}-%{ghdlshortcommit}.tar.gz` is the format defined https://fedoraproject.org/wiki/Packaging:SourceURL#Commit_Revision. However, I don't understand yet how is the sha512sum updated in the sources. Moreover, when executing `fedpkg --verbose sources`, it seems that tarballs are download from fedora's servers, not from github:

```bash
Downloading nvptx-newlib-aadc8eb0ec43b7cd0dd2dfb484bae63c8b05ef24.tar.xz
Full url: b'https://src.fedoraproject.org/repo/pkgs/rpms/ghdl/nvptx-newlib-aadc8eb0ec43b7cd0dd2dfb484bae63c8b05ef24.tar.xz/sha512/94f7
089365296f7dfa485107b4143bebc850a81586f3460fd896bbbb6ba099a00217d4042133424fd2183b352132f4fd367e6a60599bdae2a26dfd48a77d0e04/nvptx-new
lib-aadc8eb0ec43b7cd0dd2dfb484bae63c8b05ef24.tar.xz'
######################################################################## 100.0%
Downloading nvptx-tools-c28050f60193b3b95a18866a96f03334e874e78f.tar.xz
Full url: b'https://src.fedoraproject.org/repo/pkgs/rpms/ghdl/nvptx-tools-c28050f60193b3b95a18866a96f03334e874e78f.tar.xz/sha512/a688c
b12cf805950a5abbb13b52f45c81dbee98e310b7ed57ae20e76dbfa5964a16270148374a6426d177db71909d28360490f091c86a5d19d4faa5127beeee1/nvptx-tool
s-c28050f60193b3b95a18866a96f03334e874e78f.tar.xz'
######################################################################## 100.0%
Downloading gcc-9.1.1-20190503.tar.xz
Full url: b'https://src.fedoraproject.org/repo/pkgs/rpms/ghdl/gcc-9.1.1-20190503.tar.xz/sha512/9787c7f4bc6a2e918cafa5ef0bc21dd29eb8854
1297168049cffea150d08f50023e6b639bc24fe53daf6f603db286a7a8dd17357d139f0d0cfb3cee2850fee8b/gcc-9.1.1-20190503.tar.xz'
######################################################################## 100.0%
Downloading ghdl-cb34680.tar.gz
Full url: b'https://src.fedoraproject.org/repo/pkgs/rpms/ghdl/ghdl-cb34680.tar.gz/sha512/e796548f396ee4df2b8dedfdc9a3c0fae3b0e3e7e1ad9
60af3abf7c425a5e736af139575880a057cbb8c0e293956f5bdf6c0cb53389bdb33bb71796ed04bf5b5/ghdl-cb34680.tar.gz'
######################################################################## 100.0%
```

The GitHub Action I am trying to set up tries to retrieve

`https://src.fedoraproject.org/repo/pkgs/rpms/ghdl/ghdl-cb3468002497ab2090bef31d559e2eaf4747b90a.tar.gz/sha512/e796548f396ee4df2b8dedfdc9a3c0fae3b0e3e7e1ad960af3abf7c425a5e736af139575880a057cbb8c0e293956f5bdf6c0cb53389bdb33bb71796ed04bf5b5/ghdl-cb3468002497ab2090bef31d559e2eaf4747b90a.tar.gz`

instead of

`https://src.fedoraproject.org/repo/pkgs/rpms/ghdl/ghdl-cb34680.tar.gz/sha512/e796548f396ee4df2b8dedfdc9a3c0fae3b0e3e7e1ad9
60af3abf7c425a5e736af139575880a057cbb8c0e293956f5bdf6c0cb53389bdb33bb71796ed04bf5b5/ghdl-cb34680.tar.gz`

, and it fails. See https://github.com/1138-4EB/rpm-ghdl/commit/83dc19a5fb28c04d4b9429f03c83026e22d4a2bc/checks and https://github.com/1138-4EB/rpm-ghdl/compare/upstream...test.

---

Apart from that, I think that `--enable-synth` might be missing in https://src.fedoraproject.org/rpms/ghdl/blob/master/f/ghdl.spec#L439. I seems that synthesis is only enabled for GCC or LLVM backends.

sharkcz, 2019-09-16:
The sha512 checksum in `sources` is created during `fedpkg <new-source|upload>` when uploading the new source archive. I think that for the CI purpose you could do following
- `fedpkg sources` to populate the build dir with the latest source archives from Fedora
- edit `ghdl.spec` with new date/git hash
- run `spectool -g ghdl.spec`, so it downloads the ghdl snapshot from github, `spectool` can be used to download sources as defined in the `Source` tags from their upstream location.
- run `fedpkg local` for the actual build

Looks to me your URL is using the full commit hash instead of the shortened one.

eine, 2020-07-24:
Fedora 31 and 32 builds with LLVM backend are failing (mcode and GCC are successful):

- https://travis-ci.org/github/ghdl/ghdl/jobs/711458892#L388
- https://github.com/ghdl/docker/runs/904893808?check_suite_focus=true#step:4:264
- https://github.com/ghdl/docker/runs/904893825?check_suite_focus=true#step:4:263

```
gnatlink ortho_code_main.ali -o ghdl1-llvm -g llvm-cbindings.o --LINK=clang -L/usr/lib64 -lLLVM-9
/usr/bin/ld: llvm-cbindings.o: undefined reference to symbol '_ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE6lengthEv@@GLIBCXX_3.4.21'
/usr/bin/ld: /usr/lib64/libstdc++.so.6: error adding symbols: DSO missing from command line
clang-9: error: linker command failed with exit code 1 (use -v to see invocation)
gnatlink: error when calling /usr/bin/clang
gnatmake: *** link failed.
```

```
gnatlink ortho_code_main.ali -o ghdl1-llvm -g llvm-cbindings.o --LINK=clang -L/usr/lib64 -lLLVM-10
/usr/bin/ld: llvm-cbindings.o: undefined reference to symbol '_ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE6lengthEv@@GLIBCXX_3.4.21'
/usr/bin/ld: /usr/lib64/libstdc++.so.6: error adding symbols: DSO missing from command line
clang-10: error: linker command failed with exit code 1 (use -v to see invocation)
gnatlink: error when calling /usr/bin/clang
gnatmake: *** link failed.
```

sharkcz, 2020-07-24:
`... --LINK=clang -L/usr/lib64 -lLLVM-9` - this part looks wrong, it should use `clang++` instead of `clang` (see eg. https://github.com/ghdl/ghdl/blob/master/src/ortho/llvm6/Makefile#L3)

sharkcz, 2020-07-24:
https://github.com/ghdl/ghdl/blob/master/dist/ci-run.sh#L284 is the bug

eine, 2020-07-24:
@sharkcz thanks a lot! So fast and effective! https://travis-ci.org/github/eine/ghdl/jobs/711492252

eine, 2020-08-05:
I just saw the description in https://src.fedoraproject.org/rpms/ghdl. It might need to be updated, since GHDL supports synthesis now.

sharkcz, 2020-08-05:
> I just saw the description in https://src.fedoraproject.org/rpms/ghdl. It might need to be updated, since GHDL supports synthesis now.

true, do you have a good description text we could use? Or shall we just update the note about synthesis?

eine, 2020-08-05:
I would just copy/update the first paragraph of the README: https://github.com/ghdl/ghdl/blob/master/README.md

> This directory contains the sources of GHDL, **the open-source analyzer, compiler, simulator and (experimental) synthesizer** for VHDL, a Hardware Description Language (HDL). GHDL is not an interpreter: it allows you to analyse and elaborate sources to generate machine code from your design. Native program execution is the only way for high speed simulation.

eine, 2020-08-05:
Proposal:

> GHDL is the open-source analyzer, compiler, simulator and (experimental) synthesizer for VHDL, a Hardware Description Language (HDL). GHDL implements the VHDL language according to the 1987, 1993 and 2002 versions of the IEEE 1076 VHDL standard, and partial for 2008.
It compiles VHDL files and creates a binary that simulates (or executes) your design. GHDL can also translate a design into a VHDL 1993 netlist, or it can be plugged into Yosys for open-source synthesis.

Since GHDL is a compiler (i.e., it generates object files), you can call functions or procedures written in a foreign language, such as C, C++, Ada95 or Python.

sharkcz, 2020-08-28:
Updated via https://src.fedoraproject.org/rpms/ghdl/c/66e31cef7ddd804d9d58c29097918325571cc10e?branch=master