ghdl/ghdl #890: clang: support for -fstack-clash-protection

1138-4EB, 2019-08-09:
One of the patches applied when ghdl is packaged for Fedora (ref #888), is to ensure that flag `-fstack-clash-protection` is not passed to the compiler (clang). See https://src.fedoraproject.org/rpms/ghdl/blob/master/f/ghdl-llvmflags.patch

I think this is because clang does not support it. At first, it was just ignored, so no error was produced. Then, it was changed in order to produce an error. That's why this patch seems to be required.

I wonder whether the patch can be applied here. I have searched for `fstack-clash-protection` in the codebase, but I found nothing. Therefore, I don't know where might the flag be coming from.

/cc @tsailer

Refs:

- https://bugzilla.redhat.com/show_bug.cgi?id=1658311
- https://reviews.llvm.org/D42593

tgingold, 2019-08-10:
I think this is too Fedora specific.

Another possibility is to compile the c++ files using g++ (instead of clang++).  Maybe it works fine on Fedora.

sharkcz, 2019-08-23:
Yes, this is Fedora specific. It's a way how to make sure clang doesn't fail to compile sources with the system-wide CFLAGS/CXXFLAGS, whose use is mandatory. It's a valid example of a downstream-only patch.

1138-4EB, 2019-08-23:
@sharkcz, thanks a lot for clarifying.

closed 2019-08-23 by 1138-4EB