ghdl/ghdl #894: VHPIDIRECT callback on simulation termination

vdkeybus, 2019-08-15:
**Is your feature request related to a problem? Please describe.**
Add a clear and concise description of what the problem is. E.g. *I'm always frustrated when [...]*

**Describe the solution you'd like**
Add a clear and concise description of what you want to happen.

**Describe alternatives you've considered**
Add a clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

1138-4EB, 2019-08-15:
Hi @vdkeybus! Can you please provide a minimal example of what you are trying to achieve? I believe that the problem you are having is related to the version of VHDL that you are using. The behaviour is very different with `--std=93` or `--std=08`. With VHDL 2008 you should be all good.

With VHDL 93, when the simulator exists cleanly, you can use `atexit` to register an `exit_handler` as in https://github.com/VUnit/vunit/blob/36d63f1645ae64b3de86142579433a9c5feaee7a/examples/vhdl/external_buffer/src/main.c. I believe that this is equivalent to what you mean with `VHPIDIRECT_EXIT <function_exit>`, isn't it?

However, when the simulation is finished with a failure in VHDL 93, you can get an abortion. In this context, SIGABRT can be used as a handler, as shown in https://github.com/VUnit/vunit/blob/36d63f1645ae64b3de86142579433a9c5feaee7a/examples/vhdl/external_buffer/sigabrt.md. This approach is limited, tho. For example, if the GHDL binary is loaded and executed from Python, there seems not to be possible to execute any Python cleanup code after the abortion of GHDL (see https://github.com/VUnit/vunit/pull/469#issuecomment-485723516).

Note that the abortion is not expected. It is being tracked in #803, but the issue has not been updated since april. It might be worth trying with latest `master`.

vdkeybus, 2019-08-15:
@1138-4EB I am going to try the atexit() solution you propose. I wasn't aware you could use it. I'll try to catch SIGABRT for good measure too.

So the feature already exists. That's good. I'm now wondering, are FAILURE assertions actually implemented using signals ? So, could I also terminate a GHDL run that way from C ?

BTW thanks for analyzing my request and compiling the elaborate response. Is there a way we can include this kind of useful information in the foreign language manual section ?

1138-4EB, 2019-08-15:
> @1138-4EB I am going to try the atexit() solution you propose. I wasn't aware you could use it.

In practice, `ghdl_main` is "just another regular C function". Hence, you can use regular C tooling/libraries. For example, you can potentially use pthreads and dlopen to load two binaries generated with ghdl and effectively co-simulate two independent designs, while using C to allocate/provide a shared memory space.

> I'll try to catch SIGABRT for good measure too.

> So the feature already exists. That's good. I'm now wondering, are FAILURE assertions actually implemented using signals ? So, could I also terminate a GHDL run that way from C?

Note that this is likely to be a bug. According to @tgingold, ghdl should not exit with an abortion, not even after a simulation failure. See https://github.com/ghdl/ghdl/issues/800#issuecomment-486936060. So, it is probably not ghdl who is producing the signal, but some other OS logic in-between C main and ghdl_main.

Nonetheless, I think you should be able to terminate a GHDL run that way from C. I mean, this is not dependent on ghdl's codebase but on the context/system that loads the function, isn't it?

> BTW thanks for analyzing my request and compiling the elaborate response.

You are welcome.

> Is there a way we can include this kind of useful information in the foreign language manual section?

Actually, I'm the one who did latest updates to that section: https://github.com/ghdl/ghdl/blob/master/doc/using/Foreign.rst#id4. I'd be glad to review a PR if you are willing to propose it. I have not added it myself yet because I'd like to investigate the abortion issue further. Nevertheless, it is ok to add it with some hint/warning if you think this is useful.

tgingold, 2019-08-16:
There is the endofsimulation vpi/vhpi callback for this purpose.  That's heavier to use than a vhpidirect function but that's the standard way.

vdkeybus, 2019-08-17:
I need to have a look into these callbacks, since I was under the
impression that they were not intended to be used in entities / as
components. I need something lightweight that provides input to a
simulation on regular time instants.

Thanks,