ghdl/ghdl #896: reference to directory which doesn't exist

Garabito, 2019-08-20:
**Description**
i download a ghdl-0.36-mingw64-llvm prebuilded and run in windows 10, i configure the path, and run in powershell ghdl --version  and ghdl -a -v something.vhdl without problem, but when i make a ghdl -e -v something.vhdl, this drop me some error  searching libraries in a directory which i don't have in my computer. 

**Expected behaviour**
What you expected to happen, and what is happening instead.

**Context**
 ghdl -a -v compara_tb.vhdl
C:\uC\VHDL\GHDL\bin\ghdl1-llvm.exe -PC:\uC\VHDL\GHDL\lib/ghdl\ieee\v93\ -PC:\uC\VHDL\GHDL\lib/ghdl\ -c -o compara_tb.o compara_tb.vhdl  <---- no problem
---------
ghdl -e -v comara_tb
C:\uC\VHDL\GHDL\lib/ghdl/libgrt.a(win32.o): In function `ghdl_SEH_handler':
C:/projects/ghdl-psgys/build/mingw64-llvm/../../src/grt/config/win32.c:150: undefined reference to `__imp___acrt_iob_func'    **<------ i don't have this directory**
C:/projects/ghdl-psgys/build/mingw64-llvm/../../src/grt/config/win32.c:150:(.text+0x297): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `__imp___acrt_iob_func'
C:\uC\VHDL\GHDL\lib/ghdl/libgrt.a(grt-cbinding.o): In function `snprintf':
C:/msys64/mingw64/x86_64-w64-mingw32/include/stdio.h:790: undefined reference to `__ms_vsnprintf'
C:/msys64/mingw64/x86_64-w64-mingw32/include/stdio.h:790:(.text+0x3a): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `__ms_vsnprintf'
C:\uC\VHDL\GHDL\lib/ghdl/libgrt.a(grt-cbinding.o): In function `__ghdl_get_stdout':
C:/projects/ghdl-psgys/build/mingw64-llvm/../../src/grt/grt-cbinding.c:26: undefined reference to `__imp___acrt_iob_func'   **<--- neither this**
C:/projects/ghdl-psgys/build/mingw64-llvm/../../src/grt/grt-cbinding.c:26:(.text+0x5a): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `__imp___acrt_iob_func'
C:\uC\VHDL\GHDL\lib/ghdl/libgrt.a(grt-cbinding.o): In function `__ghdl_get_stdin':
C:/projects/ghdl-psgys/build/mingw64-llvm/../../src/grt/grt-cbinding.c:32: undefined reference to `__imp___acrt_iob_func'   
C:/projects/ghdl-psgys/build/mingw64-llvm/../../src/grt/grt-cbinding.c:32:(.text+0x76): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `__imp___acrt_iob_func'
C:\uC\VHDL\GHDL\lib/ghdl/libgrt.a(grt-cbinding.o): In function `__ghdl_get_stderr':
C:/projects/ghdl-psgys/build/mingw64-llvm/../../src/grt/grt-cbinding.c:38: undefined reference to `__imp___acrt_iob_func'
C:/projects/ghdl-psgys/build/mingw64-llvm/../../src/grt/grt-cbinding.c:38:(.text+0x92): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `__imp___acrt_iob_func'
C:\uC\VHDL\GHDL\lib/ghdl/libgrt.a(grt-cvpi.o): In function `loadVpiModule':
C:/projects/ghdl-psgys/build/mingw64-llvm/../../src/grt/grt-cvpi.c:149: undefined reference to `__imp___acrt_iob_func'
C:/projects/ghdl-psgys/build/mingw64-llvm/../../src/grt/grt-cvpi.c:149:(.text+0xbc): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `__imp___acrt_iob_func'
C:/projects/ghdl-psgys/build/mingw64-llvm/../../src/grt/grt-cvpi.c:162: undefined reference to `__imp___acrt_iob_func'
C:/projects/ghdl-psgys/build/mingw64-llvm/../../src/grt/grt-cvpi.c:162:(.text+0x10e): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `__imp___acrt_iob_func'
C:\uC\VHDL\GHDL\lib/ghdl/libgrt.a(grt-cvpi.o):C:/projects/ghdl-psgys/build/mingw64-llvm/../../src/grt/grt-cvpi.c:186: more undefined references to `__imp___acrt_iob_func' follow
C:\uC\VHDL\GHDL\lib/ghdl/libgrt.a(grt-cvpi.o): In function `loadVpiModule':
C:/projects/ghdl-psgys/build/mingw64-llvm/../../src/grt/grt-cvpi.c:186:(.text+0x1ba): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `__imp___acrt_iob_func'
C:/projects/ghdl-psgys/build/mingw64-llvm/../../src/grt/grt-cvpi.c:205:(.text+0x256): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `__imp___acrt_iob_func'
C:/projects/ghdl-psgys/build/mingw64-llvm/../../src/grt/grt-cvpi.c:209:(.text+0x293): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `__imp___acrt_iob_func'
C:\uC\VHDL\GHDL\lib/ghdl/libgrt.a(fstapi.o): In function `tmpfile_open':
C:/projects/ghdl-psgys/build/mingw64-llvm/../../src/grt/fst/fstapi.c:196:(.text+0xb6): additional relocation overflows omitted from the output
C:\uC\VHDL\GHDL\lib/ghdl/libgrt.a(fstapi.o): In function `fstRealpath':
C:/projects/ghdl-psgys/build/mingw64-llvm/../../src/grt/fst/fstapi.c:297: undefined reference to `__imp__fullpath'
C:\uC\VHDL\GHDL\lib/ghdl/libgrt.a(fstapi.o): In function `fstWriterCreate':
C:/projects/ghdl-psgys/build/mingw64-llvm/../../src/grt/fst/fstapi.c:1141: undefined reference to `__strtod'
C:\uC\VHDL\GHDL\lib/ghdl/libgrt.a(fstapi.o): In function `fstWriterSetParallelMode':
C:/projects/ghdl-psgys/build/mingw64-llvm/../../src/grt/fst/fstapi.c:2481: undefined reference to `__imp___acrt_iob_func'
C:\uC\VHDL\GHDL\lib/ghdl/libgrt.a(fstapi.o): In function `fstWriterEmitValueChange':
C:/projects/ghdl-psgys/build/mingw64-llvm/../../src/grt/fst/fstapi.c:2763: undefined reference to `__imp___acrt_iob_func'
C:\uC\VHDL\GHDL\lib/ghdl/libgrt.a(fstapi.o): In function `fstWriterEmitVariableLengthValueChange':
C:/projects/ghdl-psgys/build/mingw64-llvm/../../src/grt/fst/fstapi.c:2877: undefined reference to `__imp___acrt_iob_func'    **<---- no exist in my pc**
collect2: error: ld returned 1 exit status
C:\uC\VHDL\GHDL\bin\ghdl.exe: compilation error

 information:

- OS:
- Origin:
  - [ ] Package manager. Repo:
  - [ ] Released binaries. Tarball:
  - [ ] Built from sources. Commit SHA:

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

**How to reproduce?**
i base in some internet examples 
http://enos.itcollege.ee/~vpesonen/lisa/lab_example.html
https://blog.desdelinux.net/como-trabajar-con-vhdl-usando-software-libre/
https://hardwaregeeksblog.wordpress.com/2017/01/18/ghdl-tutorial/
https://www.youtube.com/watch?v=H2GyAIYwZbw 

func.vhd
library IEEE;
use IEEE.std_logic_1164.all;

entity func is
	port(
		I	: in  std_logic_vector(3 downto 0);
        Y	: out std_logic);
end func;

architecture behaviour of func is
begin	
	process (I)
	begin
		case (I) is
			when "0000" => Y <= '1';
			when "0001" => Y <= '1';
			when "0100" => Y <= '1';
			when "1001" => Y <= '1';
			when "1110" => Y <= '1';
			when others => Y <= '0';
		end case;
	end process;
end behaviour;

 Type ghdl -s func.vhd func_tb.vhd to check the syntax.
Type ghdl -a func.vhd func_tb.vhd to analyze the design.
Type ghdl -e t_func to build an executable for the testbench. 

**Files**
-a----    19/08/2019  08:14 p. m.          12350 compara.o
-a----    19/08/2019  08:14 p. m.            718 compara.vhdl
-a----    19/08/2019  02:54 p. m.            714 compara.vhdl~
-a----    19/08/2019  08:55 p. m.          13438 compara_tb.o
-a----    19/08/2019  08:48 p. m.            983 compara_tb.vhdl
-a----    19/08/2019  03:03 p. m.            993 compara_tb.vhdl~
-a----    19/08/2019  09:03 p. m.            329 e~compara_tb.lst
-a----    19/08/2019  09:03 p. m.           3855 e~compara_tb.o
-a----    19/08/2019  08:55 p. m.            387 work-obj93.cf

**Checklist**
Before submitting your issue, please review the following checklist:

- [ ] Add `GHDL Bug occurred` log block
- [ ] Add a MWE
- [ ] Try the latest version

tgingold, 2019-08-20:
I think you need to install the mingw64 libc package.  All the undefined symbols are C symbols.

(Or simply use ghdl with mcode).

Garabito, 2019-08-20:
Hi tgingold:

i have mingw64 and cygwin, and the problem still there.

in the powershell  the gcc -print-search-dir command prints:
install: /usr/lib/gcc/x86_64-pc-cygwin/7.4.0/
programs: =/usr/lib/gcc/x86_64-pc-cygwin/7.4.0/:/usr/lib/gcc/x86_64-pc-cygwin/7.4.0/:/usr/lib/gcc/x86_64-pc-cygwin/:/usr/lib/gcc/x86_64-pc-cygwin/7.4.0/:/usr/lib/gcc/x86_64-pc-cygwin/:/usr/lib/gcc/x86_64-pc-cygwin/7.4.0/../../../../x86_64-pc-cygwin/bin/x86_64-pc-cygwin/7.4.0/:/usr/lib/gcc/x86_64-pc-cygwin/7.4.0/../../../../x86_64-pc-cygwin/bin/
libraries: =/usr/lib/gcc/x86_64-pc-cygwin/7.4.0/:/usr/lib/gcc/x86_64-pc-cygwin/7.4.0/../../../../x86_64-pc-cygwin/lib/x86_64-pc-cygwin/7.4.0/:/usr/lib/gcc/x86_64-pc-cygwin/7.4.0/../../../../x86_64-pc-cygwin/lib/../lib/:/usr/lib/gcc/x86_64-pc-cygwin/7.4.0/../../../x86_64-pc-cygwin/7.4.0/:/usr/lib/gcc/x86_64-pc-cygwin/7.4.0/../../../../lib/:/lib/x86_64-pc-cygwin/7.4.0/:/lib/../lib/:/usr/lib/x86_64-pc-cygwin/7.4.0/:/usr/lib/../lib/:/usr/lib/gcc/x86_64-pc-cygwin/7.4.0/../../../../x86_64-pc-cygwin/lib/:/usr/lib/gcc/x86_64-pc-cygwin/7.4.0/../../../:/lib/:/usr/lib/

so, if the bin no works for me, i try to compile the source, but, i have the next problem, in powershell when i try to type "./configure PREFIX=C:/GHDL", this open a text editor (i don't understand why), and when i write "make" the terminal write "make: *** No targets specified and no makefile found.  Stop.".  Is more easy work with Linux, but i need this in windows :'(. 

Some more ideas?

Thanks

1138-4EB, 2019-08-20:
@Garabito, I think that pre-built binaries are meant for MinGW shells, not for Cygwin. Please, see:

- https://ghdl.readthedocs.io/en/latest/building/mcode/Windows-MinGW-GNAT.html
- https://ci.appveyor.com/project/tgingold/ghdl-psgys
- https://github.com/ghdl/ghdl/blob/master/appveyor.yml

1138-4EB, 2019-09-16:
@Garabito, please see [September 12, 2019 6:29 PM](https://gitter.im/ghdl1/Lobby?at=5d7a72859f5b592e32312094).