ghdl/ghdl #899: Crash when instantiating unconstrained array of unconstrained vector using generics

Schoggy, 2019-08-20:
**Description**
Defining a custom type, an unconstrained array of an unconstrained vector, as introduced in VHDL 2008, a crash occurs during elaboration when the type is instantiated using a generic to define the size of the array.
Using the generic to define the width of the data type, the vector, works as expected.

**Expected behaviour**
The custom type should be able to be instantiated using generics to define each parameter of the data type.

**Context**
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (tarball) [Dunoon edition]
Compiled with GNAT Version: 4.9.3
Target: x86_64-linux-gnu
/home/phil/EES/temp/
Command line:
/home/phil/Downloads/ghdl-master/ghdl_mcode -c --std=08 mwe.vhd -e testing
Exception ORTHO_CODE.SYNTAX_ERROR raised
Exception information:
Exception name: ORTHO_CODE.SYNTAX_ERROR
Message: ortho_code-exprs.adb:1063
Call stack traceback locations:
0x4d9b22 0x4bc141 0x53a0f4 0x53aa21 0x647b3a 0x647e80 0x64e5a7 0x64e601 0x6538a0 0x60d8e7 0x67d8d2 0x5e4ee2 0x67b969 0x5efaa2 0x597acd 0x697a15 0x5295b1 0x52d1f7 0x4097f8 0x7fa48d21b82e 0x4084a7 0xfffffffffffffffe
******************************************************************

- OS: Linux 64bit (Linux Mint)
- Origin:
  - [x] Released binaries. Tarball: version 0.36
  - [x] Built from sources. Commit SHA: c76c317

**Additional context**
This issue might be related to #756 

**How to reproduce?**

```
#>> mwe.vhd
library IEEE;
use IEEE.NUMERIC_STD.ALL;

entity testing is
generic(
    GENVAL : integer := 4);
end entity;

architecture RTL of testing is
    type pixel_line is array(natural range <>) of bit_vector;
    signal data_in : pixel_line(0 to GENVAL)(7 downto 0);
begin
end architecture;

#>> sim.sh
ghdl -a --std=08 mwe.vhd
ghdl -c --std=08 mwe.vhd -e testing

#>> run.sh
docker run --rm -tv /$(pwd):/src:z -w //src ghdl/ghdl:buster-mcode sh -c ./sim.sh

#>> end
```

tgingold, 2019-08-30:
I can reproduce the crash.

eine, 2020-08-05:
I cannot reproduce the crash with latest master. Might this be fixed? @mankeph, do you mind testing?

mankeph, 2020-08-20:
@eine I can confirm, using current master (8789de9), this example elaborates without crashes or errors.
To test I used the llvm backend.

closed 2020-08-20 by eine