ghdl/ghdl #901: New 'packaging' repo in the organization?

1138-4EB, 2019-08-27:
There is currently little to no coordination between developers/contributors of ghdl and packagers/maintainers of different package managers. After checking the existing packages for fedora and archlinux, I saw that we can partially optimize the workflow for all of us. Precisely, issues such as compatibility with new versions of GCC or LLVM were fixed downstream for latest versions, without notifying upstream. Furthermore, since we don't keep an exhaustive changelog, some of the patches might be kept downstream, even after the issues are fixed.

However, it is cumbersome for us to handle users/account in the specific forge that each distribution/platform uses, and to sign all the required agreements. This is specially so because we are not users of some of the platforms.

We could make it easier for all of us to keep package sources up-to-date and to contribute patches upstream if we had a copy of them tightly-coupled to the main repo. Nevertheless, I think that mirroring each of the downstream repos is not practical. We would need to add half a dozen repos, at least. Moreover, many of these package sources have common snippets.

References:
- #377, about debian.
- #497, about mingw32/mingw64.
- #507, about archlinux (AUR).
- #888, about fedora.

Therefore, as we did with ghdl/docker in #577, I propose to create a single new ghdl/packaging repo, with the following structure:

- `archlinux/`:
  - `mcode/`
  - `mcode-git/`
  - `llvm/`
  - `llvm-git/`
  - `gcc/`
  - `gcc-git/`
- `fedora/`:
  - `stable/`
  - `latest/`
  - `master/`
- `msys-mingw/`:
  - `mcode/`
  - `mcode-git/`
  - `llvm/`
  - `llvm-git/`
- `debian/`:
  - `ToDo`

Notes:

- On `archlinux` and `msys-mingw`, non-`*-git` variants produce packages for the latest stable release of ghdl, and `*-git` variants use the latest commit from `master`.
- On `fedora`, `stable` produces packages for the latest stable release of ghdl; `latest` uses some fixed commit which is ahead of stable but behind `master; and `master` use the latest commit from `master`.

Remotes/externals:

- archlinux:
  - https://aur.archlinux.org/ghdl-mcode-git.git maps to mcode-git.
  - https://aur.archlinux.org/ghdl-llvm-git.git maps to llvm-git.
  - https://aur.archlinux.org/ghdl-gcc-git.git maps to gcc-git.
  - mcode, llvm and gcc do not exist.
  - https://aur.archlinux.org/packages/ghdl/ seems to map to gcc.
- fedora:
  - https://src.fedoraproject.org/rpms/ghdl maps to stable (and latest?).
  - https://github.com/1138-4EB/rpm-ghdl/commits/update maps to master.
- msys-mingw:
  - None
- debian:
  - ToDo

Furthermore, the repository would have the following GitHub Actions workflows:

- `.github/workflows/`:
  - `archlinux.yml`: build and test mcode-git, llvm-git and gcc-git.
  - `fedora.yml`: build and test master.
  - `msys.yml`: build and test mcode-git and llvm-git.
  - `debian.yml`: ToDo.
  - `release.yml`: build and test archlinux (mcode, llvm, gcc), fedora (stable, latest), msys (mcode and llvm).

These workflows are periodically executed on amd64 only, as opposed to the broad list of architectures that some of the package managers support. The point is to check that the format of the packager is valid, and to ensure that notable changes are applied homogeneously.

Artifacts of regular workflows (archlinux, fedora, msys, debian) can be downloaded. Hence, amd64 users can download pre-built packages for the very latest development version of GHDL. At the same time, fedora and msys artifacts of the release workflow can be published to GitHub Releases, in replacement of current 'raw' tarballs and zipfiles.

Last, but not least:

- If ghdl/packaging is created, I think we can remove https://github.com/ghdl/MINGW-packages. As commented above, it is not practical to have forks of all the downstream package source repos.
- Using GitHub Actions and msys-mingw PKGBUILD files would partially superseed the existing build wokflow for Windows (#174).
  - Appveyor will not be used anymore.
  - Scripts in `dist/windows/appveyor` are not required anymore. However, the `check()` step in the PKGBUILD file/process chokes on expected failures in the testsuite. So, we might reuse the powershell test scripts.

---

I must admit that I am not completely happy with this proposal, because it doesn't allow commits to be easily cherry-picked by downstream maintainers. Anyway, I think we can come up with some tiny scripts that extract the relevant part of the history. It is just difficul to foresee now.

marzoul, 2019-08-27:
@1138-4EB Hi, I created the packages *-git for ArchLinux :
https://aur.archlinux.org/ghdl-mcode-git.git
https://aur.archlinux.org/ghdl-llvm-git.git
https://aur.archlinux.org/ghdl-gcc-git.git
Just wanted to say I'm aware of the discussions and of the evolution of ghdl, if any changes would make sense don't hesitate to say.

1138-4EB, 2019-08-27:
Hi @marzoul! Thanks for showing up! I was unsure about the GitHub user matching the AUR user, so I didn't ref you explicitly. Still, you were so fast :D

Let's tackle ArchLinux issues in #507.

> Just wanted to say I'm aware of the discussions and of the evolution of ghdl, if any changes would make sense don't hesitate to say.

Just for clarification, I am proposing enhancements from a technical point of view. I know that maintainers do directly or indirectly follow/watch this repo, and I have nothing to criticize. I must apologise if my wording was/is not correct.

Do you think that such a repo would be helpful? I ask it because the three repos/packages you maintain are probably the ones that makes less sense to duplicate. This is because those are `*-git` variants, so they break almost instantly downstream. The advantage we could provide is CI, so the GitHub Action would tell us when packages are broken, before any users attemps to build them. Do you actually have any CI in AUR (such as fedora's copr)?

marzoul, 2019-08-28:
Hi @1138-4EB, no problem, in fact English is not my native language so it's always a challenge when writing, to make it visible that I just speak neutral.

I really don't know how to take position for your question - whether or not your repo proposal would be helpful and how it could be helpful. There is balance effort vs added value trade-off.
It's always interesting to have some reference package scripts ready in a project, even if it's just for example purposes. These are distribution-specific, I only expect these to be up to date in the distribution context. A copy in the GHDL repo itself would make things simpler to my eyes, require less administrative tasks with multiple repos.

What makes things "break" for a distribution is often not the package script itself. Difference in lib versions and compiler accounts for most, what compiles somewhere is explicitely deprecated elsewhere, etc. Proper fix is upstream patch in the code, fixing some package scripts may come along, and it makes sense to add this in the same commit or series of commits...

I have zero knowledge about CI tools, this answers your last question, and this may be why what I say is missing the original intended point of you proposed repo xD

1138-4EB, 2019-08-28:
> I really don't know how to take position for your question - whether or not your repo proposal would be helpful and how it could be helpful. There is balance effort vs added value trade-off.

Just knowing that it is not easy for you to take position is valuable for me already.

> It's always interesting to have some reference package scripts ready in a project, even if it's just for example purposes. These are distribution-specific, I only expect these to be up to date in [...]

I did also think about adding these resources to the main repo. Precisely, the mid-term objective with these distribution-specific packages is to replace most of [`dist`](https://github.com/ghdl/ghdl/tree/master/dist).

The reason to split it to a separate package is purely organizational. @tgingold is the author of ghdl as a whole and he is focused on the code, not on chores. Keeping this in a separate repo allows to keep his main 'history' clean from 'noise'. From my point of view, and that of package maintainers from different distributions, we have a specific location to share issues with updates/releases; without producing/receiving 'noise' about user-related issues.

> What makes things "break" for a distribution is often not the package script itself. Difference in lib versions and compiler accounts for most, what compiles somewhere is explicitely deprecated elsewhere, etc.

I agree on that. Furthermore, package scripts need not to be updated frequently; there is normally a frame of several months between releases. However, there are some details about GHDL which we should account for:

- Three different backends are supported. Unlike other projects, these are not external plugins in third-party repos, but sources to build them are all together. As a result, up to 6 packages can be maintained in each distribution (3 stable and 3 unstable).
  - The naming of these up to 6 packages is not consistent. For example, in fedora `ghdl` is the one with GCC backend, and `ghdl-gcc` does not exist.
- New features were introduced in the last months, such as [ghdl/ghdl-language-server](https://github.com/ghdl/ghdl-language-server) or synthesis (either built-in or as a [module for yosys](https://github.com/tgingold/ghdlsynth-beta/)).

As a result, the proposed new repo is more of a sandbox to work on the packaging of the project as a whole, and to (try to) provide consistency to naming of packages across distributions. A kind of 'guidelines for downstream packagers'.

For example, your approach seems to be to `--enable-libghdl --enable-synth` by  default, unless you find some issue. When possible, this is the most desirable indeed, because:

- Users that want to use LSP features need to only install two python packages and a client.
- Users that want to use ghdlsynth-beta do not need to rebuild ghdl.

Python packages and clients can be published each through the corresponding package manager. However, we need to decide how to package ghdlsynth-beta (ref tgingold/ghdlsynth-beta#31). This needs to match some range of ghdl versions and some range of yosys versions. Ideally, one would install ghdl at one point, yosys at any other point and ghdlsynth at any last point. None of the stages should require rebuilding any of the previous steps. But this might not always be possible. Furthermore, should it be possible to change from, say, ghdl-mcode to ghdl-llvm without rebuilding ghdlsynth?

These are just some examples of questions/points that arise when thinking about 'packaging' ghdl as a project. Since some of the features are very experimental, this is a rapidly changing project (see #884, #848, #844, etc.). I believe that this will we stabilized in the mid-term and at that point a separate repo might not make so much sense. We can always merge it when the time comes.

> Proper fix is upstream patch in the code, fixing some package scripts may come along, and it makes sense to add this in the same commit or series of commits...

As said, @tgingold is focused on code development, not on chores such as docs or packaging. Therefore, it is unlikely for him to push changes to this areas, even if everything is contained in the same repo. However, it is possible to cross-reference issues/PRs/commits.

> I have zero knowledge about CI tools, this answers your last question, and this may be why what I say is missing the original intended point of you proposed repo xD

The main idea in this context is that a set of workflows/tasks/jobs/processes are executed:

- Each time a change to some related file is pushed to the repo.
- Every day/week/month as scheduled in a CRON job.

The tasks/jobs are:

- Checkout the latest ghdl from `master`.
- (if required) Run a script to update the PKGBUILD/spec file.
- In a 'clean' docker container, build with makepkg/makepkg-mingw/fedpkg and generate a package (tarball).
- In a different 'clean' docker container, install the package and run the testsuite.

In anything fails, we know that the next time you try to update one of your AUR packages, you will need to manually fix something. If we want/can, we may fix it now, so we keep checking any update to ghdl. Whenever you want to update a package in AUR, just check the latest succesful CI run and pick it.

As said, in your case, this (only) saves you having to try compiling the package every once in a while (I don't know how frequently you do it). Moreover:

[... cut ...]
- Build inside the container: https://github.com/1138-4EB/rpm-ghdl/blob/master/.github/actions/latest/run.sh#L14-L25

Note that, as opposed to AUR, in fedora a single spec files defines multiple packages (see artifacts in https://github.com/1138-4EB/rpm-ghdl/commit/ba9a92b8e181a3d75e1bf9ae334be3066516ec35/checks).

Anyway, you've been maintaining packages longer than me. If you feel that I'm proposing/explaining a too complex workflow, please do not hesitate to tell so. Should this complexity be resolved by just discussing the approach here, that'd be ok too.

eine, 2020-07-23:
Coming from #1214

> > @eine: The complexity related to having multiple backends installed at the same time is that all VHDL sources can be shared, but LLVM's and GCC's `*.o` objects are NOT compatible. In Debian, they create separate subdirs (`/usr/lib/ghdl/mcode`, `/usr/lib/ghdl/llvm` and `/usr/lib/ghdl/gcc`) and all VHDL sources are duplicated too.
> > Ideally, I think that all packagers (Debian, Fedora, Arch, MSYS2...) should agree on a similar structure, so we can advance towards providing a similar CLI experience to users on any platform.
> 
> @FFY00: I am totally up for it, but IMO it is something that would preferably be provided by the upstream Makefile. Would it be possible to add this scheme? Or perhaps entirely switch to this scheme.

@FFY00, agree. Ideally, the makefile would have a base `install` that copies common resources, and an `install backend` which installs an specific backend in a multi-backend compatible structure. I guess that, for it to work, GHDL's default search paths, and the pre-compilation of the libs need to be modified.

I don't think this is prioritary, but it'd be very desirable.

FFY00, 2020-07-23:
Great! Unfortunately I don't have time to send a patch for this :confused:. When this scheme is added please let me know so that I can update the PKGBUILD.

eine, 2020-07-23:
Packages for MSYS2 are about to be published, and the issue about versioning was raised again. I talked with Tristan about it, and he is ok with publishing a tag (not a release) every month. That is, no GitHub Release will be created and no binary assets will be uploaded. The format we have discussed is the following:

- `1.0`: regular tag and stable release.
- `1.0-dev`: regular version string in the codebase (no corresponding tag).
- `1.0-pkg.20200802`: tag for packagers (no release).

There are two points that might be discussed:

- Should we use `1.0.0` instead of `1.0` to be semver compilant? https://semver.org/
- Do we want to attach the date to the pkg tags, or can we number them sequentially?

@FFY00, @marzoul, @sharkcz the purpose of these tags is mostly to avoid packagers using arbitrary commits SHAs. Please, let us know what you think about the format and the frequency.

FFY00, 2020-07-23:
> I talked with Tristan about it, and he is ok with publishing a tag (not a release) every month.

Yeah, you can call it a snapshot.

> `1.0-pkg.20200802`: tag for packagers (no release).

I'd prefer something like `1.0.r5425.01c82f09` (`<tag>.r<commit number>.<commit ID>`).

The commit number can be given by `git rev-list --count HEAD` and the short commit ID by `git rev-parse --short HEAD`.

> Should we use `1.0.0` instead of `1.0` to be semver compilant? [semver.org](https://semver.org/)

If you plan to follow semver, yes.

eine, 2020-07-23:
@FFY00 I think it would need to be `<major>.<minor>.<patch>-g<commit_number>.<commit_ID>` (`-` instead of `.` between the version and the extra data). Apart from that, I like the format and I like naming it snapshot. Regarding semver, I think we might consider it after `1.0` is released; not important ATM.

FFY00, 2020-07-23:
Why? Also, `r<commit number>` is a pretty well established standard, I would prefer it over `g<commit number>`.

eine, 2020-07-23:
We are already using git (e.g. `GHDL 1.0-dev (v0.37.0-614-g90c92fe9)`) and I didn't understand what does `r` stand for. But I'm ok with `r`, as long as it does not stand for "release".

FFY00, 2020-07-23:
`r` stands for "revision".

tmeissner, 2020-07-23:
Tag and commit id are redundant.

tmeissner, 2020-07-23:
The same for commit number. What’s the use case of these redundant fields?

FFY00, 2020-07-23:
These fields are meant to be read by users, not computers. And they are only redundant if you have access to the repo history.

The current approach allows me to do several things by just looking at the version string:
  - Know with which that the commit is associated with
  - See which of multiple versions is the latest (by looking at the commit number)
  - Know which commit is being packaged
  - ...

eine, 2020-07-23:
@tmeissner, strictly, we only need numbers starting from 1. The date, the revision, the SHA, any other data is redundant. However, I understand one might one to see easily the difference between two commits (either by comparing dates or number of commits).

eine, 2020-08-09:
@Paebbels let me know about [repology.org](https://repology.org/). It seems interesting in the context of this issue: [repology.org/project/ghdl](https://repology.org/project/ghdl). However, some packages are not properly gathered:

- https://repology.org/project/ghdl/versions
- https://repology.org/project/ghdl/related