ghdl/ghdl #912: GHDL doesn't seems to add file dependencies in VHDL-2000

suzizecat, 2019-09-01:
**Description**
Using `--std=00`, GHDL doesn't seems to properly resolve dependencies between files which results in components not being mapped.

`--gen-makefile` will *not* list required files.

Using `--std=02` works as expected

**Expected behaviour**
All dependencies to appear as such in the Makefile and the entities to be mapped.

**Context**
Please paste the `GHDL Bug occurred` log block here. Also, provide the following information:
```sh
$ghdl --version
GHDL 0.37-dev (v0.36-640-gf977ba0d) [Dunoon edition]
 Compiled with GNAT Version: 8.3.0
 GCC back-end code generator
```
- OS: Ubuntu 19.04
- Origin:
  - [ ] Package manager. Repo:
  - [ ] Released binaries. Tarball:
  - [x] Built from sources. Commit SHA: f977ba0dd5b152e97619ecfe1d848d83f2e062ff

**Additional context**

**How to reproduce?**
In a `test` folder, with `mwe_entity.vhdl` and `mwe_tb.vhdl` files as mentioned below :
```vhdl
-- >> mwe_entity.vhdl
library ieee;
use ieee.std_logic_1164.all;

entity mwe_entity is
  port (
        input : in std_logic;
        output : out std_logic
  );
end mwe_entity;

architecture behav of mwe_entity is
begin
	process(input)
	begin
		output <= input;
	end process;
end behav;	

-- >> mwe_tb.vhdl
library ieee;
use ieee.std_logic_1164.all;

entity mwe_tb is
end mwe_tb;

architecture testbench of mwe_tb is
    component mwe_entity is
        port(
            input : in std_logic;
            output: out std_logic
        );
    end component;
    signal i,o : std_logic;
begin
    dut:mwe_entity
    port map(input => i,output => o);

    process
    begin
        for t in 0 to 10 loop
            i <= '1';
            wait for 5 ns;
            i <= '0';
            wait for 5 ns;
        end loop;
    end process;
end testbench;
-- >> end_src
```
run :
```sh
# >> sim.sh
mkdir work
ghdl -i --workdir=work --std=00 *.vhdl
ghdl --gen-makefile --workdir=work --std=00 mwe_tb
#>> run.sh
docker run --rm -tv /$(pwd):/src:z -w //src ghdl/ghdl:buster-mcode sh -c ./sim.sh
# >> end
```

**Files**
Generated output : 
```
mwe_tb.vhdl:16:5:warning: instance "dut" of component "mwe_entity" is not bound [-Wbinding]
mwe_tb.vhdl:7:14:warning: (in default configuration of mwe_tb(testbench))
```
```Makefile
# Makefile automatically generated by ghdl
# Version: GHDL 0.37-dev (v0.36-640-gf977ba0d) [Dunoon edition] - GCC back-end code generator
# Command used to generate this makefile:
# ghdl --gen-makefile --workdir=work --std=00 mwe_tb

GHDL=ghdl
GHDLFLAGS= --workdir=work --std=00

# Default target
all: mwe_tb

# Elaboration target
mwe_tb: /opt/ghdl/lib/ghdl/ieee/v93/std_logic_1164.o /opt/ghdl/lib/ghdl/ieee/v93/std_logic_1164-body.o work/mwe_tb.o
	$(GHDL) -e $(GHDLFLAGS) $@

# Run target
run: mwe_tb
	$(GHDL) -r mwe_tb $(GHDLRUNFLAGS)

# Targets to analyze files
/opt/ghdl/lib/ghdl/ieee/v93/std_logic_1164.o: /opt/ghdl/lib/ghdl/ieee/v93/../../src/ieee/v93/std_logic_1164.vhdl
	@echo "This file was not locally built ($<)"
	exit 1
/opt/ghdl/lib/ghdl/ieee/v93/std_logic_1164-body.o: /opt/ghdl/lib/ghdl/ieee/v93/../../src/ieee/v93/std_logic_1164-body.vhdl
	@echo "This file was not locally built ($<)"
	exit 1
work/mwe_tb.o: mwe_tb.vhdl
	$(GHDL) -a $(GHDLFLAGS) $<

# Files dependences
/opt/ghdl/lib/ghdl/ieee/v93/std_logic_1164.o: 
/opt/ghdl/lib/ghdl/ieee/v93/std_logic_1164-body.o:  /opt/ghdl/lib/ghdl/ieee/v93/std_logic_1164.o
work/mwe_tb.o:  /opt/ghdl/lib/ghdl/ieee/v93/std_logic_1164.o /opt/ghdl/lib/ghdl/ieee/v93/std_logic_1164-body.o
```
**Checklist**
Before submitting your issue, please review the following checklist:

- [x] Add `GHDL Bug occurred` log block
- [x] Add a MWE
- [ ] Try the latest version

1138-4EB, 2019-09-02:
Although I cannot explain why it is required, you can 'fix' your issue by defining `for all : mwe_entity use entity work.mwe_entity;` in the architecture `mwe_tb(testbench)`. This will bind the component, get rid of the warning and produce the same output with `--std=00` as it does with `--std=02`.

Maybe 'automatic binding' was introduced in VHDL-2002?

tgingold, 2019-09-02:
ghdl is correct by strictly following the VHDL rules described in section 5.2.2 of the LRM (default binding rule).  In vhdl 2000, the entity should be visible while in vhdl 2002, an entity could also simply exist in the library.  You can use the option `--syn-binding` to extend the vhdl rules.

closed 2019-09-02 by tgingold

JimLewis, 2021-06-03:
@tgingold    
I think `--syn-binding` is not necessary in this case.   

If we look at 5.2.2. of 2002, or 7.3.3 of 2019:    

```quote
5.2.2 Default binding indication     
In certain circumstances, a default binding indication will apply in the absence of an explicit binding indication.
The default binding indication consists of a default entity aspect, together with a default generic map
aspect and a default port map aspect, as appropriate.
If no visible entity declaration has the same simple name as that of the instantiated component, then the
default entity aspect is open. A visible entity declaration is either
a) An entity declaration that has the same simple name as that of the instantiated component and that is
directly visible (see 10.3),
b) An entity declaration that has the same simple name as that of the instantiated component and that
would be directly visible in the absence of a directly visible (see 10.3) component declaration with
the same simple name as that of the entity declaration, or
c) An entity declaration denoted by “L.C”, where L is the target library and C is the simple name of
the instantiated component. The target library is the library logical name of the library containing
the design unit in which the component C is declared.
```

The important part is the last sentence of c).   
`The target library is the library logical name of the library containing
the design unit in which the component C is declared.`

The usage of the word "component" here is a little confusing.  
So in investigating this, I looked at 
2002:  12.1 Elaboration of a design hierarchy.   Example on P 162:

```vhdl
-- then its architecture contains the following implicit configuration specification at the
-- end of its declarative part:
for C: comp use configuration anonymous;
-- and the following configuration declaration is assumed to exist when E(A) is
-- elaborated:
configuration anonymous of L.E is   -- L is the library in which E(A) is found.
  for A                                               -- The most recently analyzed architecture
                                                          -- of L.E.
  end for;
end configuration anonymous;
```

Based on the example here it is clear here that the word "component"
here is used in this case to mean design entity.   

I should note that commercial simulators seem to have an interpretation that is the consistent with setting `--syn-binding` in GHDL.   Hence, I think the VHDL community would benefit if we all understood this the same way.   If the text still bothers you, and you are willing to suggest a wording change to `c)` , I would be happy to enter the issue in to the VHDL WG issues list.   

@eine - this is exactly what we were talking about today.

tgingold, 2021-06-04:
So I think we agree.  `--syn-binding` is not necessary if the 2002 (or later) standard is selected, but it is if the 2000 standard (or earlier) is selected.

JimLewis, 2021-06-04:
@tgingold   I am compiling with -08:    
ghdl -a --std=08 -Wno-hide --work=LIB_MEMIO --workdir=/c/SynthWorks/Dev/_lab_build/VhdlTest_test/sim_g/VHDL_LIBS/GHDL-2.0.0-dev/lib_memio/v08 -P/c/SynthWorks/Dev/_lab_build/VhdlTest_test/sim_g/VHDL_LIBS/GHDL-2.0.0-dev /c/SynthWorks/Dev/_lab_build/VhdlTest_test/MemIO/MemIO_Structural.vhd

but then when I elaborate:    
ghdl --elab-run --std=08 --work=LIB_TB1_BASICS --workdir=/c/SynthWorks/Dev/_lab_build/VhdlTest_test/sim_g/VHDL_LIBS/GHDL-2.0.0-dev/lib_tb1_basics/v08 -P/c/SynthWorks/Dev/_lab_build/VhdlTest_test/sim_g/VHDL_LIBS/GHDL-2.0.0-dev TbMemIO
C:/SynthWorks/Dev/_lab_build/VhdlTest_test/Tb1_Basics/TbMemIO_Uart1_UartRx.vhd:164:3:warning: instance "u_memio" of component "memio" is not bound [-Wbinding]
  U_MemIO : MemIO
  ^
C:/SynthWorks/Dev/_lab_build/VhdlTest_test/Tb1_Basics/TbMemIO_Uart1_UartRx.vhd:40:14:warning: (in default configuration of tbmemio(uart1_uartrx))
%% Alert FAILURE in CpuTbPkg, during CPU Write operation at 295 ns   

I turn on `--syn_binding` and it elaborates fine.  

I am guessing this is a different issue.

tgingold, 2021-06-04:
Yes, this looks to be a different issue.  I cannot reproduce the warning with the testcase in the issue.

Could you open an issue for your design ?

JimLewis, 2021-06-04:
Ok.  In the morning.   Late here.

umarcor, 2021-08-15:
@JimLewis, friendly ping. Did you open an issue about the binding rules discussed above?