ghdl/ghdl #913: Crash from function returning record assigned in return statement

zznaki, 2019-09-02:
**Description**
Return in a record assigned within the return record causes a crash during analysis. Simple work around is to use an intermediate variable.

**Expected behaviour**
No crash

**Context**
```
ghdl -a --std=08  ./Pkg_HkPacket_BUG.vhd 

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (BaselineCompiler-650-gbaeaef81) [Dunoon edition]
Compiled with GNAT Version: Community 2018 (20180523-73)
Target: x86_64-apple-darwin16.7.0
xxx/ghdl/
Command line:
/opt/local/bin/ghdl1-llvm --std=08 --work=juice -P/opt/local/lib/ghdl/ieee/v08/ -P/opt/local/lib/ghdl/ -c -o Pkg_HkPacket_BUG.o ./Pkg_HkPacket_BUG.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans.adb:1668
Load address: 0x103123000
Call stack traceback locations:
0x10319b342 0x10333764d 0x103338322 0x10333ac34 0x10333ada3 0x10333a5f4 0x1032f29a7 0x1032f2ac5 0x1032f3687 0x1032f44de 0x1032f390c 0x1032f5ec1 0x1032f5f7c 0x1032fa13e 0x1032f96d2 0x1032a14cd 0x1032b8794 0x1032b8973 0x103340c78 0x1032def8b 0x103341ebf 0x1032913bc 0x10334f9a4 0x103350ce1 0x1033521b9 0x103126804
******************************************************************
ghdl:error: compilation error

```
- OS: MacOSX 10.14.6

- Origin: Compiled from source Repo:
Fix UPF (#905)	baeaef81	1138-4EB <1138-4EB@users.noreply.github.com>	1 Sep 2019 at 09:15

**Additional context**
No
**How to reproduce?**

```
#>> ent.vhd
library IEEE;

use IEEE.Std_logic_1164.all;
use IEEE.Numeric_std.all;
package aPackage is

	subtype T_DBYTE is std_logic_vector(7 downto 0);					
  	type T_RECORD is record
    	flgA   		: std_logic;	
    	flgB	: std_logic;	
    	flgC 		: std_logic;	
    	flgD	: std_logic;	
    	counter	: unsigned(3 downto 0);	
  	end record T_RECORD;  

  	function to_HKBYTE(inVal : T_RECORD) return T_DBYTE;
end package aPackage;

package body aPackage is

	function to_HKBYTE(inVal : T_RECORD) return T_DBYTE is
	begin
		return (7 downto 4 => std_logic_vector(inVal.counter), 
				3 => inVal.flgB, 2 => inVal.flgD, 
				1=> inVal.flgC, 0 => inVal.flgA,
				others => '0');
	end to_HKBYTE; 	
	
end package body aPackage; 

#>> sim.sh
ghdl -a  --std=08 ent.vhd

#>> run.sh
docker run --rm -tv /$(pwd):/src:z -w //src ghdl/ghdl:buster-mcode sh -c ./sim.sh

#>> end
```

**Checklist**
All done

zznaki, 2019-09-02:
Now have also discovered that this crash also occurs as a standard variable assignment within the function (rather than as part of a return statemenbt). In both cases the crash does not  occur in every instance of such an assignment of record fields although I have not been able to determine  exactlky why

diogratia, 2019-09-02:
A work around:

```vhdl
         return std_logic_vector(inVal.counter) & 
                       inVal.flgB & inVal.flgD & inVal.flgC & inVal.flgA;
```

Another work around:
```vhdl
-- subprogram declarative item
        constant counter: std_logic_vector(inVal.counter'range) :=
                    std_logic_vector(inVal.counter);
-- return statement:
        return (7 downto 4 => counter,
                3 => inVal.flgB, 2 => inVal.flgD,
                1 => inVal.flgC, 0 => inVal.flgA,
                others => '0');  -- others choice not needed here
```

The problem can be demonstrated not to be associated with record elements:

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity type_conv_agg is
end entity;

architecture foo of type_conv_agg is
    subtype T_DBYTE is std_logic_vector(7 downto 0);
    signal return_val:  T_DBYTE;
    constant uns_4_bits:    unsigned (3 downto 0) := "1101";
    constant fourbits:      std_logic_vector (3 downto 0) := "1010";
    constant slv_4_bits:    std_logic_vector (3 downto 0) :=
                            std_logic_vector(uns_4_bits);
begin
    process
    begin
        return_val <= (7 downto 4 => std_logic_vector(uns_4_bits),
                       3 downto 0 => fourbits);
        wait;
    end process;
end architecture;
```
 The above test case exhibits the same error.  The second solution can also be demonstrated to work as well:

```vhdl
architecture fie of type_conv_agg is
    subtype T_DBYTE is std_logic_vector(7 downto 0);
    signal return_val:  T_DBYTE;
    constant uns_4_bits:    unsigned (3 downto 0) := "1101";
    constant fourbits:      std_logic_vector (3 downto 0) := "1010";
    constant slv_4_bits:    std_logic_vector (3 downto 0) :=
                            std_logic_vector(uns_4_bits);
begin
    process
    begin
        -- return_val <= (7 downto 4 => std_logic_vector(uns_4_bits), 3 downto 0 => fourbits);
        return_val <= (7 downto 4 => slv_4_bits,
                       3 downto 0 => fourbits);
        wait;
    end process;
end architecture;
```
 
The problem is a -2008 bug where element index ranges matching elements of composites has been added to aggregates. There appears to be element type confusion for the actual going on.

Works:

>element range => element value   
>  
>element range => composite of element types   

Doesn't work:

>element range => type conversion (composite of element types)

commit 324403d7 (2021-09-14): testsuite/gna: add a test and close #913

Also close #926 (looks similar, but without a proper reproducer)
closed 2021-09-14 by tgingold