ghdl/ghdl #915: The testsuite should not depend on the docs.

1138-4EB, 2019-09-03:
Package system managers' build commands, such as `makepkg`, `git clone` a new copy of this repo in a clean temporal path, checkout a specific version, proceed with the build, and remove the temporal dir. I'm using the `prepare()` step, which is executed just before the build, to copy subdir `testsuite` out of the temporal path; so it can be executed later, after the path is removed.

The procedure works as expected, but during execution I found that [testsuite/sanity/005examples](https://github.com/ghdl/ghdl/tree/master/testsuite/sanity/005examples) depends on the docs. Precisely, `doc/using/QuickStartGuide.rst` is parsed to extract `.. code :: VHDL` blocks (a similar procedure to [issue-runner](https://github.com/1138-4EB/issue-runner)). I feel this is not as 'clean' as it could be. Possible alternatives:

1. Invert the dependency. I.e., put VHDL sources in `testsuite/sanity/005examples` and make the `doc/conf.py` copy them.
2. Create a 'neutral' subdir named `examples` and put VHDL files there. Then, import them from the testsuite and/or from the docs.
3. Add a target to the Makefile that allows to 'export' the testsuite to a (standalone) dir.

I did not find this issue before because Python is usually not available in the docker containers where GHDL is built/tested. Now, I saw that 8 tests/issues have python sources:

```
# find testsuite/ -name "*.py"
testsuite/gna/issue283/example.py
testsuite/gna/issue43/run.py
testsuite/gna/issue450/cocotb/top_cocotb.py
testsuite/gna/issue456/gen.py
testsuite/gna/issue456/gen_hex.py
testsuite/gna/issue531/test.py
testsuite/gna/testsuite.py
testsuite/python/units01/show_units.py
testsuite/sanity/005examples/extract_vhdl.py
```

This is ~1.5-2% of the tests. Should we add some anywhere explaining that python might be required for some tests?

Other requirements/dependencies of the testsuite that I found on an `archlinux/base` image are `diffutils` and `grep`.

tgingold, 2019-09-03:
I think that only sanity/005examples and python/ need python.  For the other tests, the python files are not used.  They just come directly from the testcase.

I am ok by moving the examples in the testsuite and change conf.py so that it is included (if this is possible).

Paebbels, 2025-04-27:
@tgingold Do you think this problem still exists? I haven't noticed any strange dependency.

closed 2025-04-27 by tgingold