ghdl/ghdl #919: Core Infrastructure Initiative (CII) Best Practices badge

1138-4EB, 2019-09-06:
https://bestpractices.coreinfrastructure.org/en

> The Linux Foundation (LF) Core Infrastructure Initiative (CII) Best Practices badge is a way for Free/Libre and Open Source Software (FLOSS) projects to show that they follow best practices. Projects can voluntarily self-certify, at no cost, by using this web application to explain how they follow each best practice. The CII Best Practices Badge is inspired by the many badges available to projects on GitHub. Consumers of the badge can quickly assess which FLOSS projects are following best practices and as a result are more likely to produce higher-quality secure software. 

I registered ghdl (https://bestpractices.coreinfrastructure.org/en/projects/3157) and I filled most of the items of the 'passing level criteria'. We are at 80%. I'd like to have some help from the community to complete the remaining 13 item and then add the corresponding badge to the README:

[image]
[image]
[image]

---

- [ ] I set http://ghdl.free.fr as the URL for the project. However, it does not meet the following criteria: *The project sites (website, repository, and download URLs) MUST support HTTPS using TLS.*. Should use https://github.com/ghdl/ghdl as the main URL instead? Or maybe setup ghdl.github.io so it is a mirror/redirection to ghdl.free.fr.
- [ ] *It is SUGGESTED that the Semantic Versioning (SemVer) format be used for releases*. Do we want to adhere to it (https://semver.org/)?
- [ ] *It is SUGGESTED that projects be maximally strict with warnings in the software produced by the project, where practical*. idk what to answer.
- [ ]  'Secure development knowledge'. @tgingold, I think that you should decide these two items.
  - By the same token, please see 'Publicly known vulnerabilities fixed' (2 items) and 'Other security issues' (1 item).
- [ ] *A cryptographic hash (e.g., a sha1sum) MUST NOT be retrieved over http and used without checking for a cryptographic signature.* We do not meed criteria when it is related to the CI. However, I think that GHDL itself does not need any checksum at all, does it?
- [ ] Regarding 'Analysis' I answered that there is no FLOSS static code analysis tool that fits the languages used in this project; but I don't know how to reply to 'Dynamic code analysis' items.

@tgingold, if you login with your GitHub account, I can add you as an editor. Note that you don't need to give permissions to access the GitHub organization. We don't want to created badges for all the repos, just for this one, and it the project is registered already.

tgingold, 2019-09-07:
I have logged in.

1138-4EB, 2019-09-07:
Mmm, it seems that I need your user ID (number). I cannot find/add you as `tgingold`: *Change using + or - followed by a comma-separated user id list*.

tgingold, 2019-09-07:
What's that ?

1138-4EB, 2019-09-07:
Just login and go to Account -> Profile. The id will be provided in the URL. For example, mine is `4248`:

https://bestpractices.coreinfrastructure.org/en/users/4248

tgingold, 2019-09-08:
4251

1138-4EB, 2019-09-08:
Done. You should be able to edit now.

> We expect that normally only one person will edit a particular badge entry at a time. This application uses optimistic locking to prevent saving stale data if multiple users try to edit a badge entry simultaneously. If you have multiple editors, we recommend saving badge entry data incrementally and often (that is a wise practice anyway).

1138-4EB, 2019-09-13:
I filled an explanation about why we don't use SemVer, and I left the two remaining items about 'Dynamic code analysis' as 'unmet' without further explanation. The label is now 'passing'.

> Now updated. 

Thanks!

> Not sure how very relevant it is.

I think this is mostly useful an exercise for us to see what *other communities* expect from FOSS software. Since both Ada and VHDL are minority languages, this is a reference of what we can achieve compared to other larger projects with better tooling. Also, it can be useful for users searching for specific features/procedures in some standard form, should the page become a kind a of standard database of FOSS.