ghdl/ghdl #922: Crash on aggregate in function

slaweksiluk, 2019-09-13:
**Context**
```
[root@324c9de07a6c workdir]# ghdl -a --std=08 a.vhd

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-711-gf857fe22) [Dunoon edition]
Compiled with GNAT Version: 8.3.1 20190223 (Red Hat 8.3.1-2)
Target: x86_64-redhat-linux
/workdir/
Command line:
/usr/local/bin/ghdl1-llvm --std=08 -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ -c -fpic -o a.o a.vhd
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : trans-chap4.adb:307
Call stack traceback locations:
0x7f0306449a11 0x5afd7a 0x57f3b0 0x594cef 0x594e39 0x619453 0x5baef2 0x59e264 0x622a13 0x56c3aa 0x628c03 0x62a213 0x62ba6b 0x409051 0x7f0306056411 0x40823c 0xfffffffffffffffe
******************************************************************
ghdl:error: compilation error
[root@324c9de07a6c workdir]# ghdl -v
GHDL 0.37-dev (v0.36-711-gf857fe22) [Dunoon edition]
 Compiled with GNAT Version: 8.3.1 20190223 (Red Hat 8.3.1-2)
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2019 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
[root@324c9de07a6c workdir]# 
```
- OS:
- Origin:
  - [ ] Package manager. Repo:
  - [ ] Released binaries. Tarball:
  - [x] Built from sources. Commit SHA: f857fe22

**How to reproduce?**
```vhdl
#>> a.vhd
library ieee;
use ieee.std_logic_1164.all;

entity dut is
generic (
    adr_width : positive := 16
);
end entity dut;
architecture rtl of dut is
    function deser(b : bit_vector) return bit_vector is
        variable ab : bit_vector(adr_width-1 downto 0);
        variable ba : bit_vector(adr_width-1 downto 0);
    begin
        -- Below causes crash
        (ab, ba) := b;
        return ab&ba;
    end function;
    signal s1 : bit_vector(adr_width-1 downto 0);
    signal s2 : bit_vector(adr_width-1 downto 0);
begin
    s1 <= deser(s2);
end architecture rtl;

#>> sim.sh
ghdl -a --std=08 a.vhd
ghdl -e --std=08 dut
./dut

#>> run.sh
docker run --rm -tv /$(pwd):/src:z -w //src ghdl/ghdl:buster-mcode sh -c ./sim.sh

#>> end
```

**Checklist**
Before submitting your issue, please review the following checklist:

- [x] Add `GHDL Bug occurred` log block
- [x] Add a MWE
- [x] Try the latest version

tgingold, 2019-09-14:
I am a little bit busy with synthesis, but there are now several issues related to unbounded arrays.  I will have a look at them sometimes in the future...

slaweksiluk, 2019-09-14:
Most of my recent issues have easy workarounds and I'm posting them here just for the record. I don't know if this relevant, but: this issue, https://github.com/ghdl/ghdl/issues/821, https://github.com/ghdl/ghdl/issues/917 are all related to generic parameter in entity - I mean it is impossible to reproduce them without generic parameter usage.

eine, 2020-08-05:
With latest master, analysis and elaboration seems to succeed, but execution fails:

```sh
# ghdl --version
GHDL 1.0-dev (v0.37.0-924-g36f25166@buster-llvm-7) [Dunoon edition]
 Compiled with GNAT Version: 8.3.0
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2020 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO     
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.   

# ghdl -a --std=08 a.vhd

# ghdl -e --std=08 dut

# ./dut
./dut:error: bound check failure at a.vhd:21
in process .dut(rtl).P0
./dut:error: simulation failed
```

slaweksiluk, 2020-08-06:
Please find modified reproducer which  now works:
```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity dut is
generic (
    adr_width : positive := 16
);
end entity dut;
architecture rtl of dut is
    function deser(b : bit_vector) return bit_vector is
        variable ab : bit_vector(adr_width-1 downto 0);
        variable ba : bit_vector(adr_width-1 downto 0);
    begin
        -- Below causes crash
        (ab, ba) := b;
        return ab&ba;
    end function;
    signal s1 : bit_vector(adr_width*2-1 downto 0);
    signal s2 : bit_vector(adr_width-1 downto 0);
begin
    s1 <= deser(s2);
    process is begin
        wait for 1 ns;
        std.env.finish;
    end process;

end architecture rtl;
```

eine, 2020-08-06:
Ouch! Stupid mistake 😆 Guess it's fixed too!

closed 2020-08-06 by eine