ghdl/ghdl #926: Internal error assigning array to integer cast to vector

ozbenh, 2019-09-16:
**Description**
The following code causes ghdl to produce an internal error:

     reg_info <= (15 downto 0 => std_ulogic_vector(to_unsigned(CLK_FREQ, 16)),
		 16 => has_uart_l,
		 17 => has_dram_l,
		 others => '0');
reg_info is a signal std_ulogic_vector(63 downto 0), has_uart_l and has_dram_l are std_ulogic but don't seem to be related to the error. CLK_FREQ is an "integer" generic

**Expected behaviour**

I expect the bottom 16 bits of reg_info to be hard wired to the 16-bit numerical value in CLK_FREQ

**Context**
ghdl -a --std=08 syscon.vhdl

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-645-gef26f6ae) [Dunoon edition]
Compiled with GNAT Version: 7.4.0
Target: x86_64-linux-gnu
/home/ANT.AMAZON.COM/benh/hackplace/microwatt/
Command line:
/usr/local/bin/ghdl1-llvm --std=08 -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ -c -fpic -o syscon.o syscon.vhdl
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans.adb:1668
Call stack traceback locations:
0x4860f3 0x620237 0x620eee 0x62377b 0x6238e2 0x623187 0x5db7f3 0x5db90f 0x5dc4c3 0x5dd2e8 0x5dc72c 0x5dee58 0x5def14 0x5e2c74 0x5e263e 0x59c08d 0x59c244 0x59caaa 0x59d75a 0x5a125e 0x5a1391 0x5a588b 0x5aad03 0x633273 0x57763b 0x6397dc 0x63aebe 0x63c7d1 0x408667 0x7f70776feb95 0x407998 0xfffffffffffffffe
******************************************************************

- OS: Ubuntu 18.04.3
- Origin:
  - [ ] Package manager. Repo:
  - [ ] Released binaries. Tarball:
  - [ X] Built from sources. Commit SHA: ef26f6ae06c01187403be345f5997930cbd44a10

**How to reproduce?**

I'm in an airport lounge, I won't have time to do that for a little while...

1138-4EB, 2019-09-16:
Hi @ozbenh! I think that your issue is because of `15 downto 0 => std_ulogic_vector(to_unsigned(CLK_FREQ, 16)),` not being supported yet. You should be able to work around it by using an intermediate signal.

ozbenh, 2019-09-18:
Yes, I've done that for now.

commit 324403d7 (2021-09-14): testsuite/gna: add a test and close #913

Also close #926 (looks similar, but without a proper reproducer)
closed 2021-09-14 by tgingold