ghdl/ghdl #927: synth: support PSL cover directive for formal verification

tmeissner, 2019-09-18:
**Is your feature request related to a problem? Please describe.**
It would be very nice to have the PSL cover directive supported. With that, we could use the `mode cover` with SymbiYosys to do an analyze of reachability. At the moment, you get an error like this:

```
ghdl --synth --std=08 ../syn/WishBoneMasterE.vhd -e wishbonemastere
synth_statements: cannot handle IIR_KIND_PSL_COVER_DIRECTIVE (../syn/WishBoneMasterE.vhd:153:5)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-738-g1ff70cdd) [Dunoon edition]
Compiled with GNAT Version: Community 2019 (20190517-83)
Target: x86_64-apple-darwin17.7.0
/Users/torsten/Projects/git.goodcleanfun.de/libvhdl/formal/
Command line:
ghdl --synth --std=08 ../syn/WishBoneMasterE.vhd -e wishbonemastere
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:32
Load address: 0x1049ca000
Call stack traceback locations:
0x104a57cea 0x104b5229e 0x104b51bbe 0x104b5216f 0x104b6444c 0x104b644d2 0x104b610ea 0x104b9cf8a 0x104af13bc 0x104ba579d 0x1049cc35e
******************************************************************
```

**Describe the solution you'd like**
It could be implemented the same way as the other directives I assume.

**Describe alternatives you've considered**
There are no real alternatives.

**Additional context**
I've tried to added PSL cover by myself:

https://github.com/tmeissner/ghdl/commit/a5fe2d5696dd27147e057821f03bc985774061eb

But all I got is following sightly different error message:

```
ghdl --synth --std=08 ../syn/WishBoneMasterE.vhd -e wishbonemastere

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-752-geb3d32a6-dirty) [Dunoon edition]
Compiled with GNAT Version: Community 2019 (20190517-83)
Target: x86_64-apple-darwin17.7.0
/Users/torsten/Projects/git.goodcleanfun.de/libvhdl/formal/
Command line:
/opt/ghdl/bin/ghdl --synth --std=08 ../syn/WishBoneMasterE.vhd -e wishbonemastere
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : netlists.adb:207
Load address: 0x10bced000
Call stack traceback locations:
0x10bee1ca1 0x10bd019ab 0x10bd025c3 0x10bd2704b 0x10be75cca 0x10be767f0 0x10be76134 0x10be766e7 0x10be8955c 0x10be895f4 0x10be861fa 0x10bec22f4 0x10be158c8 0x10becab05 0x10bcef3b4
******************************************************************
```

tgingold, 2019-09-18:
It is almost correct, but you have forgotten to create the gate.

tmeissner, 2019-09-18:
Oh, yes. Have added the appropiate lines in `netlist-builders.adb`, `netlist-gates.ads` & `ghdlsynth_gates.h`. With that changes, the arror disappeares.

But I don't know, what to add to netlist-disp_vhdl.adb. For assert & assume it looks like this

```
when Id_Assert =>
  Disp_Template
    ("  \l0: assert \i0 = '1' severity error;" & NL, Inst);
when Id_Assume =>
  Disp_Template
    ("  \l0: assert \i0 = '1' severity warning; --  assume" & NL,
    Inst);
```

I think I can't break down the cover directive to assert as it is done for assume.

tgingold, 2019-09-18:
Well, just use an assert with severity note and "covered" as a message.  This output is mainly used for debugging.

tmeissner, 2019-09-18:
Okay, I added an assert with severity note as you suggested.
Maybe you can look my additions in my fork, if that's all what's needed to support cover directive in synthesis: https://github.com/tmeissner/ghdl/commits/synth_psl_cover

If it's okay, I would combine the 2 commits and create a pull request.

tgingold, 2019-09-18:
It's ok.  It would be nice to add a testcase too.
>

closed 2019-09-19 by tgingold