ghdl/ghdl #931: -c command line option doesn't work correctly

dosadi, 2019-09-21:
Description: When I attempt to use the -c command line option to analyze and elaborate in one step, I can't get the steps described in https://ghdl.readthedocs.io/en/latest/using/InvokingGHDL.html to produce correct behavior:

ghdl-gcc  -c src/controls/hello_world.vhdl --workdir=src/controls/workdir_vhdl -e work
error: cannot open --workdir=src/controls/workdir_vhdl
ghdl-gcc:error: compilation error
make: *** [make/Makefile.controls:2: all] Error 1

ghdl --version:
GHDL 0.36-dev (Ubuntu 0.35+git20181129+dfsg-3ubuntu1) [Dunoon edition]
 Compiled with GNAT Version: 8.2.0
 mcode code generator

1138-4EB, 2019-09-21:
Does directory `src/controls/` exist?

dosadi, 2019-09-21:
#response_container_BBPPID{font-family: initial; font-size:initial; color: initial;}  Yes, I have that command line set in a makefile, and if I remove the -c switch and it's argument to go back to the original command line I used before, it works again.                                                                                                                                             Sent via BlackBerry Hub+ Inbox for Android                                                                                                                                                                           From: notifications@github.comSent: September 20, 2019 11:25 PMTo: ghdl@noreply.github.comReply-to: reply@reply.github.comCc: dosadi82@gmail.com; author@noreply.github.comSubject: Re: [ghdl/ghdl] -c command line option doesn't work correctly (#931)  Does directory src/controls/ exist?

1138-4EB, 2019-09-21:
What about `ghdl-gcc -c src/controls/hello_world.vhdl -e work`?

dosadi, 2019-09-21:
I tried what you suggested, and wow, look at that!  I can't tell if the initial error is legitimate or a side effect of whatever caused that backtrace....

$ ghdl-gcc -c src/controls/hello_world.vhdl -e work
error: cannot find entity or configuration work

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36-dev (Ubuntu 0.35+git20181129+dfsg-3ubuntu1) [Dunoon edition]
Compiled with GNAT Version: 8.2.0
Target: x86_64-linux-gnu
In directory: /home/jon/controlix/
Command line:
/usr/lib/ghdl/gcc/libexec/gcc/x86_64-pc-linux-gnu/8.2.0/ghdl1 -P/usr/lib/ghdl/gcc/vhdl/ieee/v93/ -P/usr/lib/ghdl/gcc/vhdl/ --anaelab work --ghdl-source=src/controls/hello_world.vhdl -quiet -o e~work.s e~work
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : iirs.adb:839
Call stack traceback locations:
0x7f0f386f7521 0x468031 0x518908 0x5101ce 0x50ef77 0x445438 0x95619c 0x43e202 0x44a1d9 0x57dca9 0x43c6bf 0x7f0f381aeb69 0x4405a8 0xfffffffffffffffe
******************************************************************

Execution of /usr/lib/ghdl/gcc/libexec/gcc/x86_64-pc-linux-gnu/8.2.0/ghdl1 terminated by unhandled exception
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : iirs.adb:839
Call stack traceback locations:
0x7f0f386f7521 0x468031 0x518908 0x5101ce 0x50ef77 0x445438 0x95619c 0x43e202 0x44a1d9 0x57dca9 0x43c6bf 0x7f0f381aeb69 0x4405a8 0xfffffffffffffffe
ghdl-gcc:error: compilation error

1138-4EB, 2019-09-21:
According to https://ghdl.readthedocs.io/en/latest/using/InvokingGHDL.html#analyze-and-elaborate-c:

> But, the work library is neither read from disk nor saved.

Hence, it seems that `--workdir` is not required/supported in this case, and it was trying to read `--workdir=src/controls/workdir_vhdl` as a path (with obviously, does not exist).

This might be a legitimate error now. It'd be good if you could provide `src/controls/hello_world.vhdl`, to make reproduction easier.

Nonetheless, note that the version of GHDL that you are using is rather old. It'd be useful if you could try with a more recent version. There have been lots of enhancements this last year.

dosadi, 2019-09-22:
Well, I just tried to build GHDL from a new zip of the repository, and I can't seem to get it to work.  I'm going to open a new issue report here on GH (if necessary) for this new bug, but I guess I'm not going to be able to provide further info on this bug here.  Sorry about that....

dosadi, 2019-09-22:
Here is the hello_world example code you asked for:

$ cat src/controls/hello_world.vhdl 
-- "Hello, world!" Example

entity hello_world is
end;

architecture behavior of hello_world is
begin
	proc_main:process
	begin
		report "Hello, World!";
	end process;
end behavior;

1138-4EB, 2019-09-23:
Thanks! The file looks very simple. I am pretty sure that this is fixed in the latest version. I tried with:

```sh
# ghdl --version
GHDL 0.37-dev (v0.36-684-gc37f7a10-dirty) [Dunoon edition]
 Compiled with GNAT Version: 9.2.0
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2019 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

# ghdl -c test.vhd -e hello_world

# ls
e~hello_world.o test.vhd hello_world.exe*

# ./hello_world.exe
test.vhd:10:1:@0ms:(report note): Hello, World!
test.vhd:10:1:@0ms:(report note): Hello, World!
test.vhd:10:1:@0ms:(report note): Hello, World!
test.vhd:10:1:@0ms:(report note): Hello, World!
test.vhd:10:1:@0ms:(report note): Hello, World!
test.vhd:10:1:@0ms:(report note): Hello, World!
test.vhd:10:1:@0ms:(report note): Hello, World!
test.vhd:10:1:@0ms:(report note): Hello, World!
test.vhd:10:1:@0ms:(report note): Hello, World!
test.vhd:10:1:@0ms:(report note): Hello, World!
test.vhd:10:1:@0ms:(report note): Hello, World!
```

This is with LLVM backend, which is easier to build than GCC. You might want to give it a try, unless you need code coverage analysis (which is only supported with GCC).

dosadi, 2019-09-26:
OK, I got the LLVM backend version of GHDL to build and install, and closed the other bug I had open for that issue.  Now I have yet another new issue coming up:

/opt/ghdl/bin/ghdl -c src/controls/hello_world.vhdl -e hello_world
/usr/bin/ld: /opt/ghdl/lib/ghdl/libgrt.a(grt-backtraces-gcc.o): in function `grt__backtraces__gcc__symbolizer':
/home/jon/Downloads/ghdl-master/src/grt/grt-backtraces-gcc.adb:117: undefined reference to `backtrace_create_state'
/usr/bin/ld: /home/jon/Downloads/ghdl-master/src/grt/grt-backtraces-gcc.adb:126: undefined reference to `backtrace_pcinfo'
collect2: error: ld returned 1 exit status
/opt/ghdl/bin/ghdl:error: compilation error

1138-4EB, 2019-09-26:
Do `ghdl -a` and/or `ghdl -e` work?

dosadi, 2019-09-26:
-a yes, -e no:

$ /opt/ghdl/bin/ghdl -a src/controls/hello_world.vhdl
jon@horizon:~/controlix$ ls
bin  DEADJOE  doc  e~hello_world.o  hello_world.o  LICENSE  make  Makefile  Makefile~  README  src  svn-commit.tmp~  work  work-obj93.cf
jon@horizon:~/controlix$ /opt/ghdl/bin/ghdl -e hello_world
/usr/bin/ld: /opt/ghdl/lib/ghdl/libgrt.a(grt-backtraces-gcc.o): in function `grt__backtraces__gcc__symbolizer':
/home/jon/Downloads/ghdl-master/src/grt/grt-backtraces-gcc.adb:117: undefined reference to `backtrace_create_state'
/usr/bin/ld: /home/jon/Downloads/ghdl-master/src/grt/grt-backtraces-gcc.adb:126: undefined reference to `backtrace_pcinfo'
collect2: error: ld returned 1 exit status
/opt/ghdl/bin/ghdl:error: compilation error

1138-4EB, 2019-09-30:
@tgingold, any guess?

tgingold, 2019-09-30:
ghdl should be built without backtraces.

dosadi, 2019-10-01:
How do you do that?  All I can find in the GHDL docs and on the web show to
to enable, not disable, backtraces....

Jon

tgingold, 2019-10-01:
The easiest way is to edit the Makefile (after configuration) and to clear backtrace_lib.

closed 2020-08-07 by tgingold