ghdl/ghdl #934: Failed to build with LLVM on Ubuntu 19.04

dosadi, 2019-09-23:
**Description**
A clear and concise description of what the issue is about.

Description: GHDL with LLVM backend from snapshot from date 09-21-2019 fails to build

Expected behavior: GHDL configures, builds and installs in /opt/ghdl

Configure/Build logs:

$ ./configure --with-llvm-config --prefix=/opt/ghdl
Use full IEEE library
Build machine is: x86_64-linux-gnu
Debugging is not enabled with llvm 8.0.0
Creating ghdl.gpr
Creating Makefile
for d in src/std src/std/v87 src/std/v93 src/std/v08 src/synopsys src/synopsys/v08 std/v87 ieee/v87 synopsys/v87 std/v93 ieee/v93 synopsys/v93 std/v08 ieee/v08 src/openieee src/ieee src/vital95 src/vital2000 src/mentor src/ieee/v87 src/ieee/v93 src/ieee2008 mentor/v93 synopsys/v08; do \
  mkdir -p lib/ghdl/$d; \
done
Generate ghdlsynth_maybe.ads
Generate default_paths.ads

$ make
if test -d ./.git \
   && desc=`cd .; git describe --dirty`; then \
          sub="s/[(].*[)]/($desc)/"; \
else \
  sub="s/tarball/tarball/"; \
        fi; \
sed -e "$sub" -e "s/@VER@/0.37-dev/" < src/version.in > version.tmp; \

if [ ! -r version.ads ] || ! cmp version.tmp version.ads > /dev/null; then cp version.tmp version.ads; fi
make -f ./src/ortho/llvm4-nodebug/Makefile \
 ortho_srcdir=./src/ortho ortho_exec=ghdl1-llvm \
 GNATFLAGS="-aI./src -aI./src/vhdl -aI./src/grt -aI./src/psl -aI./src/vhdl/translate -aI./src/ghdldrv -aI./src/ortho -aI./src/ortho/llvm4-nodebug -aI./src/synth -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe" LDFLAGS="" \
 LLVM_CONFIG="llvm-config" CXX="clang++" GNATMAKE="gnatmake" \
 all
make[1]: Entering directory '/home/jon/Downloads/ghdl-master'
clang++ -c `llvm-config --cxxflags` -o llvm-cbindings.o src/ortho/llvm4-nodebug/llvm-cbindings.cpp
In file included from src/ortho/llvm4-nodebug/llvm-cbindings.cpp:18:
In file included from /usr/lib/llvm-8/include/llvm-c/Target.h:22:
In file included from /usr/lib/llvm-8/include/llvm-c/Types.h:17:
/usr/lib/llvm-8/include/llvm-c/DataTypes.h:28:10: fatal error: 'cmath' file not found
#include <cmath>
         ^~~~~~~
1 error generated.
make[1]: *** [src/ortho/llvm4-nodebug/Makefile:18: llvm-cbindings.o] Error 1
make[1]: Leaving directory '/home/jon/Downloads/ghdl-master'
make: *** [Makefile:335: ghdl1-llvm] Error 2

1138-4EB, 2019-09-23:
It seems that you need to install some dependencies: `gnat make zlib1g-dev clang llvm-dev`.

dosadi, 2019-09-23:
Well, the only one of those I hadn't installed was gnat, so I went ahead and installed it, and I still see what I presume is the same error, but here is the log from 'make':

$ make
if test -d ./.git \
   && desc=`cd .; git describe --dirty`; then \
          sub="s/[(].*[)]/($desc)/"; \
else \
  sub="s/tarball/tarball/"; \
        fi; \
sed -e "$sub" -e "s/@VER@/0.37-dev/" < src/version.in > version.tmp; \

if [ ! -r version.ads ] || ! cmp version.tmp version.ads > /dev/null; then cp version.tmp version.ads; fi
make -f ./src/ortho/llvm4-nodebug/Makefile \
 ortho_srcdir=./src/ortho ortho_exec=ghdl1-llvm \
 GNATFLAGS="-aI./src -aI./src/vhdl -aI./src/grt -aI./src/psl -aI./src/vhdl/translate -aI./src/ghdldrv -aI./src/ortho -aI./src/ortho/llvm4-nodebug -aI./src/synth -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe" LDFLAGS="" \
 LLVM_CONFIG="llvm-config" CXX="clang++" GNATMAKE="gnatmake" \
 all
make[1]: Entering directory '/home/jon/Downloads/ghdl-master'
clang++ -c `llvm-config --cxxflags` -o llvm-cbindings.o src/ortho/llvm4-nodebug/llvm-cbindings.cpp
In file included from src/ortho/llvm4-nodebug/llvm-cbindings.cpp:18:
In file included from /usr/lib/llvm-8/include/llvm-c/Target.h:22:
In file included from /usr/lib/llvm-8/include/llvm-c/Types.h:17:
/usr/lib/llvm-8/include/llvm-c/DataTypes.h:28:10: fatal error: 'cmath' file not found
#include <cmath>
         ^~~~~~~
1 error generated.
make[1]: *** [src/ortho/llvm4-nodebug/Makefile:18: llvm-cbindings.o] Error 1
make[1]: Leaving directory '/home/jon/Downloads/ghdl-master'
make: *** [Makefile:335: ghdl1-llvm] Error 2

1138-4EB, 2019-09-23:
Unfortunately, I cannot reproduce it in a docker container:

```sh
docker run --rm -t ubuntu:19.04 bash -c "
apt update -qq
apt install -y gnat make zlib1g-dev clang llvm-dev git
git clone https://github.com/ghdl/ghdl
cd ghdl
./configure --with-llvm-config --prefix=/opt/ghdl
make
make install
/opt/ghdl/bin/ghdl --version
"

...
grep -h "constant Module_Id :=" src/synth/netlists.ads src/synth/netlists-gates.ads | sed -e '/constant Module_Id :=/s/:.*://' -e 's/;/,/' -e 's/ *--.*//' >> src/synth/ghdlsynth_gates.h
echo "};" >> src/synth/ghdlsynth_gates.h
install -m 644 -p ./src/synth/ghdlsynth.h /opt/ghdl/include/
install -m 644 -p ./src/synth/ghdlsynth_gates.h /opt/ghdl/include/
/opt/ghdl/bin/ghdl --disp-standard --std=87 > /opt/ghdl/lib/ghdl/src/std/standard.v87
/opt/ghdl/bin/ghdl --disp-standard --std=93 > /opt/ghdl/lib/ghdl/src/std/standard.v93
/opt/ghdl/bin/ghdl --disp-standard --std=08 > /opt/ghdl/lib/ghdl/src/std/standard.v08
GHDL 0.37-dev (v0.36-782-g06d92830) [Dunoon edition]
 Compiled with GNAT Version: 8.3.0
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2019 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Fortunately, you can use the procedure above to build ghdl and then copy `/opt/ghdl` from the container to your host.

tgingold, 2019-09-23:
cmath is a standard header.  Your clang++ compiler is broken.  I don't know which distribution you are using, but maybe one dependency is missing.

dosadi, 2019-09-26:
Sorry about the delay, It took me a little while to figure out what to do... I ended up having to install the gcc 9 version of the libstdc++-dev packages, even though both GCC and LLVM install on my box are version 8.  WIerd, but I got the code to build and install a binary in /opt, like the ticket wanted....

closed 2019-09-26 by dosadi