ghdl/ghdl #938: synth: add support for latches

Xiretza, 2019-09-23:
**Description**
Latches are usually not what you want in your design, but sometimes they are necessary afterall. Currently, ghdl crashes when trying to synthesize one.

**Expected behaviour**
GHDL synthesizes a latch as it is described.

**Context**
```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-798-g4ec17bb5) [Dunoon edition]
Compiled with GNAT Version: 9.1.0
Target: x86_64-pc-linux-gnu
/tmp/ghdlsynth_playground/
Command line:
ghdl --synth rs_latch.vhd -e ent
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : synth-inference.adb:448
Call stack traceback locations:
0x558ad8369cf9 0x558ad8369e26 0x558ad836cef6 0x558ad840ef6e 0x558ad84063a2 0x558ad840644f 0x558ad8402c4c 0x558ad8460ea8 0x558ad83c4bcd 0x558ad846b001 0x558ad82ebccc 0x7fc512587ee1 0x558ad82eac7c 0xfffffffffffffffe
******************************************************************
```

- OS: Arch Linux
- Origin:
  - [ ] Package manager. Repo:
  - [ ] Released binaries. Tarball:
  - [x] Built from sources. Commit SHA: 4ec17bb5

**How to reproduce?**

```
#>> ent.vhd
entity ent is
    port (
        r : in bit;
        s : in bit;
        q : out bit
    );
end entity;

architecture a of ent is
begin
    process(r, s)
    begin
        if r = '1' then
            q <= '0';
        elsif s = '1' then
            q <= '1';
        end if;
    end process;
end;

#>> sim.sh
ghdl --synth ent.vhd -e ent

#>> run.sh
docker run --rm -tv /$(pwd):/src:z -w //src ghdl/ghdl:buster-mcode sh -c ./sim.sh

#>> end
```

**Checklist**
Before submitting your issue, please review the following checklist:

- [x] Add `GHDL Bug occurred` log block
- [x] Add a MWE
- [x] Try the latest version

tgingold, 2019-09-27:
It will be handled (in the future) but that's not the highest priority.

rodrigomelo9, 2020-05-18:
FYI @tgingold 

Hi. I reported something related [here](https://github.com/ghdl/ghdl-yosys-plugin/issues/99), which was closed. With previous versions, when a latch was detected, the synthesis aborts with an error:
```
$ ghdl -a examples/ise/HDL_Coding_Techniques/latches/latches_0.vhd
$ ghdl --synth latches_0
examples/ise/HDL_Coding_Techniques/latches/latches_0.vhd:18:5:error: latch infered for net "Q"
```
I check with the last version in `ghdl/synth:beta`, and now the word `error` do not appears, but is still there:
```
$ bash ../helpers/docker_run.sh ghdl --synth latches_0
examples/ise/HDL_Coding_Techniques/latches/latches_0.vhd:18:5: latch infered for net "Q"
ram@ram-notebook2:~/repos/yosys-versus/xilinx$ echo $?
1
```
I don't know if was the idea to solve this issue or is an intermediate state. Regards

tgingold, 2020-05-19:
I have closed your issue as it is a duplicate of this one.
Latch are still not yet supported!

rodrigomelo9, 2020-05-19:
It is ok about the closed issue. Here I only remark, that is still an error (exit code 1) but the word "error" disappears,  which could be confusing.

tgingold, 2020-05-20:
The presence of 'error' depends on whether colors are used or not.

stnolting, 2021-12-06:
Hey @tgingold!
I wonder if latches are "officially" supported now?

I have setup a design that uses latches and comb. loops to implement a true random number generator (see https://github.com/stnolting/neorv32/pull/226). The bitstream works as expected but I have not further investigated on the actual synthesis results yet.

commit 88425cb3 (2021-12-06): synth: add --latches option to enable latches.  Fix #938
closed 2021-12-06 by tgingold