ghdl/ghdl #943: Synopsys conv_integer argument too large

letFunny, 2019-09-25:
Long story short, we have to use synopsys and conv_integer for my school and I cannot change it. We use ModelSim and the code works. However, in ghdl it gives:
`../../src/synopsys/std_logic_arith.vhdl:2014:8:@0ms:(assertion failure): ARG is too large in CONV_INTEGER`
When I print the integer in quesiton in a report it is just 32 zero bits.
```vhd
 readline (bin_file, current_line);
         hread(current_line, address);
         hread(current_line, datum);
         -- TODO he cambiado esto porque si no daba error no sé por qué
         assert CONV_INTEGER(address(31 downto 0))<MEM_SIZE
```
The file that it reads is [here](https://pastebin.com/479JpGC5)

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a ent.vhd
ghdl --elab-run ent
```

**Context**
Please, provide the following information:

- OS: Linux 5.3.1 (Arch Linux)
- Origin:
  - [ ] Package manager: `0.36`

Sorry if any information is missing, this is my first bug request. Thanks for the inconvenience.

1138-4EB, 2019-09-25:
Hola! Welcome and thanks for reporting the issue. Some thoughts that might help triage this issue:

- Please, provide a minimal working example. I.e., add the entity/architecture around the code block you posted, and provide also the commands you are using to execute ghdl. For example, this is some incomplete example that I just made up:

```vhd
entity myent is
end;

architecture arch of myent is
const bin_file: string := "my_bin_file.bin";
begin
process() begin
 readline (bin_file, current_line);
         hread(current_line, address);
         hread(current_line, datum);
         -- TODO he cambiado esto porque si no daba error no sé por qué
         assert CONV_INTEGER(address(31 downto 0))<MEM_SIZE
end process;
end;
```

``` sh
ghdl -a --std=02 myfile.vhd
ghdl -e --std=02 myfile.vhd
```

This is because the issue might be related to the CLI options you use, and not only to the VHDL code.

- Note that you can drag & drop files to a GitHub message/comment, in order to upload them. This might be easier to handle than pastebin or other external sites.

- The comment in spanish might be translated to *i changed this because it would fail otherwise (i don't know why)*. Does the comment mean that you were using some other code and you changed to this for it to work on ModelSim? Then, you tried with GHDL. If this is the case, I think it'd be helpful if you provided/tried the previous version of the code too.

tgingold, 2019-09-25:
GHDL uses the `std_logic_arith` package as defined by synopsys and the first line of conv_integer is:

```
	assert ARG'length <= 31
	    report "ARG is too large in CONV_INTEGER"
	    severity FAILURE;
```

Hence the failure.

letFunny, 2019-09-25:
First of all, sorry. When creating the issue I was in a hurry to get it to work so I did not fill the necessary details. The code I am running now is the following:

```vhd
variable address : std_logic_vector(31 downto 0);
file bin_file : text open READ_MODE is INIT_FILENAME;
```

```vhd
      while (not endfile (bin_file)) loop
         readline (bin_file, current_line);
         hread(current_line, address);
         hread(current_line, datum);
         assert CONV_INTEGER(signed(address))<MEM_SIZE
            report "Direccion fuera de rango en el fichero de la memoria"
            severity failure;
         memo( CONV_INTEGER( address(31 downto 2) ) ) <= datum;
      end loop;
```

And compilation:
```sh
ghdl -i --workdir=work --ieee=synopsys --std=08 -fexplicit rtl/*.vhd sim/*.vhd
ghdl -m --workdir=work --ieee=synopsys --std=08 -fexplicit processor_tb
```

I have looked into the source code provided in ghdl and found the snippet you included. The issue is that my vector is 32 bits long and for unsigned it only accepts 31 bit long vectors. I have solved it by adding `signed(address)` instead, as you can see.

Now it works. However the "bug" prevails because when I run the original code in ModelSim it works but in ghdl it does not. Somehow ModelSim uses the signed variant and ghdl uses the unsigned one.

Thanks in advance.

Note: The comment in Spanish did not mean anything important. It was just a placeholder because when I change `vector` to `vector(30 downto 0)` it worked. I meant to delete the comment as it was not working but I forgot. Anyway, good translation skills.

1138-4EB, 2019-09-25:
> First of all, sorry. When creating the issue I was in a hurry to get it to work so I did not fill the necessary details.

No need to apologize. As you see, @tgingold was able to provide an answer easily. It is just good practice to provide a MWE for more complex/tricky issue. Sometimes we cannot, for whatever reason, and it is ok too.

> Note: The comment in Spanish did not mean anything important. It was just a placeholder because when I change `vector` to `vector(30 downto 0)` it worked. I meant to delete the comment as it was not working but I forgot.

Once again, it's ok. I didn't mean to criticize it. I translated it quickly to help other users, but I was interested in the actual code you had changed. I.e., this explanation you provided now.

> Anyway, good translation skills.

I'm native ;)

tgingold, 2019-09-26:
> Now it works. However the "bug" prevails because when I run the original code in ModelSim it works but in ghdl it does not. Somehow ModelSim uses the signed variant and ghdl uses the unsigned one.

That's why we need a reproducer as the details are important.

tgingold, 2019-10-01:
I am closing this issue because I don't think there is a bug in ghdl.  Feel free to reopen it with a complete reproducer.

closed 2019-10-01 by tgingold