ghdl/ghdl #949: Cannot build in OSX

anquegi, 2019-09-30:
**Description**
When I tried to build in OSX I get the following:
```
ghdl on  master [?]
➜ ./configure --prefix=/usr/local
Use full IEEE library
Build machine is: x86_64-apple-darwin17.7.0
Creating ghdl.gpr
Creating Makefile
for d in src/std src/std/v87 src/std/v93 src/std/v08 src/synopsys src/synopsys/v08 std/v87 ieee/v87 synopsys/v87 std/v93 ieee/v93 synopsys/v93 std/v08 ieee/v08 src/openieee src/ieee src/vital95 src/vital2000 src/mentor src/ieee/v87 src/ieee/v93 src/ieee2008 mentor/v93 synopsys/v08; do \
	  mkdir -p lib/ghdl/$d; \
	done
Generate ortho_code-x86-flags.ads
Generate elf_arch.ads
Generate ghdlsynth_maybe.ads
Generate default_paths.ads
```

```
make
gcc -c -g -DWITH_GNAT_RUN_TIME -o jumps.o /Users/toni/learn/vhdl/ghdl/./src/grt/config/jumps.c
/Users/toni/learn/vhdl/ghdl/./src/grt/config/jumps.c:29:10: fatal error: signal.h: No such file or directory
 #include <signal.h>
          ^~~~~~~~~~
compilation terminated.
make: *** [jumps.o] Error 1
```

**Expected behaviour**
to be build correctly
**How to reproduce?**
I directly clone the repo and folllow the instructions

**Context**
Please, provide the following information:
```

uname -a
Darwin MacBook-Pro-de-Antonio.local 18.7.0 Darwin Kernel Version 18.7.0: Tue Aug 20 16:57:14 PDT 2019; root:xnu-4903.271.2~2/RELEASE_X86_64 x86_64
```

I have xcode installed 
and 
`open /Library/Developer/CommandLineTools/Packages/macOS_SDK_headers_for_macOS_10.14.pkg`

tgingold, 2019-09-30:
You need to install the C include files from Xcode.  I don't know the procedure but it should be documented with the compiler you have installed.

anquegi, 2019-09-30:
Hi thanks, I finally managed to get this working following  https://stackoverflow.com/questions/52509602/cant-compile-c-program-on-a-mac-after-upgrade-to-mojave and https://github.com/nodejs/node-gyp/issues/569, but now I get another error:

```
make
gcc -c -g -DWITH_GNAT_RUN_TIME -o lz4.o /Users/toni/learn/vhdl/ghdl/./src/grt/fst/lz4.c
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/wait.h:110,
                 from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/stdlib.h:66,
                 from /Users/toni/learn/vhdl/ghdl/./src/grt/fst/lz4.c:105:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/resource.h: In function 'getiopolicy_np':
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/resource.h:443:34: error: expected declaration specifiers before '__OSX_AVAILABLE_STARTING'
 int     getiopolicy_np(int, int) __OSX_AVAILABLE_STARTING(__MAC_10_5, __IPHONE_2_0);
                                  ^~~~~~~~~~~~~~~~~~~~~~~~
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/resource.h:449:39: error: expected '=', ',', ';', 'asm' or '__attribute__' before '__OSX_AVAILABLE_STARTING'
 int     setiopolicy_np(int, int, int) __OSX_AVAILABLE_STARTING(__MAC_10_5, __IPHONE_2_0);
                                       ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/libkern/_OSByteOrder.h:66,
                 from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/_endian.h:130,
                 from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/i386/endian.h:99,
                 from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/machine/endian.h:35,
                 from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/wait.h:186,
                 from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/stdlib.h:66,
                 from /Users/toni/learn/vhdl/ghdl/./src/grt/fst/lz4.c:105:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/libkern/i386/_OSByteOrder.h:49:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
 {
 ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/libkern/i386/_OSByteOrder.h:58:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
 {
 ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/libkern/i386/_OSByteOrder.h:96:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
 {
 ^
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/stdlib.h:77,
                 from /Users/toni/learn/vhdl/ghdl/./src/grt/fst/lz4.c:105:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/_types/_ct_rune_t.h:32:28: error: storage class specified for parameter 'ct_rune_t'
 typedef __darwin_ct_rune_t ct_rune_t;
                            ^~~~~~~~~
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/stdlib.h:78,
                 from /Users/toni/learn/vhdl/ghdl/./src/grt/fst/lz4.c:105:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/_types/_rune_t.h:31:25: error: storage class specified for parameter 'rune_t'
 typedef __darwin_rune_t rune_t;
                         ^~~~~~
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/stdlib.h:81,
                 from /Users/toni/learn/vhdl/ghdl/./src/grt/fst/lz4.c:105:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/_types/_wchar_t.h:34:26: error: storage class specified for parameter 'wchar_t'
 typedef __darwin_wchar_t wchar_t;
                          ^~~~~~~
In file included from /Users/toni/learn/vhdl/ghdl/./src/grt/fst/lz4.c:105:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/stdlib.h:86:3: error: storage class specified for parameter 'div_t'
 } div_t;
   ^~~~~
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/stdlib.h:91:3: error: storage class specified for parameter 'ldiv_t'
 } ldiv_t;
   ^~~~~~
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/stdlib.h:97:3: error: storage class specified for parameter 'lldiv_t'
 } lldiv_t;
   ^~~~~~~
In file included from /Users/toni/learn/vhdl/ghdl/./src/grt/fst/lz4.c:105:
[... cut ...]
 int     getiopolicy_np(int, int) __OSX_AVAILABLE_STARTING(__MAC_10_5, __IPHONE_2_0);
                        ^~~
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/resource.h:443:29: error: parameter name omitted
 int     getiopolicy_np(int, int) __OSX_AVAILABLE_STARTING(__MAC_10_5, __IPHONE_2_0);
                             ^~~
/Users/toni/learn/vhdl/ghdl/./src/grt/fst/lz4.c:1516: error: expected '{' at end of input

make: *** [lz4.o] Error 1
```

1138-4EB, 2019-09-30:
@anquegi, might this help? https://github.com/mattn/go-sqlite3/issues/653#issuecomment-433249471

tgingold, 2019-10-01:
Or use clang to compile C files!

anquegi, 2019-10-01:
> Or use clang to compile C files!

1138-4EB, 2019-10-01:
@anquegi, I think that @tgingold meant to use `--with-llvm-config`. See https://ghdl.readthedocs.io/en/latest/getting/LLVM.html#build-llvm.

anquegi, 2019-10-01:
Thanks for your reply, at least some advance I tried to use llvm installed with homebrew, I changed to this:

```
../configure --with-llvm-config --prefix=7urs/local
Use full IEEE library
Build machine is: x86_64-apple-darwin17.7.0
cannot run llvm-config
```

So after reading this  [](https://stackoverflow.com/questions/32959627/where-is-the-system-llvm-config-on-macos/35753922#35753922) I started to use the llvm I installed with homebrew:

using 

```
➜ $(brew --prefix llvm)/bin/llvm-config --version
8.0.0
```

and change CC=gcc to CC=clang in Makefile, it started working, bug fails again:

```
../configure --prefix=/usr/local --with-llvm-config=$(brew --prefix llvm)/bin/llvm-config
Use full IEEE library
Build machine is: x86_64-apple-darwin17.7.0
Debugging is not enabled with llvm 8.0.0
Creating ghdl.gpr
Creating Makefile
for d in src/std src/std/v87 src/std/v93 src/std/v08 src/synopsys src/synopsys/v08 std/v87 ieee/v87 synopsys/v87 std/v93 ieee/v93 synopsys/v93 std/v08 ieee/v08 src/openieee src/ieee src/vital95 src/vital2000 src/mentor src/ieee/v87 src/ieee/v93 src/ieee2008 mentor/v93 synopsys/v08; do \
	  mkdir -p lib/ghdl/$d; \
	done
Generate ghdlsynth_maybe.ads
Generate default_paths.ads
```

Then it seams to work but:

make >correct.txt 2>error.txt

```
cat correct.txt
if test -d ../.git \
	   && desc=`cd ..; git describe --dirty`; then \
          sub="s/[(].*[)]/($desc)/"; \
	else \
	  sub="s/tarball/tarball/"; \
        fi; \
	sed -e "$sub" -e "s/@VER@/0.37-dev/" < ../src/version.in > version.tmp; \

if [ ! -r version.ads ] || ! cmp version.tmp version.ads > /dev/null; then cp version.tmp version.ads; fi
make -f ../src/ortho/llvm4-nodebug/Makefile \
	 ortho_srcdir=../src/ortho ortho_exec=ghdl1-llvm \
	 GNATFLAGS="-aI../src -aI../src/vhdl -aI../src/grt -aI../src/psl -aI../src/vhdl/translate -aI../src/ghdldrv -aI../src/ortho -aI../src/ortho/llvm4-nodebug -aI../src/synth -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe" LDFLAGS="-Wl,-no_compact_unwind" \
	 LLVM_CONFIG="/usr/local/opt/llvm/bin/llvm-config" CXX="clang++" GNATMAKE="gnatmake" \
	 all
gnatmake -o ghdl1-llvm -aI../src/ortho/llvm4-nodebug -aI../src/ortho \
	-aI../src -aI../src/vhdl -aI../src/grt -aI../src/psl -aI../src/vhdl/translate -aI../src/ghdldrv -aI../src/ortho -aI../src/ortho/llvm4-nodebug -aI../src/synth -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe ortho_code_main -bargs -E \
	-largs llvm-cbindings.o --LINK=clang++ \
	-Wl,-no_compact_unwind `/usr/local/opt/llvm/bin/llvm-config --ldflags --libs --system-libs`
gnatmake -aI../src -aI../src/vhdl -aI../src/grt -aI../src/psl -aI../src/vhdl/translate -aI../src/ghdldrv -aI../src/ortho -aI../src/ortho/llvm4-nodebug -aI../src/synth -aI../src/ghdldrv -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe\
	 ghdl_llvm -bargs -E -largs -Wl,-no_compact_unwind
gcc -c -g -o lz4.o /Users/toni/learn/vhdl/ghdl/build/../src/grt/fst/lz4.c
```

```
cat error.txt
gnatmake: "ghdl1-llvm" up to date.
gnatmake: "ghdl_llvm" up to date.
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/wait.h:110,
                 from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/stdlib.h:66,
                 from /Users/toni/learn/vhdl/ghdl/build/../src/grt/fst/lz4.c:105:
[... 461 lines cut ...]
/Users/toni/learn/vhdl/ghdl/build/../src/grt/fst/lz4.c:1497:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
 {
 ^
/Users/toni/learn/vhdl/ghdl/build/../src/grt/fst/lz4.c:1506:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
 {
 ^
/Users/toni/learn/vhdl/ghdl/build/../src/grt/fst/lz4.c:1511:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
 {
 ^
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/wait.h:110,
                 from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/stdlib.h:66,
                 from /Users/toni/learn/vhdl/ghdl/build/../src/grt/fst/lz4.c:105:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/resource.h:443:9: error: old-style parameter declarations in prototyped function definition
 int     getiopolicy_np(int, int) __OSX_AVAILABLE_STARTING(__MAC_10_5, __IPHONE_2_0);
         ^~~~~~~~~~~~~~
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/resource.h:443:24: error: parameter name omitted
 int     getiopolicy_np(int, int) __OSX_AVAILABLE_STARTING(__MAC_10_5, __IPHONE_2_0);
                        ^~~
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/resource.h:443:29: error: parameter name omitted
 int     getiopolicy_np(int, int) __OSX_AVAILABLE_STARTING(__MAC_10_5, __IPHONE_2_0);
                             ^~~
/Users/toni/learn/vhdl/ghdl/build/../src/grt/fst/lz4.c:1516: error: expected '{' at end of input

make: *** [lz4.o] Error 1
```

anquegi, 2019-10-01:
Finally I could build it just for the sake of completion I would add here the steps, and thanks for helping me. with your tips.

First of all I reclone de ghdl project and upgrade de llvm from hombre with brew upgrade llvm

then:

```
mkdir build
cd build
../configure --prefix=/usr/local --with-llvm-config=$(brew --prefix llvm)/bin/llvm-config
```

after that I made two changes in the generated Makefile

`CC=clang`

and from this: 

```
brew info llvm

....

For compilers to find llvm you may need to set:
  export LDFLAGS="-L/usr/local/opt/llvm/lib"
  export CPPFLAGS="-I/usr/local/opt/llvm/include"
....
```
`
LDFLAGS= -L/usr/local/opt/llvm/lib,-Wl,-no_compact_unwind`

save the file and 

`make`

```
./ghdl_llvm --version
GHDL 0.37-dev (v0.36-878-g2388da39) [Dunoon edition]
 Compiled with GNAT Version: Community 2019 (20190517-83)
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2019 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

closed 2019-10-01 by anquegi

1138-4EB, 2019-10-01:
Good! And thanks for explaining the details. Just in case it can help me fix #867, would you mind telling how did you install GNAT? Was it graphically?

anquegi, 2019-10-01:
Yes I installed it graphically, never used, only installed for this purpose.

I installed this binary

```
➜ ls gnat-community-2019-20190517-x86_64-darwin-bin.dmg
➜ which gnatmake
/Users/toni/opt/GNAT/2019/bin/gnatmake
```

following the instructions from [here](https://www.radford.edu/~nokie/classes/320/compileInstall.html) in the section  Installing the AdaCore GNAT compiler on Mac, only click next and finish and add the PATH to my .zshrc