ghdl/ghdl #950: 64 bit integer support

antonblanchard, 2019-09-30:
VHDL-2019 apparently adds 64 bit integers. This is something I struggled with a bit when first learning VHDL and writing test benches, and I think adding it would be useful.

tgingold, 2019-09-30:
It should be possible to add it as a build time option.  I prefer not to have it by default because I fear that the performance impacts will be high on 32 bit machines.

But why do you need it ?  64 bit integers won't provide 64 bit unsigned numbers.

Paebbels, 2025-03-06:
@tgingold can you implement that VHDL-2019 feature? We run into problems with GHDL supporting only 31 bits (effectively) for integers when running functional coverage with OSVVM. We have a coverage model collecting memory addresses for the AXI network on an AMD MPSoC.

This use model is already supported by Riviera-PRO.

> I prefer not to have it by default because I fear that the performance impacts will be high on 32 bit machines.

No one should use 32-bit machines to run VHDL simulations. High-end modern servers are needed with lots of single-thread performance and high over-boost factors.

-----
/cc @JimLewis

tgingold, 2025-03-15:
There is the `-finteger64` flag, but I have to check if it is fully supported.

Paebbels, 2025-03-15:
Can this flag be activated automatically by `--std=19`?