ghdl/ghdl #951: synth: wrongly recognized as latch due to order of "if" predicates

Xiretza, 2019-09-30:
**Description**
Latches aren't supported yet (see #938), so trying to create one results in a crash - however the below example should never result in a latch, since everything is guarded in a `rising_edge(clk)`. Now honestly I'm not 100% how the standard works here, maybe the `'event` actually has to be the first thing in an `if`, in that case it's not a bug.

**Expected behaviour**
The order of predicates in an `if` statement should not have an impact on synthesis results.

**How to reproduce?**
```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all;

entity ent is
    port (
        clk : in std_logic;
        enable : in std_logic;
        i : in std_logic;
        o : out std_logic
    );
end;

architecture a of ent is
begin
    process(i)
    begin
        -- works:
        --if rising_edge(clk) and enable = '1' then
        if enable = '1' and rising_edge(clk) then
            o <= i;
        end if;
    end process;
end;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl --synth ent.vhd -e ent
```

**Context**
Please, provide the following information:

- OS: Arch Linux
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: aefa14ad

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

(This is the same bug check as would be expected for any latch while it's not supported yet.)

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-852-gaefa14ad) [Dunoon edition]
Compiled with GNAT Version: 9.1.0
Target: x86_64-pc-linux-gnu
/home/xiretza/temp/ghdl_synth_tests/
Command line:
ghdl --synth if_order.vhd -e ent
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : synth-inference.adb:448
Call stack traceback locations:
0x55dac55ea139 0x55dac55ea266 0x55dac55ed7b6 0x55dac5692c34 0x55dac56883b6 0x55dac5688463 0x55dac5684bb6 0x55dac56e4c48 0x55dac56467fd 0x55dac56eeda1 0x55dac556bccc 0x7f35f9239ee1 0x55dac556ac7c 0xfffffffffffffffe
******************************************************************
```

1138-4EB, 2019-09-30:
@Xiretza, in this kind of processes, I normally set the sensitivity list to `clk` only. If I want a purely synchronous register, I think that no other signal should trigger the process. I'd say that the order of the predicates would be negligible in that case. Could you please try it? Also, please let me know, should you think that my approach is not correct.

Nevertheless, I'll mark this as a bug, because GHDL should not crash.

Xiretza, 2019-09-30:
Right, not sure where I copy-pasted that MWE from - locally, my sensitivity list only contains `clk`, as it should. Doesn't make a difference though.

```vhdl
architecture a of ent is
begin
    process(clk)
    begin
        -- works:
        --if rising_edge(clk) and enable = '1' then
        if enable = '1' and rising_edge(clk) then
            o <= i;
        end if;
    end process;
end;
```

The commented-out line works, the second one results in the crash associated with unimplemented latches.

commit 91676ae0 (2019-09-30): synth-inference: handl 'EXPR and CLK' condition.  Fix #951
closed 2019-09-30 by tgingold