ghdl/ghdl #953: synth: support multiplication and division (at least by powers of 2)

Xiretza, 2019-09-30:
**Description**
I know there was some discussion about supporting generic multiplication and division somewhere before, but can't remember the outcome. Either way, mul/div by a power of 2 should be easy to implement, since it's just adding or removing bits.

**Expected behaviour**
Multiplication or division by a constant 2^n should have the same result as shifting the number left or right by n.

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all,
    ieee.numeric_std.all;

entity ent is
end;

architecture a of ent is
    signal x : unsigned(7 downto 0);
    signal y : unsigned(7 downto 0) := x / 2;
    signal z : unsigned(7 downto 0) := x * 2;
begin
end;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl --synth ent.vhd -e ent
```

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: aefa14ad

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

For division:

```
clean_mul_div.vhd:10:45:error: unhandled predefined IEEE operator "/"
../../src/ieee/v93/numeric_std.vhdl:207:12:error:  declared here

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-852-gaefa14ad) [Dunoon edition]
Compiled with GNAT Version: 9.1.0
Target: x86_64-pc-linux-gnu
/home/xiretza/temp/ghdl_synth_tests/
Command line:
ghdl --synth clean_mul_div.vhd -e ent
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : synth-expr.adb:1498
Call stack traceback locations:
0x560c213d8244 0x560c213da262 0x560c213da534 0x560c213c039c 0x560c213c0463 0x560c213bcbb6 0x560c2141cc48 0x560c2137e7fd 0x560c21426da1 0x560c212a3ccc 0x7fca493afee1 0x560c212a2c7c 0xfffffffffffffffe
******************************************************************
```

For multiplication:

```
clean_mul_div.vhd:11:45:error: synth_dyadic_operation: unhandled IIR_PREDEFINED_IEEE_NUMERIC_STD_MUL_UNS_NAT

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-852-gaefa14ad) [Dunoon edition]
Compiled with GNAT Version: 9.1.0
Target: x86_64-pc-linux-gnu
/home/xiretza/temp/ghdl_synth_tests/
Command line:
ghdl --synth clean_mul_div.vhd -e ent
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : synth-oper.adb:709
Call stack traceback locations:
0x55ee4ad39c83 0x55ee4ad441e8 0x55ee4ad46262 0x55ee4ad46534 0x55ee4ad2c39c 0x55ee4ad2c463 0x55ee4ad28bb6 0x55ee4ad88c48 0x55ee4acea7fd 0x55ee4ad92da1 0x55ee4ac0fccc 0x7f17ce880ee1 0x55ee4ac0ec7c 0xfffffffffffffffe
******************************************************************
```

1138-4EB, 2019-09-30:
> I know there was some discussion about supporting generic multiplication and division somewhere before, but can't remember the outcome.

FTR, tgingold/ghdlsynth-beta#38.

tgingold, 2019-09-30:
I am pretty sure there will be a follow-up....

commit 9aeab448 (2019-09-30): synth: improve support of * and /.  Fix #953
closed 2019-09-30 by tgingold