ghdl/ghdl #957: How to generate edif file.

m-kru, 2019-09-30:
On GHDL documentation website it is written 

> Although synthesis is not supported yet, there is some experimental feature to generate RTL netlists (VHDL or EDIF) from synthesisable code.

How to generate EDIF RTL netlist? By default `--synth` outputs VHDL format.

1138-4EB, 2019-09-30:
There is no such feature yet. EDIF is one of the formats that was considered at first. When the comment in the docs was written some weeks ago, `ghdl --synth` did not produce any usable output (not even the current VHDL format). Note that this is in a highly experimental and *broken* stage. Work is being focused on supporting common language features that produce a crash: https://github.com/ghdl/ghdl/issues?q=is%3Aissue+is%3Aopen+sort%3Aupdated-desc+label%3A%22FeaReq%3A+Synthesis%22

When enough language features are supported (i.e. when projects such as microwatt #903 can be synthesised), I believe it will be possible to have different output formats. However, just for the sake of discussion, why do you need/want/prefer EDIF instead of the current VHDL output?

m-kru, 2019-09-30:
Some of commercial tools are capable of detecting possible CDC, RDC problems without any simulations. I was just wondering how hard it would be to implement tool with similar functionality. I was just curious how edif format looks like in practice. I thought that maybe it would be easier to parse edif instead of VHDLs for this purpose.

1138-4EB, 2019-09-30:
Thanks a lot for the explanation. I believe that those are both (CDC/RDC and parsing) fair points.

m-kru, 2019-09-30:
Do have any experience with this? Do you know what might be the starting point for such tools? I doubt they operate on raw VHDL.

tmeissner, 2019-09-30:
Yosys has the possibility to write the design as Edif file. So, it should be possible to use the ghdlsynth plugin in Yosys to get edif files.

m-kru, 2019-09-30:
What is actually produced by ghdlsynth, that is provided to the Yosys? ghldsynth documentation is still rather modest?

tmeissner, 2019-09-30:
As far as I understand, ghdlsynth is using the Yosys API to map the synthesized design to Yosys-internal elements.

m-kru, 2019-09-30:
Do I understand correctly. VHDL is synthesized to VHDL to be mapped to Yosys internal elements?

tmeissner, 2019-09-30:
I think, the VHDL output of `ghdl --synth` is for debugging purposes only. ghdlsynth is using the IDs defined in  and tries to map them to elements defined by the Yosys API. For example:

```c++
case Id_Sub:
    module->addSub(to_str(iname), IN(0), IN(1), OUT(0));
    break;
```

addSub() for example is a Yosys API function. But I'm sure, @tgingold can give you a better (and more correct) explanation ;)

tgingold, 2019-10-01:
@tmeissner is correct.

closed 2019-10-01 by tgingold