ghdl/ghdl #959: Undefined symbols for architecture x86_64

anquegi, 2019-10-01:
**Description**
I cannot eleaborate the project.

**Expected behaviour**
to elaborate the project

**How to reproduce?**
I was just following the hello word from [here](https://ghdl.readthedocs.io/en/latest/using/QuickStartGuide.html)

```
--  Hello world program
use std.textio.all; -- Imports the standard textio package.

--  Defines a design entity, without any ports.
entity hello_world is
end hello_world;

architecture behaviour of hello_world is
begin
  process
    variable l : line;
  begin
    write (l, String'("Hello world!"));
    writeline (output, l);
    wait;
  end process;
end behaviour;

```

the ghdl -a work as expected:
```

learn/vhdl took 2s
➜ ghdl -a hello.vhdl

learn/vhdl
➜ ls
hello.o       hello.vhdl    work-obj93.cf
```

but running -e I get

```
ghdl -e hello_world
ld: warning: ignoring file /usr/local/lib/ghdl/libgrt.a, building for macOS-x86_64 but attempting to link with file built for unknown-unsupported file format ( 0x21 0x3C 0x61 0x72 0x63 0x68 0x3E 0x0A 0x2F 0x20 0x20 0x20 0x20 0x20 0x20 0x20 )
Undefined symbols for architecture x86_64:
  "___ghdl_assert_failed", referenced from:
      _std__textio__readline in textio-body.o
      _std__textio__readO2 in textio-body.o
      _std__textio__readO4 in textio-body.o
      _std__textio__readO6 in textio-body.o
      _std__textio__readO8 in textio-body.o
      _std__textio__readO10 in textio-body.o
      _std__textio__readO12 in textio-body.o
      ...
  "___ghdl_bound_check_failed", referenced from:
      _std__standard__bit_vector_AND in std_standard.o
      _std__standard__bit_vector_OR in std_standard.o
      _std__standard__bit_vector_NAND in std_standard.o
      _std__standard__bit_vector_NOR in std_standard.o
      _std__standard__bit_vector_XOR in std_standard.o
      _std__standard__bit_vector_XNOR in std_standard.o
      _std__textio__readline in textio-body.o
      ...
  "___ghdl_check_stack_allocation", referenced from:
      _std__textio__readO3 in textio-body.o
      _std__textio__writeO2 in textio-body.o
  "___ghdl_deallocate", referenced from:
      _std__textio__readline in textio-body.o
      _std__textio__readO9 in textio-body.o
      _std__textio__readO11 in textio-body.o
      _std__textio__readO13 in textio-body.o
      _std__textio__writeline in textio-body.o
      _std__textio__writeO7 in textio-body.o
      _std__textio__trim in textio-body.o
      ...
  "___ghdl_direction_check_failed", referenced from:
      _std__textio__readline in textio-body.o
      _std__textio__readO13 in textio-body.o
      _std__textio__writeO7 in textio-body.o
      _std__textio__trim in textio-body.o
      _std__textio__readO11__set_value in textio-body.o
  "___ghdl_file_endfile", referenced from:
      _std__textio__readline in textio-body.o
  "___ghdl_i64_exp", referenced from:
      _std__textio__readO15 in textio-body.o
  "___ghdl_init_top_generics", referenced from:
      ___ghdl_ELABORATE in e~hello_world.o
  "___ghdl_malloc", referenced from:
      _std__textio__readline in textio-body.o
      _std__textio__readO9 in textio-body.o
      _std__textio__readO11 in textio-body.o
      _std__textio__readO13 in textio-body.o
      _std__textio__writeline in textio-body.o
      _std__textio__writeO7 in textio-body.o
      _std__textio__trim in textio-body.o
      ...
  "___ghdl_malloc0", referenced from:
      ___ghdl_ELABORATE in e~hello_world.o
  "___ghdl_memcpy", referenced from:
      _std__standard__bit_vector_NOT in std_standard.o
      _std__standard__bit_vector_AND in std_standard.o
      _std__standard__bit_vector_OR in std_standard.o
      _std__standard__bit_vector_NAND in std_standard.o
      _std__standard__bit_vector_NOR in std_standard.o
      _std__standard__bit_vector_XOR in std_standard.o
      _std__standard__bit_vector_XNOR in std_standard.o
      ...
  "___ghdl_process_register", referenced from:
      _work__hello_world__ARCH__behaviour__STMT_ELAB in hello.o
  "___ghdl_process_wait_exit", referenced from:
      work__hello_world__ARCH__behaviour__P0( const &&&*) in hello.o
  "___ghdl_program_error", referenced from:
      work__hello_world__ARCH__behaviour__P0( const &&&*) in hello.o
  "___ghdl_rti_add_package", referenced from:
      ___ghdl_ELABORATE in e~hello_world.o
      _std__textio__ELAB_SPEC in textio.o
  "___ghdl_rti_add_top", referenced from:
      ___ghdl_ELABORATE in e~hello_world.o
  "___ghdl_stack2_allocate", referenced from:
      _std__standard__bit_vector_NOT in std_standard.o
      _std__standard__bit_vector_AND in std_standard.o
      _std__standard__bit_vector_OR in std_standard.o
      _std__standard__bit_vector_NAND in std_standard.o
      _std__standard__bit_vector_NOR in std_standard.o
      _std__standard__bit_vector_XOR in std_standard.o
      _std__standard__bit_vector_XNOR in std_standard.o
      ...
  "___ghdl_stack2_mark", referenced from:
      _std__textio__readO1 in textio-body.o
      _std__textio__readO2 in textio-body.o
      _std__textio__readO3 in textio-body.o
      _std__textio__readO4 in textio-body.o
      _std__textio__readO5 in textio-body.o
      _std__textio__readO6 in textio-body.o
      _std__textio__readO7 in textio-body.o
      ...
  "___ghdl_stack2_release", referenced from:
      _std__textio__readO1 in textio-body.o
      _std__textio__readO2 in textio-body.o
      _std__textio__readO3 in textio-body.o
      _std__textio__readO4 in textio-body.o
      _std__textio__readO5 in textio-body.o
      _std__textio__readO6 in textio-body.o
      _std__textio__readO7 in textio-body.o
      ...
  "___ghdl_text_file_elaborate", referenced from:
      _std__textio__ELAB_SPEC in textio.o
  "___ghdl_text_file_open", referenced from:
      _std__textio__ELAB_SPEC in textio.o
  "___ghdl_text_write", referenced from:
      _std__textio__writeline in textio-body.o
  "_main", referenced from:
     implicit entry/start for main executable
  "_std__textio__textio_read_real", referenced from:
      _std__textio__readO11__set_value in textio-body.o
  "_std__textio__textio_write_real", referenced from:
      _std__textio__writeO6 in textio-body.o
  "_std__textio__untruncated_text_read", referenced from:
      _std__textio__readline in textio-body.o
ld: symbol(s) not found for architecture x86_64
collect2: error: ld returned 1 exit status
ghdl:error: compilation error
```

1138-4EB, 2019-10-01:
This seems to be because #949 is not resolved yet. Likely, you have mixed artifacts in your OS, after trying to build and install GHDL multiple times. Now that you guessed how to complete the compilation, can you try setting `--prefix=/opt/ghdl` and then executing `/opt/ghdl/bin/ghdl` directly?

Apart from that, I assume that you are trying to build GHDL from sources because available [releases](https://github.com/ghdl/ghdl/releases) do not fit your host or suit your needs, isn't it?

Also, you might want to extend the conversation to https://gitter.im/ghdl1/Lobby. I'm ok with keeping this issue open, but this seems to be a very specific issue with your setup, since the example is known to work on macOS with multiple backends. Maybe some other macOS users in the chat can help.

diogratia, 2019-10-02:
The libgrt.a warning:
```none
ld: warning: ignoring file /usr/local/lib/ghdl/libgrt.a, ...
```
indicates a tool chain compatibility issue.  Archive `libgrt.a` is built during `make install`:
```none
...
rm -f -f lib/ghdl/libgrt.a
ar rcv lib/ghdl/libgrt.a `sed -e "/^-/d" -e "s!^!grt/!" < grt/grt-files` \
	 jumps.o times.o grt-cbinding.o grt-cvpi.o grt-cdynload.o fstapi.o lz4.o fastlz.o grt/run-bind.o grt/main.o
ranlib lib/ghdl/libgrt.a
...
```
The resulting archive wasn't compatible with whatever `ld` was used during elaboration. 

I'd guess you should either be operating entirely within the Homebrew tool chain or not be dipping a toe into it.

If it were me I'd find a binary package from llvm.org compatible with your 10.14 MacOS to go with the binary distribution for Adacore supplying gnat and not use Homebrew. Just set the execution search path without picking and choosing tools by manual environmental overloads (CC, CXX, LDFLAGS, CPPFLAGS , etc.) or editing makefiles:

```none
Macbook: echo $path
/Users/diogratia/share/clang+llvm-3.8.0-x86_64-apple-darwin/bin /Users/diogratia/share/gnat-gpl-2015-x86_64-darwin-bin/bin /Library/Frameworks/Python.framework/Versions/3.7/bin . /Users/diogratia/bin /usr/bin /usr/sbin /bin /sbin /usr/local/bin /usr/local/sbin /usr/X11/bin/ /usr/libexec
```
($PATH using Bash, would have a ':' separator instead of space.)

On my 10.11.6 OSX El Capitan platform clang++ for linking is from the llvm binary distribution, gcc and gnatmake are from the Adacore binary gnat distribution, where that gcc is used to build the C files you were having trouble with earlier, e.g.:
```none
gcc -c -g -o fstapi.o /Users/diogratia/Desktop/ghdl/ghdl-0.37DEV/5738c8a7/ghdl/build/../src/grt/fst/fstapi.c
```
`ld` and `ar` are found in `/usr/bin`.

anquegi, 2019-10-02:
Thanks for your replies is a problem of my system build. It shpould be abouy mixing the native llvm and the llvm installed with hombrew. 
I cannot see any binaries comparible from llvm.org. [downloads](http://releases.llvm.org/download.html#9.0.0). So I need the ones with homebrew. 
The problem is as suggested by @diogratia that I use different tools in each step.
I will try to build it correctly in OSX. 
I finally get it working with the easy way of intalling the 0.36 Release binary from ghdl. 
What shoul be better to keep this open or get all the thinngs to the other #949 issue?

diogratia, 2019-10-02:
I wanted to see how close I could get to testing the anquegi's hypothesis

_I cannot see any binaries comparible from llvm.org. downloads. So I need the ones with homebrew.
The problem is as suggested by @diogratia that I use different tools in each step.
I will try to build it correctly in OSX.._

Building from commit a462aa24 (today) using 
```none
alias nghdl_path 'set path=($HOME/share/clang+llvm-8.0.0-x86_64-apple-darwin/bin $HOME/share/gnat-gpl-2017-x86_64-darwin-bin/bin $path)'
```
clang+llvm-8.0.0 from llvm.org/downloads and gnat-gpl-2017 from Adacore with
```none
../configure --with-llvm-config=/Users/david_koontz/share/clang+llvm-8.0.0-x86_64-apple-darwin/bin/llvm-config
```
(in a build subdirectory

The build completed.

The testsuite errors on synth:

```none
...
Vests is OK
   22595 vests.log
[GHDL - test] synth

dir aggr01:
analyze aggr01.vhdl tb_aggr01.vhdl
elaborate and simulate tb_aggr01
Remove work library
Synthesis of aggr01.vhdl -e aggr01
/Users/david_koontz/Desktop/ghdl/ghdl-0.37dev/4e000b81/ghdl/build/ghdl_llvm: unknown command '--synth', try --help
#################################################################
######### FAILURE: aggr01
#################################################################
```
(meaning I need to figure out how not to run synth tests)

and the installed ghdl

```none
ghdl --version
GHDL 0.37-dev (v0.36-892-ga462aa2) [Dunoon edition]
 Compiled with GNAT Version: GPL 2017 (20170515-63)
 llvm code generator
Written by Tristan Gingold.

/usr/local/bin/ghdl -a   sbox1.vhdl
/usr/local/bin/ghdl -a   sbox2.vhdl
/usr/local/bin/ghdl -a   sbox3.vhdl
/usr/local/bin/ghdl -a   sbox4.vhdl
/usr/local/bin/ghdl -a   sbox5.vhdl
/usr/local/bin/ghdl -a   sbox6.vhdl
/usr/local/bin/ghdl -a   sbox7.vhdl
/usr/local/bin/ghdl -a   sbox8.vhdl
/usr/local/bin/ghdl -a   frk.vhdl
/usr/local/bin/ghdl -a   pipe_stage.vhdl
/usr/local/bin/ghdl -a   last_stage.vhdl
/usr/local/bin/ghdl -a   pipelined_des.vhdl
/usr/local/bin/ghdl -a   input_vector.vhdl
/usr/local/bin/ghdl -a   key_vector.vhdl
/usr/local/bin/ghdl -a   output_vector.vhdl
/usr/local/bin/ghdl -a   encrypt_vector.vhdl
/usr/local/bin/ghdl -a   pipelined_des_tb.vhdl
/usr/local/bin/ghdl -e   pipelined_des_tb
/usr/local/bin/ghdl -r  pipelined_des_tb --wave=pipelined_des_tb.ghw
```
analyzes, elaborates and simulates:

[image]

And the gist of this is that versioning of llvm and gnat packages aren't necessarily as sensitive as you might think. There's every chance these binary distributions would work under 10.14 MacOS.  

I've got the gnat gpl 2018 dmg around somewhere. I recall a minor issue with Ada versions that Tristan my have addressed.

1138-4EB, 2019-10-02:
> What should be better to keep this open or get all the things to the other #949 issue?

At this point, I think we can follow on here. We are not strict at all in this sense.

anquegi, 2019-10-02:
> I wanted to see how close I could get to testing the anquegi's hypothesis
> 
> _I cannot see any binaries comparible from llvm.org. downloads. So I need the ones with homebrew. The problem is as suggested by @diogratia that I use different tools in each step. I will try to build it correctly in OSX.._
> 
> Building from commit [a462aa2](https://github.com/ghdl/ghdl/commit/a462aa24936a64e32a79286b594490491f7c1dab) (today) using
> 
> ```
> alias nghdl_path 'set path=($HOME/share/clang+llvm-8.0.0-x86_64-apple-darwin/bin $HOME/share/gnat-gpl-2017-x86_64-darwin-bin/bin $path)'
> ```
> 
> clang+llvm-8.0.0 from llvm.org/downloads and gnat-gpl-2017 from Adacore with
> 
> ```
> ../configure --with-llvm-config=/Users/david_koontz/share/clang+llvm-8.0.0-x86_64-apple-darwin/bin/llvm-config
> ```
> 
> (in a build subdirectory
> 
> The build completed.
> 
> The testsuite errors on synth:
> 
> ```
> ...
> Vests is OK
>    22595 vests.log
> [GHDL - test] synth
> 
> dir aggr01:
> analyze aggr01.vhdl tb_aggr01.vhdl
> elaborate and simulate tb_aggr01
> Remove work library
> Synthesis of aggr01.vhdl -e aggr01
> /Users/david_koontz/Desktop/ghdl/ghdl-0.37dev/4e000b81/ghdl/build/ghdl_llvm: unknown command '--synth', try --help
> #################################################################
> ######### FAILURE: aggr01
> #################################################################
> ```
> 
> (meaning I need to figure out how not to run synth tests)
> 
> and the installed ghdl
> 
> ```
> ghdl --version
> GHDL 0.37-dev (v0.36-892-ga462aa2) [Dunoon edition]
>  Compiled with GNAT Version: GPL 2017 (20170515-63)
>  llvm code generator
> Written by Tristan Gingold.
> 
> /usr/local/bin/ghdl -a   sbox1.vhdl
> /usr/local/bin/ghdl -a   sbox2.vhdl
> /usr/local/bin/ghdl -a   sbox3.vhdl
> /usr/local/bin/ghdl -a   sbox4.vhdl
> /usr/local/bin/ghdl -a   sbox5.vhdl
> /usr/local/bin/ghdl -a   sbox6.vhdl
> /usr/local/bin/ghdl -a   sbox7.vhdl
> /usr/local/bin/ghdl -a   sbox8.vhdl
> /usr/local/bin/ghdl -a   frk.vhdl
> /usr/local/bin/ghdl -a   pipe_stage.vhdl
> /usr/local/bin/ghdl -a   last_stage.vhdl
> /usr/local/bin/ghdl -a   pipelined_des.vhdl
> /usr/local/bin/ghdl -a   input_vector.vhdl
> /usr/local/bin/ghdl -a   key_vector.vhdl
> /usr/local/bin/ghdl -a   output_vector.vhdl
> /usr/local/bin/ghdl -a   encrypt_vector.vhdl
> /usr/local/bin/ghdl -a   pipelined_des_tb.vhdl
> /usr/local/bin/ghdl -e   pipelined_des_tb
> /usr/local/bin/ghdl -r  pipelined_des_tb --wave=pipelined_des_tb.ghw
> ```
> 
> analyzes, elaborates and simulates:
> 
> [image]
> 
> And the gist of this is that versioning of llvm and gnat packages aren't necessarily as sensitive as you might think. There's every chance these binary distributions would work under 10.14 MacOS.
> 
> I've got the gnat gpl 2018 dmg around somewhere. I recall a minor issue with Ada versions that Tristan my have addressed.

I tried to reprodue it dowloaded the binaries clang llvm 8.0.0, and doesn't work the only difference is that I use 2019 GNAT.

Also I do not understand de alias, is supposed to execute to add the path of the bin directory in clang?

And I also need to change CC=clang if not I get the error from #949 

Sorry for this incoveniences

1138-4EB, 2019-10-04:
@anquegi, the alias ensures that the downloaded binaries are found before the ones available in the path. Doing so should pick `clang` automatically, without modifying the makefile. Furthermore, should you need to set `CC`, you should do it as a cli arg.

anquegi, 2019-10-04:
So I get this setup:

```
~/learn/vhdl
➜ git clone https://github.com/ghdl/ghdl.git
Cloning into 'ghdl'...
remote: Enumerating objects: 1082, done.
remote: Counting objects: 100% (1082/1082), done.
remote: Compressing objects: 100% (590/590), done.
remote: Total 37450 (delta 751), reused 789 (delta 492), pack-reused 36368
Receiving objects: 100% (37450/37450), 53.73 MiB | 3.32 MiB/s, done.
Resolving deltas: 100% (28848/28848), done.
Checking out files: 100% (8213/8213), done.

~/learn/vhdl took 21s
➜ export PATH="/Users/toni/opt/clang+llvm-8.0.0-x86_64-apple-darwin/bin:$PATH"

~/learn/vhdl
➜ which clang
/Users/toni/opt/clang+llvm-8.0.0-x86_64-apple-darwin/bin/clang

~/learn/vhdl
➜ which gcc
/Users/toni/opt/GNAT/2019/bin/gcc
```

then I configure:
```
ghdl on  master
➜ mkdifr build && cd build
ghdl/build on  master
➜ ../configure --prefix=/usr/local --with-llvm-config=/Users/toni/opt/clang+llvm-8.0.0-x86_64-apple-darwin/bin/llvm-config
Use full IEEE library
Build machine is: x86_64-apple-darwin17.7.0
Debugging is not enabled with llvm 8.0.0
Creating ghdl.gpr
Creating Makefile
for d in src/std src/std/v87 src/std/v93 src/std/v08 src/synopsys src/synopsys/v08 std/v87 ieee/v87 synopsys/v87 std/v93 ieee/v93 synopsys/v93 std/v08 ieee/v08 src/openieee src/ieee src/vital95 src/vital2000 src/mentor src/ieee/v87 src/ieee/v93 src/ieee2008 mentor/v93 synopsys/v08; do \
	  mkdir -p lib/ghdl/$d; \
	done
Generate ghdlsynth_maybe.ads
Generate default_paths.ads
```

And with this setup I cannot build anything when I do make, it starts to generate a lot of logs and finally:
```
                                                          ....
                                                          ....
gcc -c -g -o fstapi.o /Users/toni/learn/vhdl/ghdl/build/../src/grt/fst/fstapi.c -I/Users/toni/learn/vhdl/ghdl/build/../src/grt/fst
gcc -c -g -o lz4.o /Users/toni/learn/vhdl/ghdl/build/../src/grt/fst/lz4.c
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/wait.h:110,
                 from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/stdlib.h:66,
                 from /Users/toni/learn/vhdl/ghdl/build/../src/grt/fst/lz4.c:105:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/resource.h: In function 'getiopolicy_np':
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/resource.h:443:34: error: expected declaration specifiers before '__OSX_AVAILABLE_STARTING'
 int     getiopolicy_np(int, int) __OSX_AVAILABLE_STARTING(__MAC_10_5, __IPHONE_2_0);
                                  ^~~~~~~~~~~~~~~~~~~~~~~~
                                           .....
                                           ....
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/libkern/i386/_OSByteOrder.h:49:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
 {
 ^ 
                                                      ....
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/_types/_ct_rune_t.h:32:28: error: storage class specified for parameter 'ct_rune_t'
 typedef __darwin_ct_rune_t ct_rune_t;
                            ^~~~~~~~~
```

I add some different errors

diogratia, 2019-10-05:
> `gcc` -c -g -o lz4.o /Users/toni/learn/vhdl/ghdl/build/../src/grt/fst/lz4.c
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/wait.h:110,
                 from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/stdlib.h:66,
                 from /Users/toni/learn/vhdl/ghdl/build/../src/grt/fst/lz4.c:105:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/resource.h: In function 'getiopolicy_np':
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/resource.h:443:34: error: expected declaration specifiers before '__OSX_AVAILABLE_STARTING'
 int     getiopolicy_np(int, int) __OSX_AVAILABLE_STARTING(__MAC_10_5, __IPHONE_2_0);
                                  ^~~~~~~~~~~~~~~~~~~~~~~~

This appears to be telling us the _OSX_AVAILABLE_STARTING  from the compiler is declared incompatibly with it's usage in the MacOSX.sdk (It's likely part of someone's corrections to the MacOSX SDK API, look for include file size differences between Xcode versions).

The README.txt associated with the gnat-community-2019-20190517-x86_64-darwin-bin.dmg telegraphs the issue:

>On Mac OS, GNAT Community 2019 requires Xcode version 10 or above to be
installed.

clang+llvm-8.0.0-x86_64-apple-darwin/bin/gcc clearly isn't completely API compatible for at least one of the fst C files.

For the problem mentioning _OSX_AVAILABLE_STARTING:

- You could set gnat-community-2019-20190517-x86_64-darwin-bin before both clang+llvm-8.0.0-x86_64-apple-darwin/bin and /usr/bin in your execution search path. This places gnat-community-2019-20190517-x86_64-darwin-bin/bin/gcc (and g++ for CXX) first to be found in the execution search path. 

- You could put gnat-community-2019-20190517-x86_64-darwin-bin before both clang+llvm-8.0.0-x86_64-apple-darwin/bin/gcc  configure.

- You could have a separate makefile or script to produce all the C source objects files (all in ghdl/src/grt for the llvm version). 

These methods should use /usr/bin/{ar,ranlib} not the ones provide by Homebrew from gnu binutils and count on gnat-community-2019-20190517-x86_64-darwin-bin being API compatible with Xcode 10+. Otherwise a solution needs finer control (more definitions in configure/Makefile) allowing these C source files to be compiled with /usr/bin/gcc.

Whether other issues might surface is unknown.  The risk of vulnerability to other API issues is low using either /usr/bin/gcc or gnat-community-2019-20190517-x86_64-darwin-bin/bin/gcc.

There's a potential for ABI changes due to API changes but there are  very few dynamically linked libraries used in ghdl. Using ldd (utool -L) zll are low version and pretty much standard across MacOS versions. The ghdl llvm back end comes from statically linked libraries that seem mostly self referential as a group.

A ghdl built on an earlier Xcode/MacOS version should run on 10.14 MacOS. It's easier to get a commit/release version you're happy with if you build your own - there have been lots of bug fixes and some added -2008 feature support since ghdl-0.36.

As far as I can tell the gcc guys haven't released a formal fix for this issue (it's not in a release, instead there's a patch people use presumably like Adacore in gnat-community-2019-20190517-x86_64-darwin-bin).

anquegi, 2019-10-05:
@diogratia , thanks for clarifying but I do not get it working. I tried this:

>     * You could set gnat-community-2019-20190517-x86_64-darwin-bin before both clang+llvm-8.0.0-x86_64-apple-darwin/bin and /usr/bin in your execution search path. This places gnat-community-2019-20190517-x86_64-darwin-bin/bin/gcc (and g++ for CXX) first to be found in the execution search path.

I tried with this:

```
~/learn/vhdl
➜ export PATH="/Users/toni/opt/clang+llvm-8.0.0-x86_64-apple-darwin/bin:$PATH"

~/learn/vhdl
➜ export PATH="/Users/toni/opt/GNAT/2019/bin:$PATH"

~/learn/vhdl
➜ echo $PATH
/Users/toni/opt/GNAT/2019/bin:/Users/toni/opt/clang+llvm-8.0.0-x86_64-apple-darwin/bin:/Users/toni/.rvm/gems/ruby-2.6.3@learn/bin:/Users/toni/.rvm/gems/ruby-2.6.3@global/bin:/Users/toni/.rvm/rubies/ruby-2.6.3/bin:/Users/toni/perl5/bin:/Users/toni/.opam/4.07.0/bin:/Users/toni/.opam/4.07.0/bin:/Users/toni/.kiex/elixirs/elixir-1.6.1/bin:/Users/toni/.kiex/bin:/Users/toni/.nvm/versions/node/v12.4.0/bin:/Users/toni/.kerl/erlang21/bin:/Users/toni/.jenv/shims:/usr/local/opt/tcl-tk/bin:/Users/toni/.roswell/bin:/usr/local/sbin:/usr/local/bin:/Users/toni/.cargo/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Library/TeX/texbin:/usr/local/share/dotnet:/opt/X11/bin:~/.dotnet/tools:/Library/Frameworks/Mono.framework/Versions/Current/Commands:/Library/Frameworks/Mono.framework/Versions/Current/bin/:/Users/toni/bin/apache-maven-3.3.9/bin:/Users/toni/.rvm/bin

~/learn/vhdl
➜ cd ghdl

ghdl on  master
➜ echo $PATH
/Users/toni/opt/GNAT/2019/bin:/Users/toni/opt/clang+llvm-8.0.0-x86_64-apple-darwin/bin:/Users/toni/.rvm/gems/ruby-2.6.3@learn/bin:/Users/toni/.rvm/gems/ruby-2.6.3@global/bin:/Users/toni/.rvm/rubies/ruby-2.6.3/bin:/Users/toni/perl5/bin:/Users/toni/.opam/4.07.0/bin:/Users/toni/.opam/4.07.0/bin:/Users/toni/.kiex/elixirs/elixir-1.6.1/bin:/Users/toni/.kiex/bin:/Users/toni/.nvm/versions/node/v12.4.0/bin:/Users/toni/.kerl/erlang21/bin:/Users/toni/.jenv/shims:/usr/local/opt/tcl-tk/bin:/Users/toni/.roswell/bin:/usr/local/sbin:/usr/local/bin:/Users/toni/.cargo/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Library/TeX/texbin:/usr/local/share/dotnet:/opt/X11/bin:~/.dotnet/tools:/Library/Frameworks/Mono.framework/Versions/Current/Commands:/Library/Frameworks/Mono.framework/Versions/Current/bin/:/Users/toni/bin/apache-maven-3.3.9/bin:/Users/toni/.rvm/bin

ghdl on  master
➜ mkdir build

ghdl on  master
➜ cd build

ghdl/build on  master
➜ ../configure --prefix=/usr/local --with-llvm-config=/Users/toni/opt/clang+llvm-8.0.0-x86_64-apple-darwin/bin/llvm-config
Use full IEEE library
Build machine is: x86_64-apple-darwin17.7.0
Debugging is not enabled with llvm 8.0.0
create pic/ subdirectory
Creating ghdl.gpr
Creating Makefile
for d in src/std src/std/v87 src/std/v93 src/std/v08 src/synopsys src/synopsys/v08 std/v87 ieee/v87 synopsys/v87 std/v93 ieee/v93 synopsys/v93 std/v08 ieee/v08 src/openieee src/ieee src/vital95 src/vital2000 src/mentor src/ieee/v87 src/ieee/v93 src/ieee2008 mentor/v93 synopsys/v08; do \
	  mkdir -p lib/ghdl/$d; \
	done
Generate ghdlsynth_maybe.ads
Generate default_paths.ads
```
Then I get the same error,

gcc -c -g -o lz4.o /Users/toni/learn/vhdl/ghdl/build/../src/grt/fst/lz4.c
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/wait.h:110,
                 from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/stdlib.h:66,
                 from /Users/toni/learn/vhdl/ghdl/build/../src/grt/fst/lz4.c:105:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/resource.h: In function 'getiopolicy_np':
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/resource.h:443:34: error: expected declaration specifiers before '__OSX_AVAILABLE_STARTING'
 int     getiopolicy_np(int, int) __OSX_AVAILABLE_STARTING(__MAC_10_5, __IPHONE_2_0);
                                  ^~~~~~~~~~~~~~~~~~~~~~~~

I'm using xcode 11.

tgingold, 2019-10-05:
IMHO you are looking for troubles.  Use clang from Xcode for C/C++ code and gnat for Ada code.

diogratia, 2019-10-06:
Tristan's concern appears to relate to using /Users/toni/opt/GNAT/2019/bin/gcc for compiling objects from C files.  I'd expect Tristan comment to be authoritative, if uninformative.

Apple's /usr/bin/gcc is a wrapper calling their clang located by searching Xcode using xcrun to find and execute gcc (cc). Executing gcc -v for /usr/bin/gcc will report it's found in Xcode and is clang:

``` none
 gcc -v
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 8.0.0 (clang-800.0.42.1)
Target: x86_64-apple-darwin15.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```
(cc in the bin directory is linked to clang in the same bin directory. These from my Xcode 8.2.1 on 10.11.6 OSX.)

Apple's own gcc (clang) will be API change aware. 

Looking through a successful build output gcc would be invoked to compile Ada sources using /Users/toni/opt/GNAT/2019/bin/gcc to compile Ada objects. None of the Ada source files depend on Apple's API changes. Using that gcc should be okay.

Objects from C files should likely be built using /usr/bin/gcc (Xcode clang) or /usr/bin/clang (without knowing while the Homebrew llvm@8 failed to produce a ranlib output static library that ghdl couldn't elaborate with). Only nine C files are used for MacOS ghdl with an llvm back end. Yet another compiler means a Makefile change or a kludge. These no notice changes by Apple can take two or three releases to propagate. We can see when llvm.org quit providing Darwin binaries. This kind of thing makes platforms seem less attractive to open source developers.

It's probably worthwhile investigating why Homebrew llvm@8 built objects that when loaded into a static library couldn't be linked. It may have been using Gnu binutils ranlib and ar or they went around part of an API change resulting in loss of versioning and prevented ghdl from elaborating with libgrt.a.

anquegi, 2019-12-31:
I was testing gnu-apl and also rvm installing, and had similar problems, but reading on the web I removed binutils from brew and now it worked. I add some usefull links.
https://github.com/pyenv/pyenv/issues/640#issuecomment-451896377
or 
https://github.com/rvm/rvm/issues/3247

I hopw this should help , I'm going to close this issue thanks for your help, finally I can use an own compiled ghdl

closed 2019-12-31 by anquegi