ghdl/ghdl #961: VHDL2008 unconstrained array of record crash

eschmidscs, 2019-10-02:
**Description**
The provided nonsensical VHDL code crashes in elaboration with the bug report pasted below.
If the I and O ports and signals are removed, it runs ok.

**Expected behaviour**
It should simulate fine, reporting some errors. It does so with Questa 10.7b.

**How to reproduce?**
Analyze and elaborate the following code:

```vhd :file: ghdl2008test.vhd
library ieee;
context ieee.ieee_std_context;
use ieee.math_complex.all;

package testpkg is
  type SFIXED is array(integer range <>) of std_logic;
  type SFIXED_VECTOR is array(integer range <>) of SFIXED;
  type CFIXED is record RE, IM : SFIXED; end record;
  type CFIXED_VECTOR is array(integer range <>) of CFIXED;
  type CFIXED_MATRIX is array(integer range <>) of CFIXED_VECTOR;
end package testpkg;

library ieee;
context ieee.ieee_std_context;
use ieee.math_complex.all;
use work.testpkg.all;

entity dut is
  port(CLK : in  std_logic;
       I   : in  CFIXED_VECTOR;
       VI  : in  boolean;
       SI  : in  unsigned;
       O   : out CFIXED_VECTOR;
       VO  : out boolean;
       SO  : out unsigned);
end dut;

architecture test of dut is
begin
  process(CLK)
  begin
    if rising_edge(CLK) then
      VO <= false;
      if VI then
        VO <= true;
        SO <= resize(SI, SO'length);

        for a in 0 to O'length-1 loop
          O(a) <= (RE => '0' & I(a mod I'length).RE, IM => '0' & I(a mod I'length).IM);
        end loop;
      end if;
    end if;
  end process;
end architecture;

library ieee;
context ieee.ieee_std_context;
use ieee.math_complex.all;
use work.testpkg.all;

entity tb is
end entity tb;

architecture tb of tb is
  signal CLK : std_logic := '0';
  signal I   : cfixed_vector(3 downto 0)(RE(15 downto 0), IM(15 downto 0));
  signal VI  : boolean   := false;
  signal SI  : unsigned(7 downto 0);
  signal O   : cfixed_vector(4 downto 0)(RE(16 downto 0), IM(16 downto 0));
  signal VO  : boolean;
  signal SO  : unsigned(14 downto 0);
begin
  CLK <= not CLK after 2 ns;

  process
  begin
    wait for 10 ns;
    I(0) <= (RE => x"ABCD", IM => x"0123");
    I(1) <= (RE => x"ABDD", IM => x"5123");
    I(2) <= (RE => x"ACCD", IM => x"6123");
    I(3) <= (RE => x"BBCD", IM => x"7123");
    SI   <= x"88";
    VI   <= true;
    wait until rising_edge(CLK);
    VI   <= false;
    wait for 20 ns;
    std.env.finish;
  end process;

  inst_dut : entity work.dut
    port map (
      CLK => CLK,
      I   => I,
      VI  => VI,
      SI  => SI,
      O   => O,
      VO  => VO,
      SO  => SO
      );

  process(CLK)
  begin
    if rising_edge(CLK) then
      if VO then
        assert SO = 14x"0888" report "SO wrong";
        assert O(0) = (RE => 17x"0ABCD", IM => 17x"00123") report "O0 wrong";
        assert O(1) = (RE => 17x"0ABCD", IM => 17x"00123") report "O0 wrong";
        assert O(2) = (RE => 17x"0ABCD", IM => 17x"00123") report "O0 wrong";
        assert O(3) = (RE => 17x"0ABCD", IM => 17x"00123") report "O0 wrong";
        assert O(4) = (RE => 17x"0ABCD", IM => 17x"00123") report "O0 wrong";
      end if;
    end if;
  end process;
end architecture tb;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a --std=08 ghdl2008test.vhd
ghdl -e --std=08 tb
```

**Context**
- OS: Debian GNU/Linux bullseye/sid
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [X] Built from sources: `a462aa24936a64e32a79286b594490491f7c1dab`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:
```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-99-g34666ab2) [Dunoon edition]
Compiled with GNAT Version: 8.3.0
Target: x86_64-linux-gnu
In directory: /home/eschmid/403-1/task/sigchain/foo/
Command line:
ghdl -e --std=08 tb
Exception ORTHO_CODE.SYNTAX_ERROR raised
Exception information:
raised ORTHO_CODE.SYNTAX_ERROR : ortho_code-exprs.adb:451
Call stack traceback locations:
0x5583d778e929 0x5583d778e99d 0x5583d778fc1d 0x5583d77a7081 0x5583d799e539 0x5583d799e61f 0x5583d799f342 0x5583d798a134 0x5583d7988419 0x5583d798d7d4 0x5583d79378e1 0x5583d79380f2 0x5583d7939687 0x5583d79397b9 0x5583d7928954 0x5583d7928ab7 0x5583d793965a 0x5583d79397b9 0x5583d7926faf 0x5583d792705b 0x5583d793961e 0x5583d79397b9 0x5583d7926faf 0x5583d792705b 0x5583d793961e 0x5583d79397b9 0x5583d793cc0d 0x5583d7941452 0x5583d79acf2e 0x5583d7918dd9 0x5583d7915f66 0x5583d7923b4e 0x5583d79b1512 0x5583d78fadc2 0x5583d7859426 0x5583d79b4cda 0x5583d76bf46d 0x7f875a75abb9 0x5583d76be108 0xfffffffffffffffe
******************************************************************
```

eine, 2020-08-05:
I think that this might be fixed in latest master. At least, I don't get the crash. @eschmidscs can you please confirm?

eschmidscs, 2020-08-10:
Yes, the code as given compiles fine here with the current master (68b879eb404e12a6414ef56e697b6dc99618d457). I'm trying to compile and run that in the original code, but I hit some different bug there (report incoming...).

closed 2020-08-10 by eschmidscs