ghdl/ghdl #962: synth: also support division for integer types

Xiretza, 2019-10-02:
**Description**
#953 introduced division for unsigned-like types, however integer types still result in a synthesis error (`error: non-constant division not supported`).

**Expected behaviour**
Integer division can be synthesized.

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```vhd :file: ent.vhd
entity ent is
end;

architecture a of ent is
    signal x : integer;
    signal y : integer := x / 2;
begin
end;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl --synth ent.vhd -e ent
```

**Context**
Please, provide the following information:

- OS: Arch Linux
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: a462aa24

1138-4EB, 2019-10-02:
What if `signal x : integer := 8;` or `signal x : integer := -512;`?

Xiretza, 2019-10-02:
@1138-4EB neither makes a difference, the error remains.

closed 2019-10-03 by tgingold