ghdl/ghdl #967: Regression in synth_expression

pepijndevos, 2019-10-05:
After ORConf I figured I'd come back and fix the issues in my display code, only to find that all the problems in the display code are already fixed, but my CPU itself no longer compiles. It crashes with an access check in the following trace:
```
#0  synth.context.get_value (syn_inst=0x555555af5490, obj=5878)
    at /home/pepijn/code/ghdl/src/synth/synth-context.adb:313
#1  0x00005555557e6836 in synth.context.get_value_type (syn_inst=0x555555af5490, atype=5878)
    at /home/pepijn/code/ghdl/src/synth/synth-context.adb:322
#2  0x0000555555805965 in synth.expr.synth_expression (syn_inst=0x555555af5490, expr=5436)
    at /home/pepijn/code/ghdl/src/synth/synth-expr.adb:1648
#3  0x00005555557f2395 in synth.stmts.synth_selected_signal_assignment (syn_inst=0x555555af5490, stmt=5386)
    at /home/pepijn/code/ghdl/src/synth/synth-stmts.adb:1109
#4  0x00005555557f7449 in synth.stmts.synth_concurrent_statements (syn_inst=0x555555af5490, stmts=5246)
    at /home/pepijn/code/ghdl/src/synth/synth-stmts.adb:2373
#5  0x00005555557ec652 in synth.insts.synth_instance (inst=...)
    at /home/pepijn/code/ghdl/src/synth/synth-insts.adb:801
#6  0x00005555557ec6ff in synth.insts.synth_all_instances () at /home/pepijn/code/ghdl/src/synth/synth-insts.adb:825
#7  0x00005555557e8dae in synthesis.synth_design (design=5954, inst=0x555555af5490, m=5)
    at /home/pepijn/code/ghdl/src/synth/synthesis.adb:138
#8  0x0000555555833d3c in ghdlsynth.perform_action (cmd=..., args=...)
    at /home/pepijn/code/ghdl/src/ghdldrv/ghdlsynth.adb:268
#9  0x000055555578f1db in ghdlmain.main () at /home/pepijn/code/ghdl/src/ghdldrv/ghdlmain.adb:476
#10 0x00005555558e65eb in ghdl_jit () at /home/pepijn/code/ghdl/src/ghdldrv/ghdl_jit.adb:40
```

I did a git bisect and the offending commit is https://github.com/ghdl/ghdl/commit/010aca1966eeb260529041d209d69a92654465f8#diff-318a658cc2879df474ef223cb66ccd87L1725

It seems that `Get_Info (Obj)` returns null. Not 100% sure what's going on here.

Currently don't have an small testcase any smaller than my CPU: https://github.com/pepijndevos/seqpu/blob/master/cpu.vhd which can be tested with `ghdl --synth --std=08 alu.vhd inc.vhd cpu.vhd -e cpu`. Might try to dig a bit deeper, but that's what I have at the moment.

tgingold, 2019-10-05:
Should be fixed.

pepijndevos, 2019-10-05:
Works! Thanks. Now I'm back to issues I can fix myself :)

closed 2019-10-05 by pepijndevos