ghdl/ghdl #968: synth: assertion failure for --synth on T65 code

Fatsie, 2019-10-06:
**Description**
ghdl --synth fails on T65 VHDL vode

**How to reproduce?**
I am running on Ubuntu 18.04.3 LTS with freshly compiled ghdl using gnat-8 package. Following steps should reproduce the problem:
```console
eda@eda:/tmp$ git clone https://gitlab.com/Chips4Makers/t65.git
Cloning into 't65'...
remote: Enumerating objects: 97, done.
remote: Counting objects: 100% (97/97), done.
remote: Compressing objects: 100% (72/72), done.
remote: Total 97 (delta 22), reused 0 (delta 0)
Unpacking objects: 100% (97/97), done.
eda@eda:/tmp$ cd t65/rtl/vhdl
eda@eda:/tmp/t65/rtl/vhdl$ export GHDL_OPTS="--std=08 --ieee=synopsys"
eda@eda:/tmp/t65/rtl/vhdl$ ghdl -a $GHDL_OPTS T65_Pack.vhd T65_ALU.vhd T65_MCode.vhd T65.vhd
eda@eda:/tmp/t65/rtl/vhdl$ ghdl -e $GHDL_OPTS T65
eda@eda:/tmp/t65/rtl/vhdl$ ghdl --synth $GHDL_OPTS T65

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-928-g950f6b8c) [Dunoon edition]
Compiled with GNAT Version: 8.3.0
Target: x86_64-linux-gnu
/tmp/t65/rtl/vhdl/
Command line:
ghdl --synth --std=08 --ieee=synopsys T65
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : netlists-builders.adb:642
Call stack traceback locations:
0x7f858232f171 0x5565b618ce68 0x5565b62f7e35 0x5565b62f5e09 0x5565b6301f7f 0x5565b6302a03 0x5565b62eca22 0x5565b62f4b7c 0x5565b62ea52e 0x5565b62ea5ce 0x5565b62e6fdc 0x5565b632e333 0x5565b6290bd7 0x5565b63d859d 0x5565b6078261 0x7f85818f4b95 0x5565b6076bd8 0xfffffffffffffffe
******************************************************************
eda@eda:/tmp/t65/rtl/vhdl$ cat -n ~/Compile/ghdl/src/synth/netlists-builders.adb |grep '64[0-5]'
   640	      Wd : constant Width := Get_Width (L);
   641	      pragma Assert (Wd /= No_Width);
   642	      pragma Assert (Get_Width (R) = Wd);
   643	      pragma Assert (Ctxt.M_Compare (Id) /= No_Module);
   644	      Inst : Instance;
   645	      O : Net;
```

tgingold, 2019-10-06:
It's due to:
```
	VDA	<= '1' when Set_Addr_To_r /= "000" else '0';		-- Incorrect !!!!!!!!!!!!
```
which is indeed incorrect.

I will fix to not crash.

tgingold, 2019-10-06:
I think this is fixed.

closed 2019-10-06 by tgingold

tmeissner, 2019-10-06:
Fastest support in EDA "industry" 🥇

Fatsie, 2019-10-06:
Shouldn't this give warning/error during annotation or elaboration ?

BTW, was using old version of T65 code also, ...

tgingold, 2019-10-06:
Yes, it could also gives a warning for simulation.  But with synthesis, you will always get a warning even when the length is not constant.

1138-4EB, 2019-10-06:
> Shouldn't this give warning/error during annotation or elaboration ?

I'd say that GHDL does produce a warning/error during the elaboration, in terms of the VHDL standard. However, `ghdl -e` is not a complete VHDL elaboration. Some steps are delayed to `ghdl -r`. `ghdl --elab-run --no-run` will do a complete VHDL elaboration without actually simulating the design. See https://ghdl.readthedocs.io/en/latest/using/InvokingGHDL.html#elaboration-e