ghdl/ghdl #969: synth: cannot handle IIR_KIND_CONCURRENT_PROCEDURE_CALL_STATEMENT on T80 code

Fatsie, 2019-10-06:
**Description**
ghdl --synth fails on T80 VHDL code.

**How to reproduce?**
I am running on Ubuntu 18.04.3 LTS with freshly compiled ghdl using gnat-8 package. Following steps should reproduce the problem:

```console
eda@eda:/tmp$ git clone https://gitlab.com/Chips4Makers/t80.git
Cloning into 't80'...
remote: Enumerating objects: 25, done.
remote: Counting objects: 100% (25/25), done.
remote: Compressing objects: 100% (20/20), done.
remote: Total 492 (delta 6), reused 0 (delta 0)
Receiving objects: 100% (492/492), 124.50 KiB | 394.00 KiB/s, done.
Resolving deltas: 100% (245/245), done.
eda@eda:/tmp$ cd t80/rtl/vhdl/
eda@eda:/tmp/t80/rtl/vhdl$ export GHDL_OPTS="--std=08 --ieee=synopsys"
eda@eda:/tmp/t80/rtl/vhdl$ ghdl -a $GHDL_OPTS T80_Pack.vhd T80_ALU.vhd T80_Reg.vhd T80_MCode.vhd T80.vhd T80se.vhd Z80.vhd 
T80_MCode.vhd:156:16:warning: declaration of "f" hides port "f" [-Whide]
eda@eda:/tmp/t80/rtl/vhdl$ ghdl -e $GHDL_OPTS Z80
eda@eda:/tmp/t80/rtl/vhdl$ ghdl --synth $GHDL_OPTS Z80
annotate_concurrent_statements_list: cannot handle IIR_KIND_CONCURRENT_PROCEDURE_CALL_STATEMENT (T80_ALU.vhd:143:8)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-928-g950f6b8c) [Dunoon edition]
Compiled with GNAT Version: 8.3.0
Target: x86_64-linux-gnu
/tmp/t80/rtl/vhdl/
Command line:
ghdl --synth --std=08 --ieee=synopsys Z80
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:32
Call stack traceback locations:
0x560ad244f878 0x560ad2484569 0x560ad2484758 0x560ad2484ff2 0x560ad258ad29 0x560ad258b2ea 0x560ad24edbd7 0x560ad263559d 0x560ad22d5261 0x7f454c10db95 0x560ad22d3bd8 0xfffffffffffffffe
******************************************************************
```

tgingold, 2019-10-06:
This was also reported in #960.  Work in progress.

1138-4EB, 2019-10-06:
Closing as duplicate. See #960 and https://github.com/ghdl/ghdl/issues/970#issuecomment-538750552.

tgingold, 2019-10-06:
Well, it is slightly different, as it is about a concurrent procedure call.  I should have said it depends on #960.

closed 2019-10-07 by tgingold

tgingold, 2019-10-07:
Now fixed, but there are still issues with T80.

Fatsie, 2019-10-07:
OK, should I file new ticket or will you follow it up ?
BTW, I also updated T65 master and it now also gives problems.

tgingold, 2019-10-08:
No need to create a new issue, I will work on T80.

T65 should now be OK.

tgingold, 2019-10-09:
T80 is OK for me now.

Fatsie, 2019-10-09:
Do you use a yosys feature of the last days ?
ghdl --synth now works but ghdl command in yosys not:
```console
% ghdl --synth $GHDL_OPTS T80 >/dev/null
% yosys -m ghdl

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2019  Clifford Wolf <clifford@clifford.at>           |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.9+932 (git sha1 6ad29184, gcc 7.4.0-1ubuntu1~18.04.1 -fPIC -Os)

yosys> ghdl --std=08 --ieee=synopsys T80
1. Executing GHDL.
Importing module t80.
ERROR: Unsupported(1): instance \497 of $sextend.
ERROR: Assert `GetSize(ports) >= it.second->port_id' failed in kernel/rtlil.cc:1353.
```

tmeissner, 2019-10-09:
`Unsupported(1): instance \497 of $sextend` means that handling of this construct has to be added to ghdlsynth-beta, not in GHDL. It are defined in `netlist-gates.ads` but not yet handled in ghdlsyth-beta. The name suggests some kind of width extension of objects of signed data type. Maybe it can be handled in similar way as Id_Uextend, which is already implemented in ghdlsynth-beta:

```c++
case Id_Uextend:
	{
		RTLIL::SigSpec res = IN(0);
		res.extend_u0(get_width(n), false);
		return res;
	}
```

tgingold, 2019-10-09:
> ghdl --synth now works but ghdl command in yosys not:

Step by step!

commit 5a4e0827 (2019-10-07): synth: add support for concurrent procedure calls.  Fix #969