ghdl/ghdl #971: Option to import all annotated/elaborated units in yosys

Fatsie, 2019-10-06:
**Is your feature request related to a problem? Please describe.**
I am using nmigen to generate a top cell for a circuit that contains also modules written in VHDL. Problem is that I do get a list of all the VHDL files but not the list of modules that are needed in the design.

**Describe the solution you'd like**
Implement option for the `ghdl` command in yosys to import all units that have been annotated/elaborated. This would make behaviour more in line with the `read_*` commands that also import all modules from a certain file.
Alternative solution could be implementation of an extra `ghdl_read` command that reads one VHDL file and import all units from it.

tgingold, 2019-10-06:
If I understand correctly, nmigen directly generates rtlil and therefore you can use yosys to drive the elaboration.

Does it work with the verific front-end ?  I had the impression that it works like ghdl: you have to all the files.

1138-4EB, 2019-10-06:
The specific repo that connects GHDL and yosys is [tgingold/ghdlsynth-beta](https://github.com/tgingold/ghdlsynth-beta). Anyway, that repo might be merged here when it is out of beta; hence, it is ok to keep this issue here. Just putting the reference here to be aware that other related issues might exists there.

Fatsie, 2019-10-06:
> Does it work with the verific front-end ? I had the impression that it works like ghdl: you have to all the files.

Verific can read both verilog and vhdl files. I do know name of top module but not the name of the sub-modules I need out of the VHDL files. I let nmigen generate a verilog file. I read all verilog and VHDL files with verific and then import the top module.

LAK132, 2022-06-08:
> If I understand correctly, nmigen directly generates rtlil and therefore you can use yosys to drive the elaboration

amaranth (nmigen), despite supporting the ability to import vhdl, cannot *read* vhdl. I'm working on a fork of amaranth that adds support for targeting yosys/nextpnr-xilinx/prjxray for xilinx fpgas, and I'd like to use ghdl as a backend for synthesising vhdl. this is turning out to be practically impossible because of the original issue here (must specify top even though there isn't one)