ghdl/ghdl #972: synth: unsigned arithmetic hits assert in netlists.adb:699

antonblanchard, 2019-10-06:
**Description**
When synthesising the code below, I hit an assertion:

```raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : netlists.adb:699```

**How to reproduce?**

```vhd :file: test2.vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test2 is
    port(
        d_in: in std_ulogic_vector(63 downto 0);
        d_out: out std_ulogic_vector(63 downto 0)
        );
end entity test2;

architecture behaviour of test2 is
begin
    comb : process(all)
    begin
        d_out <= std_logic_vector(unsigned(d_in) + 4);
    end process;
end architecture behaviour;
```

```sh :image: ghdl/synth:beta
ghdl -a --std=08 test2.vhdl
yosys -m ghdl -p "ghdl --std=08 test2"
```

**Context**
Please, provide the following information:

- OS: Linux

- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x ] Built from sources: `commit 873285ee1`

1138-4EB, 2019-10-06:
Is `raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : netlists.adb:699` an *isolated error* or the *relevant line* of a *bug occurred* block?

tgingold, 2019-10-07:
Sorry, but I cannot reproduce the crash.  I have added the case in the testsuite so it will be run for regressions testing.

antonblanchard, 2019-10-07:
I notice that `ghdl --synth --std=08 test2` works, but it fails in yosys using `yosys -m ghdl -p "ghdl --std=08 test2"`

The current ghdl/synth:beta Docker container reproduces it:

```docker run --rm -t -i -v $(pwd):/src -w /src ghdl/synth:beta bash -c "ghdl -a --std=08 test2.vhdl; yosys -m ghdl -p \"ghdl --std=08 test2\""```

tgingold, 2019-10-07:
Ah OK, that's a crash in yosys.

1138-4EB, 2019-10-07:
> The current ghdl/synth:beta Docker container reproduces it:

Be aware that `ghdl/synth:beta` is *almost* bleeding-edge, but not *exactly*. The image is published twice a week (https://github.com/ghdl/docker#-extyml-5-jobs--max-4--15-images-twice-a-week), unless there is some failure (which skips the deployment). (un)fortunately, Tristan is so fast! Hence, there might be a 10, 20, 50... commit delay between `master` and the docker image.

You can build latest `ghdl/synth:*` images by executing `./travis.sh` from https://github.com/tgingold/ghdlsynth-beta. It should take 2min: https://travis-ci.com/1138-4EB/ghdlsynth-beta/builds/130604124

I am not suggesting/requiring you to do it. This is just for you to know how to proceed in case you get a "this should be fixed" answer in the future.

BTW, I'm looking into improving the automation between repos. I'm open to any suggestion/discussion.

tgingold, 2019-11-05:
Now passing.

Thank you for the testcase.

closed 2019-11-05 by tgingold