ghdl/ghdl #979: CLANG apparently can't link with "LLVMGetDefaultNormalizedTargetTriple".

FlatAssembler, 2019-10-16:
So, I have Ubuntu 18.04. When I try to build ghdl with LLVM, I get this error:
`llvm-cbindings.o: In function `LLVMGetDefaultNormalizedTargetTriple':
llvm-cbindings.cpp:(.text+0x66): undefined reference to `LLVMNormalizeTargetTriple'
clang: error: linker command failed with exit code 1 (use -v to see invocation)
gnatlink-5: error when calling /usr/bin/clang++
gnatmake: *** link failed.
../src/ortho/llvm4-nodebug/Makefile:12: recipe for target 'ghdl1-llvm' failed`
Do you have any idea what I am doing wrong? I am using CLANG 6.0.

And if I try to use mcode, "ghdl" apparently can't find "std" library when analyzing, I've tried everything and nothing helps.

I am very new to both Linux and VHDL.

1138-4EB, 2019-10-16:
Hi @FlatAssembler! It seems that you are using clang-6 and gnatlink-5. Can you try installing `gnat-6` instead?

Also, please provide as much info as you can to help us reproduce your issue. This is: which packages/dependencies you install, the specific commands that you introduce in the shell, the specific errors you get, etc.

FlatAssembler, 2019-10-16:
> Hi @FlatAssembler! It seems that you are using clang-6 and gnatlink-5. Can you try installing `gnat-6` instead?
>
> Also, please provide as much info as you can to help us reproduce your issue. This [...]

OK, I've installed gnat-6, but I still get the same error. In fact, now not even mcode builds, here is what I get if I try to build mcode (so I literally just invoked "configure" and then "make"):
`./ortho_mcode.o: In function `ortho_mcode__finish':
/home/teo/ghdl/ghdl-master/src/ortho/mcode/ortho_mcode.adb:746: undefined reference to `ortho_ident__disp_stats'
./ortho_mcode.o: In function `ortho_mcode__free_all':
/home/teo/ghdl/ghdl-master/src/ortho/mcode/ortho_mcode.adb:757: undefined reference to `ortho_ident__finish'
collect2: error: ld returned 1 exit status
gnatlink-6: error when calling /usr/bin/gcc-6`

FlatAssembler, 2019-10-16:
> Is `gcc-6` installed on your system?

Apparently not, here is what I get when I run "gcc --version":
`gcc (Ubuntu 7.4.0-1ubuntu1~18.04.1) 7.4.0`

1138-4EB, 2019-10-16:
What about LLVM backend? Did it work? These are the combinations of LLVM and GNAT versions that we do test:

| LLVM | GNAT |
|---|---|
| 4.0 | 6 |
| 7 | 8 |
| 8 | 8 |
| 7 | 7 |
| 6.0 | 7 |

tgingold, 2019-10-16:
1) Which version of LLVM are you using ?  Try llvm-config --version

2) If you are very new to linux, simply get a pre-built binary.

FlatAssembler, 2019-10-17:
1) What's the difference between CLANG and LLVM? As far as I understand, LLVM is something like Java Virtual Machine and CLANG is a program that compiles C and C++ programs into a binary program compatible with it. So, if any program compiles and runs, you must be using the same version of LLVM and CLANG, right?

2) The pre-build binary for Ubuntu relies on GNAT-4, which cannot be installed on Ubuntu 18.

tgingold, 2019-10-17:
1) clang+llvm is the compiler.  clang being the front-end while llvm the code generator.  ghdl doesn't need clang, but uses llvm.
  I am not sure that the analogy with java+jvm is correct.
  (And you don't have answered to my question)

3) Build for mcode (from a clean state).  That's way simpler.

closed 2020-03-04 by tgingold