ghdl/ghdl #980: Revisit how VHDL versions are handled

1138-4EB, 2019-10-16:
Coming from VUnit/vunit#569...

> [1138-4EB]
> IRRC, subdirs named v93 and v08 are generated by GHDL automatically and you cannot modify their names, you can only set the parent ***@***.***, please correct me if I'm wrong).

> [tgingold]
> No, GHDL never creates directories.  However, it searches libraries within subdirectories of -Pdir

I should have said *GHDL's build scripts*. My point is that users or VUnit should not tinker with `v87`, `v93` and `v08` subdir names. Those are used internally by GHDL, as in e.g. `-Pdir`. Although GHDL does not create them, all the built-in libraries are compiled/installed to that directory structure:

```sh
root@f5e52e7835ea:/# tree -d 2 /usr/local/lib/ghdl/
/usr/local/lib/ghdl/
|-- ieee
|   |-- v08
|   |-- v87
|   `-- v93
|-- mentor
|   `-- v93
...
|-- std
|   |-- v08
|   |-- v87
|   `-- v93
|-- synopsys
    |-- v08
    |-- v87
    `-- v93
```

According to @GlenNicholls' [comment](https://github.com/VUnit/vunit/issues/569#issuecomment-542369175), this applies to vendor scripts too.

---

Therefore, there is a slight inconsistency in the sense that all the libraries in the *standard location* follow a structure to tell versions apart, but this is not preserved in the *work* directory. As a result, if the same source is analysed with two versions, some files are overwritten. For example, copying the https://ghdl.readthedocs.io/en/latest/using/QuickStartGuide.html#the-heartbeat-program in a `test.vhd` file:

```sh
root@f5e52e7835ea:~# ghdl -a test.vhd 
root@f5e52e7835ea:~# ls -la
total 36
drwx------ 1 root root 4096 Oct 16 13:52 .
drwxr-xr-x 1 root root 4096 Oct 16 13:51 ..
-rw-r--r-- 1 root root 2408 Oct 16 13:52 test.o
-rw-r--r-- 1 root root   70 Oct 16 13:52 test.vhd
-rw-r--r-- 1 root root  168 Oct 16 13:52 work-obj93.cf
root@f5e52e7835ea:~# ghdl -a --std=08 test.vhd 
root@f5e52e7835ea:~# ls -la
total 40
drwx------ 1 root root 4096 Oct 16 13:53 .
drwxr-xr-x 1 root root 4096 Oct 16 13:51 ..
-rw-r--r-- 1 root root 2408 Oct 16 13:53 test.o
-rw-r--r-- 1 root root   70 Oct 16 13:52 test.vhd
-rw-r--r-- 1 root root  168 Oct 16 13:53 work-obj08.cf
-rw-r--r-- 1 root root  168 Oct 16 13:52 work-obj93.cf
```

Surprisingly, even though `test.o` was *updated* when the source was analysed with `--std=08`, `ghdl -e heartbeat` is successful. However, I believe this is an edge-case, because the source is indeed compatible with v93 (i.e., no v08 features are used). In most cases, `*.o` will not be compatible.

---

Therefore, I'd like to suggest two enhancements:

- Making subdirs `v87`, `v93` and `v08` an internal feature (derived directly from `--std`), instead of doing it in the makefiles/build scripts only. This would imply having subdirs created in the work dir too.

- Emit an specific error message when trying to elaborate a design with a standard version but files analyzed with a different version are available only. Instead of 'unit not found in library lib', something like 'unit not found in library lib for std version XX, but available for std version(s): ... Did you forget --std?'.

---

Nonetheless, because of https://github.com/ghdl/ghdl/issues/908#issuecomment-526893271, it might not be worth applying these suggestions. @tgingold, wdyt?

tgingold, 2019-10-16:
My idea:

1) have only one work-obj.cf
2) keep options used to analyze each file (in particular, --std, -f*).
3) Add --base-std= to select the base standard (not for common users).  This selects which version of std.standard and ieee packages are used.  The default is the latest.
4) Use --std per file which select keyword/construct allowed.  The version for --std should be less than the --base-std.

The problems:
1) What about --ieee= or -Pxxx ?
2) Should new declarations be hidden.  Example: if the --base-std=08 and --std=93, is integer_vector visible ?  And is std_logic_vector a subtype of std_ulogic_vector ?

I fear that problem 2) might not be easy to solve.

1138-4EB, 2019-10-16:
On the one hand, I assume you prefer to skip transitient fixes and go for a *unified* solution. I'm good with this decision, I'd just want to confirm it.

On the other hand, I'm not sure I understand the practical use of the scheme you propose. Please, let me know if this rewrite makes sense:

> 1) Have only one `work-obj.cf`.
> 2) Add `--base-std=` to select which version of `std.standard` and `ieee` packages are used. The default is the latest, which is expected to be used by common users.
> 3) Keep options used to analyze each file (in particular, --std, -f*). The version for `--std` should be less than the `--base-std`.

Questions:

- Is `--base-std=` an (optional) analysis param only? Is it an elaboration param only? Is it required for both of them?
- Is it possible to save `--base-std=` in `work-obj.cf` so that it only needs to be set/updated once?
  - Allow to *upgrade* it after it is defined, but not *downgrade* it.
- Is it possible to build multiple versions of some (standard) packages into the final binary/object? This approach is used in multiple environments (golang, javascript).

> The problems:

I feel that you are proposing an actual *internal transpilation* to integrate sources for older versions of the standard libraries with more recent sources of them. As suggested in the last question above, I think that actually building as many versions of the standard as source types are provided might be a less painful approach. For example, https://ghdl.readthedocs.io/en/latest/using/InvokingGHDL.html#cmdoption-ghdl-frelaxed-rules might not be required for *pure functions*. I have not a deep knowledge of the LRMs, but at some point there might be features which are just impossible conciliate. Hence, trying to conciliate as much as possible might not be worth the effort. Does this make any sense?

arcturus140, 2019-10-17:
> 4) Use --std per file which select keyword/construct allowed.  The version for --std should be less than the --base-std.

is `--base-std` referring to the group of standards and `--std` to VHDL standard? If so shouldn't `--std` >= `--base-std` ?

tgingold, 2019-10-17:
> Questions:
> 
> 	• Is --base-std= an (optional) analysis param only? Is it an elaboration param only? Is it required for both of them?

I will be required for all of them

> 	• Is it possible to save --base-std= in work-obj.cf so that it only needs to be set/updated once?

Not sure.  Again it is not supposed to be for every user.

> 		• Allow to upgrade it after it is defined, but not downgrade it.

No way.

> 	• Is it possible to build multiple versions of some (standard) packages into the final binary/object? This approach is used in multiple environments (golang, javascript).

But at some point you have to choose which one is used.

> The problems:
>
> I feel that you are proposing an actual internal transpilation to integrate sources for older versions of the standard libraries with more recent sources of them. As suggested in the [...]

No, there is no transpilation.  When using --std=93, you will use the 08 version of ieee and std, but allowed to only use 93 features for your file.  My problem is that you will also be able to use some 08 features of ieee and std.

1138-4EB, 2019-10-17:
> > • Is --base-std= an (optional) analysis param only? Is it an elaboration param only? Is it required for both of them?
> 
> I will be required for all of them

> > • Is it possible to save --base-std= in work-obj.cf so that it only needs to be set/updated once?
> 
> Not sure. Again it is not supposed to be for every user.

These two are related. Saving it in `work-obj.cf` was just an idea for the user not to be required to repeat it in every `-a` and `-e` call.

> > • Allow to upgrade it after it is defined, but not downgrade it.
> 
> No way.

Then, as a result, `--base-std=` must be provided with the same value for all the `-a` and `-e` calls that compose a design, isn't it?

> • Is it possible to build multiple versions of some (standard) packages into the final binary/object? This approach is used in multiple environments (golang, javascript).
> But at some point you have to choose which one is used.

This is what I don't understand yet. Isn't it possible to *version* all the types, funcs, etc. so that internally `std_logic_v87`, `std_logic_v93`, `std_logic_v08` exist? Then sources analyzed with `--std=93` would use `std_logic_v93`, unless the interface needs to be bind to a source analyzed with `--std=08`. I believe that, for may resources `*_v87 == *_v93 == *_v08`. Why does one need to be chosen?

> No, there is no transpilation. When using --std=93, you will use the 08 version of ieee and std, but allowed to only use 93 features for your file. My problem is that you will also be able to use [...]

So, you/we are assuming that VHDL will always be incremental? I.e., 08 contains all the features of 93 plus some new, and the concern is that some 08 features should be hidden. What about *deprecated* features, those that might exist in 93 but not in 08? Or incompatible features, where 93 requires an explicitly different implementation than 08? Note that 93,08 can be replaced with 08,19 or 19,2x.

tgingold, 2019-10-18:
> • Allow to upgrade it after it is defined, but not downgrade it.
> 
> No way.
> 
> Then, as a result, --base-std= must be provided with the same value for all the -a and -e calls that compose a design, isn't it?

Yes.

> • Is it possible to build multiple versions of some (standard) packages into the final binary/object? This approach is used in multiple environments (golang, javascript).
> But at some point you have [...]

At the end, you only want one version of each package (a package can declare a signal or a shared variable).
How do you deal with the interfaces ?

> So, you/we are assuming that VHDL will always be incremental? I.e., 08 contains all the features of 93 plus some new, and the concern is that some 08 features should be hidden. What about deprecated [...]

I think that most problems are corner cases.  Either ignore them and use --base-std if you want to be 100% compliant.

1138-4EB, 2019-10-18:
I'm now starting to understand... 

> > Then, as a result, --base-std= must be provided with the same value for all the -a and -e calls that compose a design, isn't it?
>
> Yes.

Then, I'd strongly suggest to use any other mechanism to define it. Be it an specific subcommand `ghdl --base-std=`, without any other option; an environment variable; some JSON/YAML config file; whatever... I find it to be more of a project/design level option, rather than a command option.

> At the end, you only want one version of each package (a package can declare a signal or a shared variable).
>
> How do you deal with the interfaces ?

Either casting or conversion functions. If the representation of data is the same (the feature is compatible), casting should allow to match e.g. `std_logic_v93` and `std_logic_v08`. Otherwise, a function body is required to define a custom conversion.

I see what you mean with *global variables* (i.e. package signals and shared variables), in the sense that you need a unique value/ref to be the *canonical*. However, I thought that these used some GHDL's internal types, not bind to any std version. I understand now that `--base-std` is precisely to set the *canonical internal types*.

> I think that most problems are corner cases.  Either ignore them and use --base-std if you want to be 100% compliant.

Therefore, GHDL will be less pedantic about distinction of standard versions. As you said, it will allow for strict VHDL 1993 by using `--base-std=93 --std=93`, but the default `--std=93` with implicit `--base-std=08` will allow users to write *93 on steroids*. On the one hand, this sounds good, coherent and easy to use. On the other hand, it might be misleading and frustrating for new users. We will need to make an effort to properly document/communicate it.

Now, going back to option `--ieee`. Does it make sense for it to be a *suboption* of `--base-std`? Hence, it would not be an `-a` or `-e` option anymore. Of course, this won't work for designs with modules that require different ieee variants. However, supporting it would take us back to the discussion about casting/conversion above...

Regarding `-Pxxx` all should be good unless `base-std` of external libs is greater than `base-std` of the current/work project/design, isn't it?

---

> is --base-std referring to the group of standards and --std to VHDL standard? If so shouldn't --std >= --base-std ?

@arcturus140, `base-std` means *the version of the standard libraries that is used*. Therefore, any source you analize must not use features which are newer than the standard library, but you can use/assume older versions. As commented above, we are assuming that any newer version of the standard is a superset of older versions.

---

BTW, what will happen if a user analyses a source with `ghdl -a --std=93` `ghdl -a --std=08`? The last one will be used?

rhinton, 2020-03-13:
I am using David Bishop's compatibility packages that back-port many VHDL-2008 standard library changes that are VHDL-93 compatible.  For example, his `std_logic_1164_additions_c` package provides the missing `"xor"`, explicit definitions for `maximum` and `minimum`, the missing shift and rotate operators, mixed scalar-vector operators, etc.  If I `use` these packages with the `base-std` approach, I believe GHDL will see multiple definitions of these functions and remove them from the namespace, so my compile will fail.  True?

I liked @eine 's concept of linking against multiple versions of the standard library, but I don't know enough about compilers in general or GHDL in particular to help here.  Even if the analysis phase used the language-specific package declarations, perhaps it could match up the `"xor"` call to the compatibility package when compiling in 93 mode, and with the `std_logic_1164` when compiling in 2008 mode.

eine, 2020-03-13:
@rhinton, please note that I am not an expert about the LRM or about compilers. Actually, I'm quite ignorant. Hence, most of my ideas are just thinking loud as an exercise to extract knowledge from Tristan's brain... This is to say that some/many of them are not technically possible, but what I'd like to see/use. My main concern is making VHDL and GHDL easier to step into for users that want to learn digital design.

In this specific case, I think that the issue is with the strong typing of VHDL. The same type/variable defined in two versions of the standard packages are effectively different VHDL types. Hence, from an implementation point of view, Tristan's approach is straighforward, in the sense that a single solution exists for each identifier (assuming that an `and` in 2008 is the same as in 2002, 1993, etc.). My proposal is to include an explicit check/filter to decide which of multiple definitions of `and` to use, and then generate all the implicit conversions between the types. As you see, my proposal involves tons of additional work, which might not be worth.

rhinton, 2020-03-14:
I'm no expert, either.  So I am blindly suggesting two closely-related solutions.

1. Analyze against the standard-specific package declaration.  During elaboration time, "link" the subprogram calls to the "base" package library.  Note that analysis has to remember which functions come from which package, since at "link" time there will be several matching candidates.  This approach assumes that later standard library packages are supsersets of the earlier packages with no changes in semantics.
2. Analyze and elaborate against the standard-specific package.  The elaboration phase needs to include "hints" to force `std93.standard.integer` to be the same type as `std2008.standard.integer` and so forth for all of the other standard library types (including `ieee` packages, of course).  This is the safer approach because it can handle changing semantics and doesn't require later packages to be supsersets of earlier packages.

I am happy to flesh out these ideas if they're useful.  Given my ignorance of the GHDL code base, they are likely both silly. :-)