ghdl/ghdl #981: Running compile-xilinx-vivado.sh fails with llvm on windows

GlenNicholls, 2019-10-16:
**Description**
When I run `compile-xilinx-vivado.sh` on windows with the 0.36-dev release of llvm, I get this error:

```
Compiling Xilinx Vivado Libraries [FAILED]
```

After digging around a bit, GHDL seems like it doesn't map the `unisim` library properly as I see this error for every library after unisim that depends on unisim:

```
Analyzing file 'C:/Xilinx/Vivado/2019.1/data/vhdl/src/unimacro/unimacro_VCOMP.vhd'
Analyzing file 'C:/Xilinx/Vivado/2019.1/data/vhdl/src/unimacro/ADDMACC_MACRO.vhd'
C:/Xilinx/Vivado/2019.1/data/vhdl/src/unimacro/ADDMACC_MACRO.vhd:28:9: cannot find resource library "unisim"
C:/Xilinx/Vivado/2019.1/data/vhdl/src/unimacro/ADDMACC_MACRO.vhd:29:12: unit "vcomponents" not found in library "unisim"
C:/Xilinx/Vivado/2019.1/data/vhdl/src/unimacro/ADDMACC_MACRO.vhd:58:14: entity 'addmacc_macro' was not analysed
```

The weird part is this does not occur with the windows mcode 0.36-dev release and I'm not sure why... I am seeing other problems with mcode crashing, so I switched to the llvm version to see if that fixed my problems, however, it looks like I'm now running into this.

The version of GHDL that I'm referring to is [ghdl-0.36-mingw64-llvm.zip](https://github.com/ghdl/ghdl/releases).

I haven't been able to figure out how to compile llvm yet on windows, but I'll try compiling it from master once I figure out what I'm missing.

1138-4EB, 2019-10-16:
> I haven't been able to figure out how to compile llvm yet on windows, but I'll try compiling it from master once I figure out what I'm missing.

You can use the following PKGBUILD in a MSYS terminal: https://github.com/1138-4EB/ghdl-packaging/blob/master/msys2-mingw/llvm-git/PKGBUILD

```sh
# Install deps
pacman -S --noconfirm base-devel git mingw-w64-x86_64-llvm mingw-w64-x86_64-clang mingw-w64-x86_64-z3 mingw-w64-x86_64-toolchain
dos2unix PKGBUILD
# Build GHDL
makepkg-mingw -sCLfc --noconfirm --noprogressbar
# Install GHDL
pacman --noconfirm -U mingw-w64-*-any.pkg.tar.xz
```

Then, open a MINGW64 and try `ghdl --version`.

Note that it is important to build it in a MSYS terminal and test/use it in a MINGW64 terminal.

GlenNicholls, 2019-10-16:
@1138-4EB Do you have this documented somewhere? That was much easier than all the hoops I was jumping through this morning to compile with llvm.

On a seperate note, this works fine with what is on master. I have no clue why the 0.36 mingw64 llvm binary doesn't work right, but this appears to be okay.

closed 2019-10-16 by GlenNicholls

1138-4EB, 2019-10-16:
I extracted the `pacman` and `makepkg-mingw` commands from https://github.com/1138-4EB/ghdl-packaging/blob/master/msys2-mingw/run.sh. However, that script is reused to build six variants (mcode, mcode-rc, mcode-git, llvm, llvm-rc, llvm-git). That's why I extracted only the relevant commands for your target (llvm).

The six variants are built weekly and artifacts are available in https://github.com/1138-4EB/ghdl-packaging/actions. Hence, you can just download one of them and run `pacman -U` directly.

Unfortunately, this is all work in progress, so there is no explicit documentation yet. In #929, PKGBUILD files similar to mcode-git and llvm-git are used as a replacement of current AppVeyor build scripts. Moreover, #901 is to discuss how to integrate packaging in the organization. You will find that both issues are tightly related. Last msys2/MINGW-packages#5757 is a proposal to upstream stable mingw32-mcode and mingw64-llvm versions.