ghdl/ghdl #982: warning: instance of component "oserdese2" is not bound [-Wbinding]

arcturus140, 2019-10-17:
OSERDESE2 is a primitive from Xilinx that is part of unisim library located in unisims/secureip. 

In GHDL the build output for  OSERDESE2 is located in secureip. Compiling my sources I receive the following warning:

`warning: instance of component "oserdese2" is not bound [-Wbinding]`

While OSERDESE2 is not an encrypted source it declares a component `OSERDESE2_WRAP` which I am unable to locate in Vivado. 

I wonder if my compilation of the vendor library is corrupted or Xilinx does not provide all sources for the primitive. Has anyone had success using OSERDESE2 with GHDL?

1138-4EB, 2019-10-17:
Hi @arcturus140! Although I have not used OSERDESE2 myself, in https://github.com/fcayci/vhdl-hdmi-out it seems that is used with GHDL and GtkWave by downgrading it to OSERDES1. Is it useful for your case?

arcturus140, 2019-10-17:
@1138-4EB I am not familiar with 6 Series Family but they are close enough in behaviour. Thanks!

closed 2019-10-17 by arcturus140