ghdl/ghdl #983: synth: Error when using bin/hex/oct literals in PSL code

tmeissner, 2019-10-20:
**Description**
When bin/hex/oct literals are used in PSL code, an error occurs.

`test.vhd:26:42:error: no declaration for ""`

**Expected behaviour**
These literals should works as they do in VHDL code.

**How to reproduce?**

```vhd :file: test.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test is
end entity test;

architecture beh of test is

  signal s_clk      : std_logic;
  signal s_register : std_logic_vector(7 downto 0);

begin

  process is
  begin
    wait until rising_edge(s_clk);
    -- In VHDL code, hex/bin literals work
    s_register <= x"00";
    s_register <= b"00000000";
    s_register <= "00" & o"00";
  end process;

  default clock is rising_edge(s_clk);

  -- In PSL code they don't work
  -- FOOTER_VALID_hex : cover {s_register = x"00"};
  -- FOOTER_VALID_bin : cover {s_register = b"00000000"};
  -- FOOTER_VALID_oct : cover {s_register = "00" & o"00"};
  FOOTER_VALID : cover {s_register = "00000000"};

end architecture beh;
```

```sh :image: ghdl/ghdl:formal
ghdl --synth --std=08 -fpsl test.vhd -e test
```

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `ad1f40b`

closed 2019-10-21 by tgingold

1138-4EB, 2019-10-23:
Can the tests in https://github.com/tgingold/ghdlsynth-beta/tree/master/testsuite/formal/shifts be updated now?

tmeissner, 2019-10-23:
Yes, I think so. I can update them.

commit 7a3a15b5 (2019-10-21): vhdl: try to convert identifier to token only for identifiers

(and not for bit string literal).
Fix #983