ghdl/ghdl #984: Assigning a constant to a generic from a configuration sometimes fails silently

Xiretza, 2019-10-22:
**Description**
When using a configuration to assigning a constant to a generic, with the constant declaration in
one package and the generic's type's declaration in another, the generic is only properly assigned
if both the type package and the constant package are included in the entity the generic belongs to.

Now that's quite a mouthful, I suggest just looking at the source to understand the problem instead. I'm trying to assign the constant `c` to `generic_check`'s generic `i` from `tester_conf`, which works fine only as long as `c` is visible in `generic_check.vhd` (by uncommenting the `use` clause at the top). If it's not visible, it seems like an array of 0's is assigned instead.

```sh
$ ./run.sh
analyze types_pkg.vhd
analyze const_pkg.vhd
analyze tester.vhd
analyze generic_check.vhd
analyze tester_conf.vhd
elaborate tester_conf
generic_check.vhd:15:9:@0ms:(assertion error): Should be 5, is 0
```

**Expected behaviour**
The constant is assigned to the generic regardless of the packages included in the target entity.

**How to reproduce?**

You can download a tarball of the reproducer [here](https://github.com/ghdl/ghdl/files/3755227/reproducer.tar.gz).

```vhd :file: types_pkg.vhd
package types_pkg is
    type generic_type is array(0 to 3) of integer;
end package;
```

```vhd :file: const_pkg.vhd
use work.types_pkg.all;

package const_pkg is
    constant c : generic_type := (others => 5);
end package;
```

```vhd :file: generic_check.vhd
use work.types_pkg.all;
-- Workaround:
--use work.const_pkg.all;

entity generic_check is
    generic (
        i : generic_type
    );
end entity;

architecture a of generic_check is
begin
    process
    begin
        assert i(0) = 5 report "Should be 5, is " & integer'image(i(0));
        wait;
    end process;
end architecture;
```

```vhd :file: tester.vhd
entity tester is
end entity;

architecture a of tester is
    component check is
    end component;
begin
    checker: check;
end architecture;
```

```vhd :file: tester_conf.vhd
use work.const_pkg.all;

configuration tester_conf of tester is
    for a
        for checker : check
            use entity work.generic_check
                generic map (
                    i => c
                );
        end for;
    end for;
end configuration;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -i *.vhd
ghdl -m tester_conf
ghdl -r tester_conf
```

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: a08e8ce8

diogratia, 2019-10-22:
This is not a VHDL-2008 feature, note the lack of `--std=08` in the command line invocation recitals.  This appears to be a bug.

The title could be more in line with **Association of an actual in the generic map aspect of a binding indication of a component configuration fails silently**.  This describes the failure succinctly.

How **sometimes** figures in it isn't readily apparent, is there a code example where the association does not fail?  That'd be informative.

Where the value 0 for the first element of i comes from is also not readily apparent, evaluating  all 4 elements of i in lieu of a stack trace might or might not be informative.

Xiretza, 2019-10-22:
Right, those may or may not be the proper LRM terms, I'm don't have much in-depth knowledgs of the language specification. As explained in the "Description" paragraph, it works fine if the constant's package is `use`d in the entity itself (even though the name of the constant only occurs in the configuration). The 0 is probably just some unitialized value, when `std_logic` are used for the array elements instead of integers, it results in `'U'`.

1138-4EB, 2019-10-22:
> This is not a VHDL-2008 feature, note the lack of `--std=08` in the command line invocation recitals. This appears to be a bug.

My bad. I normally use 'Bug' for issues that produce a 'Bug occurred' block (a crash in GHDL), and label other issues as 'yet not properly supported' features. I chose quickly between available 'FeaReq' labels and I didn't that this was not about 2008. @tgingold, @Paebbels, do we want to label this kind of issues as 'Bug' or 'Enhancement'?

diogratia, 2019-10-22:
It wouldn't be an enhancement to correct a silent failure which should likely be a higher priority than an non handled exception.

There is no default value for i. Also note the explicit binding indication in the component configuration succeeded - the assertion produced output.

The idea is to determine whether or not the problem is due to pulling a value out of the wrong place in lieu of `c` which is made directly visible in the configuration declaration by use clause.

tgingold, 2019-10-22:
I really like this issue. I think the remarks were too strong.
You have provided a reproducer, and the issue is about a feature rarely used: configuration. You also have provided a work around, which give a hint on the problem.

This looks like an elaboration order issue. Shouldn't be too difficult to fix. Hum, famous last words!

commit 0d427a74 (2019-10-24): vhdl/translate: elaborate dependencies of configurations.  Fix #984
closed 2019-10-24 by tgingold